grbl/grbl #1246: I'm not shure where the limit switches must be installed

ml-hn, 2017-06-11:
Hello

With "Homing" is enabled and per axis I have one limit switch installed. Where the limit switches must installed to? In position zero or Maximum (X,Y,Z)?

thank you very much!

jahnj0584, 2017-06-11:
usually in the bottom left corner of the machine, so that every move
referenced from that position put you into a positive work coordinate

chamnit, 2017-06-11:
Please read the wiki pages. It does explain this in there. The new Grbl v1.1 site has more updated wiki pages that goes into more detail there.

closed 2017-06-11 by chamnit

vMeph, 2017-06-11:
i use on top right position of machine sense CNC machine works on negative coordinates, then i use work positions to mark where i want to start my job