grbl/grbl #1252: Stopping a laser job

rfresh737, 2017-06-22:
I'm running Grbl 1.1e. Is there a preferred way to stop a laser job once your software has sent it? I have a stop button on my CAM program and when clicked, I send a reset 0x18 byte but I noticed it still takes a few seconds or many seconds, before the machine actually stops. Is there a way to get an instant stop?

Thanks...

tklus, 2017-06-22:
I wired up a feed hold button for my laser machine on pin A1. The machine stops instantly and the laser turns off. From that point you can reset

chamnit, 2017-06-22:
@rfresh737 : A reset command should be instantaneous, because it is processed as soon as it is received in serial RX. The only reason it would take several seconds to enact is if the GUI is delaying the send for some reason.

rfresh737, 2017-06-22:
@chamnit I had a call in the GUI that was holding up sending out the reset. I removed the GUI call and the Stop is now immediate...thank you.

closed 2017-06-22 by chamnit