grbl/grbl #1255: GRBL don't suport the resolution for a gearbox 1/10

Thorgrlm, 2017-06-23:
Hi all, I have a problem when calibrating the steps of the motor, I bought a gearbox of 1/10, the driver I have it to 1/32 microsteps, to intruducir the value of the steps GRBL does not support the resolution,  Is there any solution?

Thanks you

mayhem2408, 2017-06-23:
@Thorgrlm What resolution are you trying to achieve?. You need to keep in mind that GRBL has a limit of 30,000 steps for second. The higher the steps per millimeter, the slower it will go.

mayhem2408, 2017-06-23:
@Thorgrlm  Also keep in mind the gearboxes tend to have backlash which may counteract the high level of accuracy you might be trying to achieve. I had a 5:1 gearbox and it's backlash was over 2 full steps. In your case that would be 64 micro steps.

Thorgrlm, 2017-06-23:
I want to make pcb with 0.15mm tracks and 0.035mm copper thickness and I need the highest possible resolution. When buying the gearbox I did not realize that detail :'(, in future GRBL updates will be implemented the support of higher resolutions?

tklus, 2017-06-23:
Perhaps you can achieve the accuracy you require with a small ball screw and pulley gear reductions between the stepper and ball screw. Guess it depends on your machine design as well...

Thorgrlm, 2017-06-23:
The mechanics of my machine is very accurate, I have the standard CNC3040, that's not the problem, but I need more accuracy because the jobs don't stay the way I want

https://es.aliexpress.com/item/EU-STOCK-3-axis-3040-300W-USB-MACH3-CNC-ROUTER-ENGRAVER-ENGRAVING-DRILLING-AND-MILLING-MACHINE/32601108796.html?spm=2114.13010608.0.0.9wdl3l

tklus, 2017-06-23:
If I was going to increase the accuracy of a machine like that, I would remove the steppers, make some new mounts to locate the steppers beside the ball screw and belt drive the screws. you could basically double the accuracy of your machine by running a 2:1 gear reduction with a belt driven system.

I also know that some ball screws from china can have a fair amount of back lash. up to .0015 inches. or .038mm. I would put a dial indicator on your machine and check back lash and slop in your system. you could have the most accurate stepper system but if your linear motion system has slop, it wont matter. I suggest a .00005" dial indicator for checking back lash... https://www.amazon.com/Brown-Sharpe-599-7033-3-Indicator-Graduation/dp/B007WZPVVY/ref=sr_1_4?s=industrial&ie=UTF8&qid=1498226348&sr=1-4&keywords=.00005+indicator

https://www.robotdigg.com/ has lots of pulley and belt options.

if you can't visualize what I am describing in the text, I can draw a picture :)

Tim

Thorgrlm, 2017-06-23:
I have one already.
Now what I will do is put the driver to 1/8 or 1/16 to see which supports me and try.

I have this one
https://es.aliexpress.com/item/100-Real-Japan-Mitutoyo-Dial-Indicator-2109S-10-0-1mm-0-001mm-Dial-Test-Gauge-Micrometer/32648510046.html?spm=2114.13010608.0.0.6ynmH2

tklus, 2017-06-23:
Very good... it is amazing what the indicator can teach you about the machine. Report back with your findings...

https://es.aliexpress.com/item/100-Real-Japan-Mitutoyo-Dial-Indicator-2109S-10-0-1mm-0-001mm-Dial-Test-Gauge-Micrometer/32648510046.html?spm=2114.13010608.0.0.6ynmH2

Thorgrlm, 2017-06-23:
What is the limit in steps that GRBL can reach ?, in my configuration if I set the driver to 1/32 with the gearbox of 1:10, the steps that it gives are 16000, if I put that, I can't raise the speed of 100, GRBL goes crazy

whitetd, 2017-06-23:
Setting s high microstep does not make it more accurite.

109JB, 2017-06-23:
Are you sure it is Grbl that is the problem, or are you exceeding the capabilities of the motors? What exactly happens?  Do the motors stop and give out a whine type noise?

chamnit, 2017-06-23:
@Thorgrlm : Grbl doesn't care what the step/mm is, just as long as you don't exceed the 30kHz step/sec performance limit of the 328 processor. Accuracy is dependent on your machine and your drivetrain. 0.02mm accuracy is typical with most Grbl-based CNCs. You absolutely do not need 16,000 steps/mm to get that. 80-250 step/mm is more than sufficient. If you want accuracy lower than 0.02mm, then you need a very high precision and robust CNC machine. Most hobby machines can't really do less than 0.02mm reliably due to tolerances and lack of rigidity. This is mechanical in nature and has nothing to do with Grbl. 

So, please increase your micro stepping to 1/2 or 1/4. Again, like @tklus mentioned, 1/32 microsteps doesn't help you increase accuracy. Microstep torque between full steps is often weak and won't guarantee the machine moves (and holds) the programmed position.

closed 2017-06-23 by chamnit

langwadt, 2017-06-23:
@chamnit and all that "accuracy" is already out the window with a gear box

mayhem2408, 2017-06-23:
@Thorgrlm You are hitting the limit of GRBL. If you are attempting 16,000 steps per mm, you need to take into account that GRBL has a limit of 30,000 steps per second which equates to 1.875mm per second or a max of 112mm per minute. At 100, you at pretty much at the limit of GRBL. Even faster controllers have limits. I also have a UC100 which can step at 100,000 steps per seconds. For you, that still only a feed of 350mm/m

My machine that I use for PCB engraving uses a 0.9deg step (400 steps per revolution) using a 1/4 micro step driver on a 3mm lead screw with an anti-backlash spring nut.  I am getting 533.333333 steps per mm or 0.001875mm per step. More than accurate enough for my PCB work. 

From experience, too slow means slow chip removal and heat build up. Just like too fast is bad, too slow is also bad. I cut feed at 500mm/m and fast move 3000mm/m for my PCB Work.
The other thing to take into account is flex in the PCB. I use a touch probe to do my levelling. It has a repeatable accuracy of 0.01mm which is 5 times less accurate than my steppers.

109JB, 2017-06-24:
Actually from his description he is NOT hitting the limit of Grbl if his numbers are accurate.

16000step/mm * 100mm/min / 60 = 26.667 steps/sec. 

This is well within Grbl's capabilities. I would suspect that a motor limit has been reached. Either too high accel, current set too low, too much torque loss from microstepping, too much friction in the drive system, etc.

109JB, 2017-06-24:
I looked at the specs for the machine and it looks like it has 4mm pitch ballscrews. If this is the case you don't even need any kind of reduction drive. At 1/8 microstepping a single microstep would be 4/200/8=0.0025mm.  I would guarantee that this is well beyond the capabilities of the machine itself.

X3msnake, 2017-06-26:
Not to mention the spindle runout that most hobby machines has

Thorgrlm, 2017-06-26:
I have read all the comments, I will try to start explaining.

The problem I have is with the height, I use the bCNC software and I use the auto-leveling funtion, my pcb has a copper thickness of 0,35um.

the auto-leveling If I repeat it again, I'll never do the same measure.

When making the Gcode I set it to a pentration height of -0.036 and when manufacturing the pcb using the same Gcode, the tool never penetrates the same priofundity and this destroy the pads of smd

Now I want to test with more precision

electrokean, 2017-06-26:
@Thorgrlm and as I think most of the above comments try to explain, your problem isn't with grbl or its settings, but with the mechanical limitations of your machine.  You'd be best spending some time trying to improve the presicsion of the machine mechanically.  Using a geared motor is almost certainly going to introduce additional backlash and make things worse.
Moving faster is secondary but will become an issue later for making clean cuts as described by @mayhem2408

langwadt, 2017-06-26:
@Thorgrlm your expectation of what accuracy and repeatability is achievable seems way way off. It is a mechanical limitation not a GRBL limitation, try calculation what a few tenths degrees slop in a gearbox adds up to.  

also the copper thickness is a nominal

X3msnake, 2017-06-26:
best technique to mill PCBs is to use a PVC or MDF base and level it befor
you span in your board and mill your PCBs, usually no spacial compensation
is needed if a the board is of good quality it has little gauge variation

What probe are you using? You might consider that you can be introducing
errors in your probing...

electrokean, 2017-06-26:
@X3msnake exactly the method I use!  I have a sacrificial bed (HIPS) that I mill perfectly flat, then I use some thin double sided tape to hold the PCB down.  I can mill two 0.8mm single sided 0.5oz copper PCBs with 0402 and 0.5mm pitch parts, then put them back to back and have perfect registration of holes.  Because the copper is so thin, and the V cutter makes wider cuts the deeper you go, I usually run a few cut passes with the first being convervatively high, and then a touch deeper each time till all traces are clean.  Sometime some manual touchup is needed of fine details with a sharp knife under the microscope.
I use HIPS as it is more stable than MDF, and cuts nicer than PVC.

X3msnake, 2017-06-26:
What angle V Cutter you are using 10º?

You say the Z changes on repeatability. Do you have your Z axis Bushing
correctly tighten?

Never though of using high impact polysterene thanks for the tip i'll have
to try it.

electrokean, 2017-06-26:
I find 10° V cutters aren't great as you need a very small step between cuts, or you'll leave fine copper hairs behind.  I prefer 30°, and for larger pitch designs I was using a 90° V as it does a really good job at clearing (at the risk of removing a little too much copper if you go too deep).

I'm guessing your question on Z repeatability was for the OP?

X3msnake, 2017-06-26:
yes it is, forgot the mentions, thanks for calling it :(

Thorgrlm, 2017-06-26:
Sorry for my poor english, I'll try to explain.
I use tools in V of 10º
 I want to do are tracks of 0.15mm and smd pads for 0.3mm for microcontrollers
I upload some images of some test pcb that I have made.
[img]http://i.imgur.com/PV8aYNK.jpg?1[/img]
[img]http://i.imgur.com/DLP0anz.jpg?1[/img]
[img]http://i.imgur.com/X2OpsnB.jpg[/img]

The problem is that the Gcode that I used to manufacture that PCB does not work me to make another PCB equal.

To make the work table flat, I use a plastic board fixed to the work table, (methacrylate) more later I mill that at a height of -1mm so it is straight, I use double-sided tape very thin to fix the pcb to the Plastic table

Thorgrlm, 2017-06-28:
Hi, I still have the same problems, the software doesn't know where it is exactly and that's why it gives me those problems. Is there any GRBL that uses encoder on the axes?

chamnit, 2017-06-28:
@Thorgrlm : You are clearly ignoring everyone's suggestions on this thread, when they have already given you the answer several times. The accuracy requirement for your PCBs are common and there are lots and lots of examples of Grbl doing this everywhere on the internet. Google it, if you're not convinced. 

Again, this is a mechanical issue likely caused by backlash and inaccuracies of your machine. It's not related to Grbl. Grbl only commands and executes the motion they are given, which are guaranteed as accurate as the step/mm setting you provide. If your machine does not move the amount commanded, then it's your machine. You can always setup a step pulse counter with another Arduino if you don't believe that either.

Rather than waste everyone's time over and over again. Please try searching on the web to see if there is a version of Grbl that supports encoders. Even if there is one, it won't help with your backlash or mechanical problem.

109JB, 2017-06-28:
You are not having a software issue. If you were, everyone would be having this issue and they are not. Your problem is related to your machine whether you want to believe it or not.

1. Have you checked the backlash of the machine? 
2. Have you adjusted the acceleration and maximum velocities of the machine so that you aren't losing steps?
3. Have you adjusted your stepper drivers to the correct current rating for the motors?

Thorgrlm, 2017-06-29:
They are not paying attention to what I say

All that I have checked and I don't see lost steps, I have created repetition patterns and I have stressed the machine for hours and I don't see a lost of steps.

For example, I created a pattern that moves the motor in steps of 0.01 from a height of 0 to 30cm I left it running for 1 hour, with no missed steps (I created several diferent pattern at diferent measures all work fine)

I have said over and over again that when I create a Gcode that works, it does the work at one time and perfect, as I show in the photos above, if I had lost steps I could not do those pcb's I have done, I would have problems since the Z axis does not go down once does the work and goes up, but it goes down and goes up continuously.

And I when trying to use the same Gcode on another PCB doesn't work, I have to re-adjust the height in the Gcode until it does the PCB well, the problem is at the beginning not rest of the job.
For all this can not be a problem of the machine there is only something that escapes me

Hi 109JB
My drivers are TB6600
and yes al all you say to me

If you need me to do some test that comes to mind tell me and we see results

thanks at all ;)

X3msnake, 2017-06-29:
@Thorgrlm 
This might be a silly question, but do you have end switches and homing turned on on GRBL?

Does you keep your stepper motors always on during the time it takes for you to change PCB and do a new take? or do the steppers are allowed to unlock?

vMeph, 2017-06-29:
Most like every time you do a job you will need to start machine,and when you give tension to motors  it will cause the Motors to lock and may lose position, couldnt you use autolevel, or make it Home and then go to a work position that always will be same ?

vMeph, 2017-06-29:
Also whats is your $1=?
If you put the $1=255 the motors will always be on and will not go to sleep,that way should always mantain position

Thorgrlm, 2017-06-29:
I have switches for homing cicle.
$ 20 = 1
$ 22 = 1

No, I haven't tried to finish one job and put another.
I make a pcb, I turn off the machine and the next day I make another one.

I use the bCNC and GRBL 0.9j software
The working method I use is as follows:
I move the bridgue to the place of work, the Z axis I do the test proves, wen the tool touch the copper I stabilize the 0.
I run the autolevel, to the Z axis I do the test test and set to 0 (everything says the bCNC manual)

Now with the values ​​of the auto level start up milling, if I see that it does not reach the height, I change the Gcodes I've done until it starts working at the correct height and make the pcb at once, but if I turn off the machine, and without changing anything I try to do it again, it does not work well

I don't have the configuration of $ 1 = 255, I have it by default, I'll prove what you tell me

chamnit, 2017-06-29:
@Thorgrlm : No. Everyone is listening and giving you the right suggestions. This still isn't a Grbl issue but a machine issue. If your probe isn't repeatable, consider something wrong with your probe or something wrong with your assumptions in your process, as others have already suggested. A properly setup probe has plenty of accuracy for PCB milling. Again lots of users mill PCBs consistently and accurately with Grbl. Your process assumptions may be wrong. For example, PCBs are not consistent thicknesses so you have to probe each one, each time. Or something moved in your setup. Or something mechanical in your machine is causing inconsistency like your steppers disabling ($1=255) or backlash. 

In addition, it would have be more helpful if you explained your situation in the beginning, rather than come to an incorrect conclusion that something was wrong with Grbl. Everyone still gave the correct answered from the information you provided. We are all trying to help you, but if you don't help yourself by considering and testing these suggestions, I will lock this thread.

chamnit, 2017-06-29:
@Thorglrm : Thank you for trying. Your last post did not show up while I was writing my last one. Please set your stepper to not disable with $1=255. I'm fairly confident that this is your problem.

vMeph, 2017-06-29:
If you set your motors $1=255 and do homing go to where you start pcb job set the  G54 (G10 P1 L20 X0 Y0 Z0)work position on the start position and set a G54 on your gcode file and start job, next day you do homing and start file to do job and you should get same result, you motors going to sleep and start machine with out do homing etc, is prety normal that you will never get same results

109JB, 2017-06-29:
With $1 not equal to 255 the stepper drives will disable after a certain amount of time.  When they re-enable they will re-enable to a full step position, not a microstep position, so immediately you will lose steps.  I did testing a long time ago with other drivers in regard to disabling and them re-enabling them. This testing was done in LinuxCNC, but the results should be the same as with Grbl.  In the testing, I found that my machine would move as much as 0.002", which on my machine is 2 full steps, when I re-enabled after disabling. I have all my machines set at $1=255 for this reason.  So each time you disable and re-enable the drive it could be moving and "losing steps", and each time compounds the amount it is off.

chamnit, 2017-06-29:
@109JB : Agreed. There is a note on this behavior in the Grbl wiki configuration [page](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#1---step-idle-delay-milliseconds) (and in [v0.9 wiki too](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#1---step-idle-delay-msec) ).

BTW, I thought I responded to your step rate comment, but looks like I didn't. Your calculation is off by a factor of 1000. `16000step/mm * 100mm/min / 60 = 26667 steps/sec.` Not 26.7 steps/sec. It's below the 30kHz mark, but it's extremely close to it. Bumping up to 112mm/min hits that mark. 

And the 30kHz limit is a limit set by tests I did with what I considered normal usage and a range of expected settings. It should encompass most cases, but there might be some situations that might be a little lower. For example, having a resolution of 16000 steps/mm is well outside the tested range I was expecting.

vMeph, 2017-06-29:
@chamnit @109JB would be nice to  have  on wiki a button  for subjects like calculations that would explain how to use these formulas calculations with some explanations in what they do,belts calculations,rods,KHz etc..

X3msnake, 2017-06-29:
@vMeph I think for the belts and rods the reprap calculators can be used   http://www.prusaprinters.org/calculator/#stepspermmbelt

As for the Grbl Frequency one needs to be done

109JB, 2017-06-30:
> BTW, I thought I responded to your step rate comment, but looks like I didn't. Your calculation is off by a factor of 1000. 16000step/mm * 100mm/min / 60 = 26667 steps/sec. Not 26.7 steps/sec. It's below the 30kHz mark, but it's extremely close to it. Bumping up to 112mm/min hits that mark.

Yes. this was a typo where I forgot to add KHz after. ie: 26.667KHz.