grbl/grbl #1256: can't find the 4 axis pinouts for grbl mega 2560 anywhere

robogrow, 2017-06-26:
I am pretty sure I have the 4 axis version of grbl loaded as it has $110 thru $113 values appear.

But I can't find the pinout diagram for the 4 axises anywhere. I am testing the ports around 24 to 31 but can't seem to find which pins are x,y,z,a step and direction.

vMeph, 2017-06-26:
dont know if this is what you looking for
https://github.com/gnea/grbl-Mega/issues/15

robogrow, 2017-06-27:
>     gnea#15 https://github.com/gnea/grbl-Mega/issues/15

found a post with the info that link only has 3 axis grbl pinouts.

pin function

22  x step

23 x dir

24 y step

25 y dir

26 z step

27 z dir

28 a step

29 a dir

109JB, 2017-06-28:
Just look in the cpu_map.h file. It has all of the info you want.