grbl/grbl #1257: Problem with Y axis during arcs engraving

Sceeker, 2017-06-28:
Hello !
I recently finished the prototype of my first engraving machine, it works well, especially when engraving rectangular shapes.
But when I try to engrave a circle or a round letter, such as a "O" I get this:
[image]

The problem comes the Y axis which is too slow to start moving in the opposite direction.
And while the Y axis is static, the X axis continues to move so I obtain straights lines (as shown inside the red circles) and edges (as shown by the arrows).

Is there a way I can fix this ?
I did my best to explain my problem but there might be some english errors since it's not my native language.
Thanks you for reading.

electrokean, 2017-06-28:
This is caused by mechanical backlash

Sceeker, 2017-06-28:
Thank you, I've found where the backlash is.
Since I used a scanner for the Y axis the stepper motor has a gearbox, and it's where the backlash is.
I think the only solution now is to try to use another stepper motor.

closed 2020-07-26 by Sceeker