grbl/grbl #1258: Problem with motors

adisma57, 2017-06-30:
Hi there,

I'm trying to build a CNC coreXY machine.
I configured grbl to coreXY, and sets $100 and $101 to 80. But, my motors dont turns at the same speed.
One is really fast, the other seems more "normal"

Both drivers are set to the same tension (0.9V).

Any idea where it could come from ?

X3msnake, 2017-06-30:
are the motors the same type? same step angle?

adisma57, 2017-06-30:
Yep, both bought at the same time.

X3msnake, 2017-06-30:
I believe that on a corexy the motors only turn at the same speed if you
are going on a 45 degree angle. Leave all the same and just changr the
motors with each other and repeating the test.

chamnit, 2017-06-30:
Check your microstep jumpers on your stepper drivers.

adisma57, 2017-06-30:
I'll recheck them. 
Is there a way to test them ? Because as I didn't have jumpers, I solders pins together.. ^^

mayhem2408, 2017-06-30:
CoreXY Geometry is quite easy.
Vertical movement is the two steppers spinning at the same speed in opposite directions
Horizontal movement is the two steppers spinning at the same speed in the same direction
45deg movement is one stepper moving and one stepper not moving.

I would check your microsteps on you drivers. If they are the same, i'd check your wiring incase one of your steppers/drivers are missing steps.

adisma57, 2017-06-30:
Thanks for this explanation.
I checked my microstepper. 2 "home-made jumpers" are broken. On the crazy
motor of course.
I'll fix them and test again.

mayhem2408, 2017-06-30:
@adisma57 The following tests are based on your steps per mm being set at 80. and your steppers microstepping at 1/16

You can check each motor by first going to your origin:
`G0 X0 Y0`

To turn one motor in one direction for one full revolution over 10 seconds
`G1 X40 Y40 F240`

To turn the other motor in one direction for one full revolution over 10 seconds
`G1 X80 Y0 F240`

To turn both motors in the same direction for one full revolution over 10 seconds
`G1 X40 Y0 F240`

To turn both motors in the opposite direction for one full revolution over 10 seconds
`G1 X40 Y40 F240`