grbl/grbl #1259: Feed Rate in inch/min regardless of G21 Command?

DeltaPrinter88, 2017-07-05:
Even after disabling "AMASS" I am unable to get a feed rate of 1mm/min.  Oddly enough, it is constantly pegged to 1 in/min, or completes a 25.4 mm line in one minute when G21 F1 X25.4 is sent.  My machine does 80 steps/mm.  By my math, at 4 step/sec (quoted lowest speed) I should be able to at least get close to 1mm/min.  Am I missing something or is there a chance F is being read as in/min regardless of being in G21 mode?  I'll refine and add info to this post later today.  Gotta go to class! (Calc III summer 2017)

chamnit, 2017-07-05:
You have to lower the planner minimum speed as well to go below 1mm/min.

closed 2017-07-05 by chamnit

DeltaPrinter88, 2017-07-05:
I just want to get to 1mm/min, not go below.  I'll look into the planner speed though, thanks!

chamnit, 2017-07-05:
@DeltaPrinter88 : You should be able to get 1mm/min without AMASS enabled. And there is no chance that Grbl can misinterpret a distance with  G20 or G21, otherwise there would be a ton of people with problems. I would check if you truly did disable AMASS correctly. You may have edited the wrong config.h file. It should be the one that was copied to your Arduino library folder.

chamnit, 2017-07-05:
@DeltaPrinter88 : BTW, 4 steps/sec gives you about 240 steps/min. At 80 steps/mm, you won't get below 3 mm/min.

109JB, 2017-07-05:
Are you in G93 inverse time mode?  If so, then your commanded line (G21 F1 X25.4) would be telling Grbl to move the X axis 25.4mm and take 1 minute to accomplish it.

DeltaPrinter88, 2017-07-05:
Thanks for the quick responses guys.  

@Chamnit: you know, I really may have been editing the wrong config.h file location...:D oh boy!  I'll update when I'm home in an hour.

:D

DeltaPrinter88, 2017-07-05:
Update: Thanks Chamnit, you caught my silly mistake.  I had indeed been updating my backup file of grbl instead of the one in my arduino library.  Such a rookie mistake!  My 8W laser is now happily cutting through 1/8" acrylic at 1mm/min....which may take a while, but beats buying a more expensive machine.