grbl/grbl #1260: CNC Plasma THC integration

bduplan, 2017-07-06:
I am running 0.9j with a v3.51 grbl shield.  It works well on my 4x4 foot cnc plasma table, but I am having a hard time getting a torch height controller to work.  I programmed a second arduino uno to act as the torch height controller (THC) and it works until the plasma power source is turned on.  The plasma arc puts out a lot of EMI, and I just can't keep the THC uno from crashing  and rebooting in the middle of a cut.  It works fine some times but not others.  I've shielded and grounded everything and the grbl controller is fine.  I've tried for a couple months between designing, building, and troubleshooting.  Anyway, I'm at the point now where I'm ready to just bite the bullet and buy an integrated unit like a Proma THC.  I really need to get the machine working and use it.

The THC takes the SpinEn output, which triggers the plasma arc relay when no THC is used, and intercepts it.  It uses that as a trigger to do a probe first, and then it sends the signal to the relay to start the plasma.  It then waits a predetermined amount of time (~0.5s) and sends an "Arc OK" signal back to the CNC controller, telling it to begin X and Y movement.  My Hypertherm plasma machine can also send an Arc OK when it detects that the arc has pierced the metal.

So my problem:  The "arc OK" feedback could be sent to grbl as a resume command, but I can't figure out how to hold it in the first place.  I was hoping there was a way to put a command in the post processor that would put grbl into a hold state.  I'm pretty new to this, so I tried putting a "!" on the line immediately following the M03 command, and I'm sure you know that this threw an error for an unsupported g-code command.  But I mention that to illustrate what I am trying to accomplish.  Is there a way to make grbl essentially wait for that Arc OK signal before continuing?

Thanks for any help!

bduplan, 2017-07-07:
I stumbled around a bit and think I found a solution.

in the gcode.c file, I found the following block of code for spindle control:
`
// [7. Spindle control ]:  
if (gc_state.modal.spindle != gc_block.modal.spindle) {
  // Update spindle control and apply spindle speed when enabling it in this block.    
  spindle_run(gc_block.modal.spindle, gc_state.spindle_speed);
  gc_state.modal.spindle = gc_block.modal.spindle;
}`

And added another if statement to it.  So the intent is that if the command is to turn the spindle on, it executes a feed hold with code that I found elsewhere in the file.

`
// [7. Spindle control ]:
if (gc_state.modal.spindle != gc_block.modal.spindle) {
  // Update spindle control and apply spindle speed when enabling it in this block.    
  spindle_run(gc_block.modal.spindle, gc_state.spindle_speed);
  gc_state.modal.spindle = gc_block.modal.spindle;
  if (gc_block.modal.spindle == SPINDLE_ENABLE_CW) {
    //Execute a feed hold so that the THC has time to probe.  The THC will resume via the resume pin
    bit_true_atomic(sys_rt_exec_state, EXEC_FEED_HOLD); // Use feed hold for program pause.
  }
}`

So now when the gcode sends an M03 command, it goes into a hold.  I tested this on a gcode file that fires the torch twice in the process of cutting out some letters.  Grounding the resume pin on the shield then resumes the program.  The '~' resume command sent through UGCS will not work when it is in this hold, but I should be able to live with that.  I'm just wondering if that's an indication that I am overlooking something which could cause problems.

vMeph, 2017-07-07:
Would a G4 dwell with a certain time to pause, work for what you want?

bduplan, 2017-07-08:
No, because the probe will take different amounts of time depending upon how far the torch head is from the work piece at the start.  Once the probe is finished, the THC sends the signal to the plasma power supply to fire the arc, so the CNC needs to resume movement at a precise time with relation to that arc.

cri-s, 2017-07-08:
I think it is the wrong way.
As example if you start ARC, and ARC_OK  don't arrive in 2 seconds
error must be issued
and all must be shutdown. It is possible that you handle this
externally, same as the
pressure sensor (microswitch and silicone tube) in order to shut down
if pressure is missing,
and at the same time you must/should time ER and consumable usage.
Doing THC is simple and i don't understand you'r problem with arduino,
maybe just doing
a pic of the arduino board and schematic of connections including
grounding.can show the
problem. maybe overall photo of plasma table and cable chain is not bad.

cri-s, 2017-07-09:
THC is simple, what is needed on THC side:

 ―> ARC OK
 ―> THC UP
 ―> THC DOWN
 <― Torch ON
 <― protocol /Charge Pump
 <― THC LOCK

If instead of UP/DOWN it should emit pulses, instead of up/down
 <― Z DIR
 <― Z STEP
 ―> THC STEP
 ―> THC DIR
and if not  plasma is not on Z is passed directly, and during run
simple pulse generator
is emitted

During plasma run, normally Z axis is not driven, it is only driven
in two cases:
material height checking for setting correct starting height and when
using small material
that is uneven cutting faster as the control loop.

Basically there is 2V 50/60Hz noise overimpressed and this is the
reason why higher
voltage is preferable to lower voltage as  having 150V and 2V noise is
different as having
2.94V and 2V noise.  That is no problem, but it means that as example
having 50Hz one
need integrate 2500 samples (50khz) or with 60hz 3000 samples. This reduces max
single sample value to 50 or 60 / sec . If averaging 4 samples, that
is low averaging,
this is 12.5 or 15  real values / second. As there is a feedback for
up/down this works only
good for thicker material. Thinner material require either  previously
mapping using
ohmic sensor and software correction or using higher sensing voltage.

What is required on Grbl.
It depends what you use and if you use the WDT.
Example for  TORCH-ON  it have to fire automatically the WDT with 2Sec
interval and if
that gives interrupt it should interrupt waiting for feedback signal
and return ERR.
Plasma have two sensors, one is the ohmic sensor, that usually is
automatically deactivated during run and the other is the contact
sensor and/or collision sensor.
Both can use the probe input and maybe m4 to switch the sensors.
As both inputs if this are triggered need to pause immediatly it
should be mapped
to limit swith and if plasma is running and probe input is active it
should stop ASAP.
A4 could be used for reading back the signal for ARC-OK
What other requirement plasma table have.

Generally plasma table can work without limit switches and need Z
control only for
pierce height and often there have servo actuated microswitch for
checking pierce height.
During work, (ARC-OK)  there don't use Z pins and this pins can be
used for other purpose if relay exchange this externally. Further Z
pins

DIR XYZ
STEP XYZ
ENABLE
LIMIT XYZ
M3 Spindle enable
M4/M12 Spindle direction
ABORT/FEED-HOLD/START
M8
M7/M10
Probe

plasma cut need:
DIR XY(Z)
STEP XY(Z)
ENABLE
ABORT/FEED-HOLD/START
M3 (Torch ON)
M4 (Scribe ON)
M8 (LOCK)
M7 (exchange Z axis or for pneumatic Z, enable it)
Probe (ohmic sensor)
Limit Z (Limit + and - + collision switch) operator panel have a
switch to route collision switch to probe input
Limit X is setup for homing and limit switches.
Limit Y is used for ARC ON input

The modifications needed are:
M4 need to be off using M5, hovewer M12/M13 could be used instead to
have the same behaviour.
2 sec timer for ok and check for limit X including enabling the
software method to set and get method on (arduino) THC in order to
count consumables usage , seconds)
Limit Y should not go to Stop mode, instead it should do pause and/or
record line numeber it have executed
because it is frequently, that plasma works for 10 minute, pause 90
seconds and then again works for 10 minutes.
If this happen it is crucial for the sender software to know what code
and position it have executed
because it need to fire the torch again, and if that don't work wait
30 seconds, fire it again, wait 60 seconds,
fire it again, wait 120 seconds fire it again, wait 300 seconds, fire
again, wait 600 seconds and tray the last time.
and continue to work if it is possible.

Postprocessor must instead of M3 generate code like this:
( G31Z-100 )
G38.2Z-100 F100
G92Z0
G0Z4.8 ( or the height calc from program , use variables inside
postprocessor for pierce height)
M3 (here the 2 sec timer should be fired and after limitY is ON the
limit should be included into limit mask)
G4P0.8 ( Pierce delay, thc delay is set at 1 second )
F3200  ( or the speed calculated from program, use variables inside
postprocessor for pierce height).

for setting it or to store/recall it. Eventually G70
The big problem is that for different material or speed one need to
run the postprocessor again because it is not possible to use
variables
and every pierce (m3) trash the F speed. and that is a big problem.
In this case i propose to add 10 variables that could be set using
$99N and recalled using #99N , #97N could be input pins , ..., #90N is
the current system position, #91N is the planner/Gcode position ...
and add the notation [ #990 = X + 3.2 ] adding a small expression
calculator, ignoring the [ and ] beside using it as ( ) and [ #NNN =
calls should be equivalent as $NNN = , but allowed inside gcode and
limited as example to 95x to 999 range. Expression parser is simple,
but it is not needed at all, just adding the #9NN notation and setting
it manually could be enought.

What program for sending code you use or write ?

2017-07-08 17:17 GMT+02:00, Cri S <phone.cri@gmail.com>:

bduplan, 2017-07-09:
Thank you for the comprehensive reply.  I have to admit that I will need a little time to digest most of that.  I understand THC operation, as I put a lot of thought into designing my first one.  The second Uno handled all of the functionality that you outlined.  The problems that I had were istability.  It crashed when cutting sometimes.  It would do fine for a few cuts and then reset.  Because I had material thickness identification handled through the serial monitor, it would reset and ask the user for the thickness in the middle of a cut.  The other problem I had was getting an accurate arc voltage.  It read battery voltages and anything from 0-5V very accurately, but the cut voltages were a bit noisy and generally lower than they should have been.  The Hypertherm Powermax 45XP that I have gives a divided arc voltage, but I guess it is not filtered well.  That is something that I need to test.  I can make an RC filter on the op-amp board for the voltage measurement.  I had to use an op amp because the Hypertherm gives divided arc voltage (50:1) on a (+) and a (-) pin, but neither are ground-referenced.  So I made an op-amp circuit that ground references the voltage measurement so that the arduino can read it on an analog pin.  I have one more idea for making that arduino more stable, and I will test it out by simply measuring the arc voltage and seeing how the Uno fairs.

Yesterday, I managed to get a reliable probe sequence in my post-processor.  Interestingly, if the probe is followed by an arc, I get Error 33 and it is not executed.  The arcs do indeed lie on the path to within 0.0005".  By changing my lead in from an arc to a line, it works.  I verified this behavior on the latest grbl 1.1 as well as 0.9j.  

(As a side note, the latest stable UGCS (1.0.9) doesn't show machine position on grbl 1.1 but does on 0.9j.)

I bought SheetCAM to write the gcode.  SheetCAM stores tools, which I have one for each material thickness.  Re-processing for different thicknesses isn't a problem for me.  I used the Mach3_plasma post and modified it a bit.  This part gives me the probe:
`function OnPenDown()
   --Probe
   post.ModalText (" G38.2")
   post.ModalNumber (" Z", -10, "0.0000")
   post.ModalNumber (" F", 300, "0.####")
   post.Eol()
   --Reset Z axis and feed rate
   post.Text (" G92 Z-4.8500\n")
   post.ModalNumber (" F", 4500, "0.####")
   post.Eol()
   --Move up above pierce height (for backlash)
   post.Text (" G00 Z7.0000\n")
   --Move to pierce height
   post.NonModalNumber (" Z", pierceHeight * scale, "0.0000")
   --Below here was already in the post
   post.Text ("\n M03\n")
   if (pierceDelay > 0.001) then
      post.Text (" G04 P")
      post.Number (pierceDelay,"0.###")
      post.Eol()
   end
end`

The above is working well, so all that is left is the adjustments while cutting in response to the divided arc voltage.  If I can get my other Uno to handle that, I may use that.  The issues that I expect to see with that, from experience, are:
1.) Slow adjustments.  Using the AccelStepper library might be the cause.  Maybe I should just use the stepper library for faster moves.  I tried turning the acceleration up but it was still slow.  It was averaging the values at 5Hz, but when it had to make adjustments, it was no faster than 1Hz.
2.) Passing grbl Z steps through the THC.  Perhaps this is a job for an interrupt function?  Remember, I am new to this.  This is my first arduino project actually.  I had it working before by having the THC Uno controlling a relay that passed either grbl steps or THC steps and direction.  It worked, but was crude and a bit unreliable.  The Proma THC SD actually appears to use relays as well, based on the documentation.
3.) Accurately measuring the incoming voltage, as I mentioned before.
4.) Storing arc-on time for consumable life after the Arduino powers down.  Is this just something I would have to enter manually on power-up?  I don't run a production shop, so my consumables should last many sessions.

cri-s, 2017-07-09:
The output of hypertherm as stated 50:1 in reality is 51:1 and for
tollerances, it is maybe
different too. Further i think you don't have the required cap on Vref
and maybe yoiu need
to calibrate the 5V value too.
Store values inside eeprom, if it crash you don't need reenter the value.
Do you have connected you'r second arduino using maybe usb or if not
without galvanic
isolation to the pc ?  What do you mean with ground reference ?
pierce delay and thc delay need to be correlated, and because that
often it  is fix .
10 mm is a bit small for checking height, and you should return to
homing Z height first,
G53Z0 as example. and you have confusion with pierce hieght,
hardcoding it at 11,85 mm and then using pierceHeight * scale.

1) use pwm instead  or emit it between adc readings.
2) this is simple if you have selected two differrent ports with same pins.
   as example and just a guess:
#define plasma_start  ((PLASMA_PORT&_BV(PLASMA_PIN))==0)
#define Z_MASK (_BV(Z_DIR_PIN)+_BV(Z_STEP_PIN))

   noInterrupts();
   while(!plasma_start) {
        PORTB=(PORTB&~Z_MASK)|(PORTA&Z_MASK)
  }
  interrupts();

  here PORTB is step/dir output and PortA is step/dir input using the
same pins on both
ports.

I don't know the other software, but just the millis timer interrupt
from arduino can be the
cause of loss step if not using something like above.

3)  without any connection diagram and description and photo, i can
only guess, based
on the response in this mail, the guess is that you made big mistakes
with grounding/emi.
4) if nozzle is rated for 100.000 seconds and you need changing it at
20.0000 seconds
indicate something is wrong.

bduplan, 2017-07-10:
Re: Arc voltage cal.  The hypertherm has an amphenol port on the back that has pins for triggering the arc, a ground pin, an arc ok pin (dry contact to ground), and (+) and (-) for divided arc voltage.  I first tried measuring the voltage by wiring the (+) pin to an analog pin on the THC Uno and grounding the (-) pin.  That gave very inconsistent values.  The (-) pin read voltage when measured to ground while cutting, and it was both positive and negative at times.  So I built a differentiator op amp circuit that measured the difference between (+) and (-) and then output that voltage (that is what I called ground-referencing it, because the voltage was now measured from the common ground that all of the controls use.)  At first, my gain was 1 on the op amp circuit, and you are right, it did have problems at 5V.  So then I turned the gain down to 1/2 by adding resistance and now it measures a battery at 1.63V, two in series at 3.26V, and Vref at 5V accurately.  At 5V, it measured 4.96.

I'll have to look up how to write to eeprom, but I didn't know that was possible programmatically so that's great news.

Both the grbl Uno and my THC Uno are connected to a laptop via USB.  The THC Uno seemed like it was running out of power so I power that with a 1A 9V DC wall wart as well.  I think that is a big contributing factor to my crashing problem, because the wire is not shielded and runs outside of my electrical box.

I have a gutted household circuit breaker box that houses the electronics.  It is mounted to the table frame with rubber isolators (electrical and vibration isolation) and grounded directly to an 8' ground rod with 4ga stranded wire.  The table frame is also grounded separately to the ground rod.  All wiring (other than that power wire) is shielded and grounded to the tale frame.  I have ferrite cores on the wires just before they go into the electronics box.  All holes in the box (one for a 120mm fan, and three more for air exhaust) are covered with aluminum window screen.  The grbl Uno never crashes.

My torch head floats on bearings and 4.85mm was the measurement between the torch head touching the work piece and tripping the probe switch.  So when the probe switch opens, the Z axis below zero because it has to come back up before the torch starts raising.  I then take the torch up to 7mm above zero and come back down to account for backlash in the screw drive that runs the z-axis.  This ensures that I get the right pierce height (0.15") and that the z axis responds as soon as grbl calls for it to drop to cut height (0.06").  I've verified that this works with a dial indicator and it's within 0.003" consistently.

Thank you very much for taking the time to help me out!  I will look into the _BV function and the interrupts.  I was using interrupts to measure X and Y steps so that I could determine if they were feeding quickly enough to make Z adjustments or if I should stop THC while slowing for a turn.  I'll post my THC code later on, after I remove the probing routine since that's working in grbl.

Have you experienced the Error 33 after putting a G02/G03 arc after a probe?  If I comment out the G38.2 line and leave everything else the same, the arc works fine.  I have it working now with linear lead-ins rather than an arc.

cri-s, 2017-07-10:
Normally you ground the plasma table, that is (+) and (-) is negative
relative to earth,
but if clamp is not attached to table, it is floating relative to
earth , literally hundred of volts.
THC don't have to ground, Never, Never, it is a security risk beside
having wrong readings and instabilitiy, and use only real
transformators, not dc/dc .
If you need to connect the THC to laptop for several reason, use only
optoisolated serial
connection, two cheap optocoupler and resistors are enought. If you
don't have serial
converter, you can usa arduino as usb-ttl converter connecting reset
to ground and
using TX and RX (pin 0 and 1) for rs232. Connect 1kohm resistor in
order to protect pins
if you connect the wrong pin, as TX receives and RX sends.
THC have to adjust height based on current, during plasma run, grbl
don't have to emit any Z movement except from piercing.

error 33, increase decimals to 4(mm) or 5(inch) or make convert
project to metric and
then do CAM and use metric gcode filie, it is accuracy problem,
especially seen inside
autodesk products.

bduplan, 2017-07-10:
The plasma table is grounded to the ground rod that's driven into the earth 8 feet.  I clamp the big clamp from the plasma power supply to the workpiece usually, but I have clamped it to the table and to the ground rod (a suggestion from Hypertherm's Jim Colt to someone else.)

I don't understand when you say not to ground the THC.  How do you power it or have it send signals to/from the grbl Uno without sharing a ground?

My gcode is in metric with 4 decimal precision.  It only errors if the G38.2 is there.  It doesn't error if I comment that out and leave the G92.  (The coordinates are all wrong, but it doesn't throw the error.)

cri-s, 2017-07-10:
Eventually i can run it on simulator and check the real reason.

post or send schematic and connection diagrams or scetches and i can
tell you more if
you don't understand the grounding statement. If you want g**gle
search for "Class II" device.

bduplan, 2017-07-10:
Will do. Hopefully I'll have some time when I get home this evening.

zentoolworks, 2017-07-10:
Hi, guys, 

I just got our test platform hooked up and will run a plasma/grbl test tomorrow 6:00 pm PDT,  live on our youtube channel.  We are using Lotos LTP5000D with CNC Straight torch. Grlb version 1.1f. 

[Youtube Live!](https://www.youtube.com/live_cards?v=yy4WT_8bHLQ)

Thanks

Xin

mkeyno, 2017-07-11:
@zentoolworks  is there more technical data such as pinout , wiring and tech data about LTP5000D ?

bduplan, 2017-07-11:
Here is how I have the Uno THC wired up with the relays and the Z driver.  The X and Y drivers (and the cloned X on A) are 8825's on the grbl shield.  My next step was to take the wall wart that's powering the THC and get power from inside of the metal box, after the EMI filter.  It's similar to this: [https://www.picclickimg.com/d/w1600/pict/121235665835_/1pc-NEW-AC-Power-Line-EMI-Filter-10A.jpg](url)

Diagram:
[http://sickspeedmonte.com/Wiring.jpg](url)

I don't really need to have the THC hooked to the computer while cutting, if it's a problem.

zentoolworks, 2017-07-11:
@mkeyno

LTP4000D Lotos is a 50A machine without any probe feedback. I am using Pin12, spindle enabling, to control the plasma torch trigger. Currently, I just have X and Z axis setup, so it will only cut out a straight line. Eventually, I want to probe the cutting area first, and use that data to compensate the torch height.

Here is a sample gcode sequence that will be sent over to grbl shield. We assume that the surface is flat for now.

S1000
M5
G4 P0.5 (the delay can be adjusted based on test results)
G1 X20 F100 (the federate will also need to be adjusted according to result)
M3
X0

cri-s, 2017-07-11:
As you don't have send schematic or block schematic of THC i cannot
full advice the problem..

- do you have inductors 1mh / 220mh / whatever , LC filter  that block the HF ?
  it is not needed but it block a lot of noise.
- The power supply and THC you have is grounded on GND.
- plasma table, the grounding is on (+) and not on (-) , and if the
THC is not galvanically
isolated (optocoupler, .... ).
 I'm sure you not have connected plasma(+) to gnd on thc, but plasma
(-) and this create the problems for galvanic isoltation.

bduplan, 2017-07-11:
By schematics do you mean the sketch? As for hardware, its just an Uno.

I dont have inductors but I do have resistors and capacitors for an RC
filter that I haven't put together yet.

Should I just unhook the ground to the power supply, leaving the load and
neutral wires? The metal chassis of the power supply contacts the main
electrical box which is also grounded to a different ground rod.

The plasma ground wire does go to my ground bus bar. I can disconnect it.
The (-) of the divided voltage goes to that op amp so the difference bw (+)
and (-) can be determined. If I hook it to the THC ground, then the (-)
will have a negative voltage. I didnt think that could be read on an Uno's
analog pins.

Thanks
Bernie

bduplan, 2017-07-11:
Here is the code on the THC Uno.  I haven't run this particular sketch yet.  I had been running another version which included probing, so I just removed the probe pieces and this is what's left for the closed-loop arc voltage feedback portion.

// Libraries ===========================================
#include <AccelStepper.h>

// Pin Assignments =====================================
const byte arcVoltPin = 1;                              //analog input

const byte xStepInPin = 2;                              //digital input  0 and 1 are reserved for TX/RX
const byte yStepInPin = 3;                              //digital input  2 and 3 are the only valid interrupt pins
const byte spinEnInPin = 5;                             //digital input

const byte zStepControl = 6;                            //digital output  High Level Trigger Relay
const byte zDirControl = 7;                             //digital output  Low Level Trigger Relay
const byte zStepOutPin = 10;                            //digital output
const byte zDirOutPin = 13;                             //digital output

// Parameters to tune ==================================
float vHyst = 1;                                        //Hysteresis for OK-Band during THC closed-loop operation (undivided arc voltage, V).  If the measured voltage is within the target, +/- vHyst, tortch height is not changed
float zBacklash = 1.81;                                 //mm, backlash in z axis
float THCsteps = 0.5;                                   //mm of adjustment on every THC iteration in closed feedback loop control
float loopFreq = 5;                                     //Hz, frequency of Z adjustment in closed-loop
float zMaxSpd = 10;                                     //mm/s
float zMaxAccel = 1000;                                 //mm/s^2
float ArcVoltDivisor = 100.6;                           //Voltage divider, set at 50:1 (times a 2.012 op amp gain)
float xstepspmm = 20.32;                                //X steps per mm
float ystepspmm = 20.32;                                //Y steps per mm
float zstepspmm = 102.00;                               //Z steps per mm
float closedLoopSpeed = 400;                            //Z steps per sec
float closedLoopAccel = 50000;                          //Z steps per sec^2
int sampleSize = 80;                                    //Number of samples for running average.  Max of (1000/loopFreq)

// Initialize variables ================================
String tTarg;
float startx;
float starty;
float startt;
float endt;
float freqx;
float freqy;
float freqPythag;
volatile unsigned int xsteps;
volatile unsigned int ysteps;
float arcVsum;
float arcVavg;
float vTarg;
float frTarg;
boolean validInput = false;
boolean lastMoveUp;
unsigned int charCount;
float i;
unsigned int floatLowCount;
unsigned int floatHighCount;
float arcV2Counts = (1024/(5*ArcVoltDivisor));
unsigned int SpinEnInCount;

// Define a stepper and the pins it will use ===========
AccelStepper stepper(AccelStepper::DRIVER, zStepOutPin, 
  zDirOutPin);

void setup() {
  // Set input pins ====================================
  pinMode(xStepInPin, INPUT);
  pinMode(yStepInPin, INPUT);
  pinMode(spinEnInPin, INPUT);
  pinMode(zStepControl, OUTPUT);
  pinMode(zDirControl, OUTPUT);
  pinMode(zStepOutPin, OUTPUT);
  pinMode(zDirOutPin, OUTPUT);
  
[... 182 lines cut ...]
          moveWithBlock();
        }
        Serial.println("^");
        charCount++;
        lastMoveUp = false;                                 //Indicate that the last move was up
      }
      // OK, No Move
      else {
        Serial.println("-");
        charCount++;
      }
    }
    else {
        Serial.println("S");
        charCount++;
    }
    
    // Carriage return Up/Down feedback
    if (charCount >= 50) {
        Serial.println("");
        charCount = 0;
    }
  }
  Serial.println("");
}

cri-s, 2017-07-12:
yes, reading negative voltages is possible, need different circuit.
The problems are different ground connections.
I suggest you to buy usb to rs232 ttl interface (6 pin) or use arduino
for that holding it in
reset state for using only the usb converter. and use optoisolator in
order to protect
the pc/laptop.
http://www.deep-shadows.com/hax/wordpress/?page_id=372
https://softsolder.com/2010/01/13/arduino-serial-optical-isolator/
Just RX and TX is needed

Probably PC is grounded and 24V supply is grounded, maybe use power
cord with multiple power connectors where you remove the ground
connections.

If you have finished the debugging, just add optocouplers to the THC
for the signals that
come in and going out. If motor driver have optocouplers, output
optocouplers are not needed.

for eeprom, here is a example of settings configurations.
It works without problems if you add new variables at the end, in case
you change
structure on already written data to eeprom , not only adding
additional variables,
you must increase the VERSION, and VERSION cannot be 0xff or 0.

#include <avr/eeprom.h>
#define VERSION 1

struct settings_t
{
  long alarm;
  int mode;
} settings = {
	1234L,
	0,
};

void setup()
{
  if(eeprom_read_byte ((void*)0)==VERSION)
  eeprom_read_block((void*)&settings, (void*)3,
eeprom_read_word((void*)1)&0x3ff);
  eeprom_update_byte ((void*)0,VERSION);
  eeprom_update_word ((void*)1,sizeof(settings));
  eeprom_update_block((const void*)&settings, (void*)3, sizeof(settings));
  // ...
}
void loop()
{
    // let the user adjust their alarm settings
    // let the user adjust their mode settings
    // ...

    // if they push the "Save" button, save their configuration
    if (digitalRead(13) == HIGH)   eeprom_update_block((const
void*)&settings, (void*)3, sizeof(settings));
}

For adc reading, you need to use external VREF or at least calibrate
it and add 0.1 cap to Vref pin on arduino.

long readVcc() {
  long result;
  // Read 1.1V reference against AVcc
   uint8_t mux = ADMUX;
   if(settings.flag&F_VREF) return settings.calib; // external Vref,
no need for this

 #if defined(__AVR_ATmega32U4__) || defined(__AVR_ATmega1280__) ||
defined(__AVR_ATmega2560__)
   ADMUX = _BV(REFS0) | _BV(MUX4) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1);
 #elif defined (__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) ||
defined(__AVR_ATtiny84__)
   ADMUX = _BV(MUX5) | _BV(MUX0);
 #elif defined (__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) ||
defined(__AVR_ATtiny85__)
   ADMUX = _BV(MUX3) | _BV(MUX2);
 #else
   ADMUX = _BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1);
 #endif
  delay(2); // Wait for Vref to settle
  ADCSRA |= _BV(ADSC); // Convert
  while (bit_is_set(ADCSRA,ADSC));
  result = ADCL;
  result |= ADCH<<8;
  ADMUX = mux;
  result = settings.calib / result; // Back-calculate AVcc in mV
  return result;
}

on you'r code, arcVsum if using float this could introduce errors, use
 unsigned long instead. and increase loops to 10 hz.
Check if it works better
if  (arcVavg > vTarg + vTarg>>3)  do_nothing
instead of the 80% speed.

If doing correction, you must speedup the loop as example averaging 10
readings only
and emitting steps.

just start at 2500 uS for step rate and  every step decrease by 4 to
minmal of 700 uS ,
and if ok, increase the steps by 16 up to 2500 uS.   This is for full step,.
For microstepping use 625 down to 175 and decrase by 1.

bduplan, 2017-07-18:
I ordered some opto-isolators that should be in Monday.  I also ordered a serial cable and a serial to USB convertor.  I will opto-isolate the TX/RX pins, use 1k resistors, and connect them to the serial cable for communicating with my laptop.  I will also isolate the rest of the inputs/outputs going into the arduino THC.  The Z-driver I'm using does have isolation on the board.

I'm also going to try to get rid of the ground loops.

I've been trying to make sense of the port mask code for the Z steps and direction.  First of all, I assume that I will need diodes to protect the grbl arduino when the THC step/dir pins go high if the grbl pins are low.  

As far as the code, I've been reading about the arduino ports, and I think I follow how you are changing the input/output status of the Z step/dir pins.  According to this: https://www.arduino.cc/en/Reference/PortManipulation , Port B is digital pins 8-13.  I don't see anything about Port A though.

lalo-uy, 2017-07-26:
> Reference/PortManipulation , Port B is digital pins 8-13. I don't see

Be sure to polarize the base of the optos, other ways they will deform the
115k bps signal.

RichardFalcos, 2017-11-15:
Oi, para todos, sei que esta discussão ficou um pouco para trás, estava acompanhado os comentários, gostaria de saber sobre o desfecho final, se o colega bduplan conseguiu resolver seus problemas com o THC, e se possível passe para nós suas experiencias tais como, circuitos, placas, programas e dicas, pois creio que muitos estão também com o mesmo problema, inclusive eu. Valeu...

bduplan, 2017-11-16:
I needed it to work and after a couple months, I caved in and bought a Proma SD. It works great now and I have been cutting ever since.

closed 2017-11-16 by bduplan

nicolaux, 2017-12-30:
Stò usando grbl per il mio cnc taglio al plasma autocostruito tutto funziona bene fino a che non accendo il plasma ,allora il programma si blocca e devo staccare usb e ricominciare ma aimè si blocca di continuo .In principio questo difetto lo faceva  meno e ho potuto tagliare alcune cosette ma adesso non va .Cosa posso fare ?

bduplan, 2017-12-30:
I had the same problem. You need to shield all motor and switch cabling and
ground the shielding, only at one end, to the table. Then you need to
ground the table to a grounding rod. The controller needs to be in a
shielded enclosure that is electrically isolated from the table and
grounded with a separate wire to the ground rod. So its not technically
isolated, but it has to pass through the ground rod before they are
connected, so the interference is isolated. I removed the ground connection
to my home wiring so that there was only one path to ground.

nicolaux, 2017-12-30:
grazie mille  cercherò di fare queste cose che dici tu ,buon anno