grbl/grbl #1261: problem with y axis

samik648, 2017-07-14:
i am constructing cnc with arduino uno,a4988 motordriver,cnc shield v3.0 and PK264-03B STEPPER MOTOR.

problem is x axis is rotating good but y axis motor is holding torque while rotating.y axis motor is not rotating..
i will be highly obliged if u give me any solution..
                                                  THANK YOU

chamnit, 2017-07-14:
That's pretty vague and there are a lot of things that could be wrong. Check your wiring and if your stepper driver not setup with the right current or lots of friction in your y-axis. This is not a problem related to Grbl. Closing.

closed 2017-07-14 by chamnit

samik648, 2017-07-14:
when i connect x axis dir and step to each x and y axis motor driver and motor,both motor are moving.
but when i connect y axis dir and step to each x and y axis motor driver and motor,both motor are not moving

vMeph, 2017-07-14:
Check the wings of motor if they connected correctly

jringling, 2019-01-02:
Can I comment here? I have the same problem, no y axis movement. Swapped cables, driver boards, and verified motor is working. Grbl 1.1z on a bachinmaker nano board.

whitetd, 2019-01-02:
Didable end stops, does it work then?

jringling, 2019-01-02:
I can find the limit settings, but not "end stops". Which line is the y axis end stop?