grbl/grbl #1263: How to connect a CO2 laser power supply to arduino?

jef1992, 2017-07-19:
My CO2 Laser Power Supply have these connections:

H: Laser output control (high level effective)
L: Laser output control (low level effective)
P: Water protectio switch (valid if grouded)
G: Ground signal
IN: Laser energy control signal (0-5V)
5V: Output 5V/50mA

I don't understand the laser guide very well.

The PWM pin 11 goes to "IN" pin of the power supply?
And Spindle Direction pin 13 goes to "H" pin of the power supply?

Also, I have to connect the GND and 5V of the Arduino to the GND and 5V of the power supply?

My power supply is a MYJG-100W.
Image of connections of the power supply: https://images.ua.prom.st/824079939_w800_h640_connect.jpg

allstar87, 2017-08-01:
I have a similar project, and here's what I've found thusfar.  Note I don't have it working 100% yet, but this what I think I know:

The +5V pin on the power supply is a regulated 5V output.  It's there mostly if you're going to use a potentiometer to control laser power rather than the controller's PWM.  It can be left disconnected.

Spindle speed (PWM) needs to be hooked to IN.
Spindle Direction can be attached to either TH or TL.  I'm hooking mine to TH.  Gcode commands M3 and M4 toggle this pin LOW and HIGH respectively, in addition to changing between static and dynamic laser mode.  I want to run in dynamic mode (so that if it stutters due to buffering or whatever, it'll turn the laser off and not put a burn mark on a raster image).  The Spindle Direction pin is HIGH when in dynamic mode, so we'll use TH on the power supply.

I'm using LaserWeb, so I put an M4 in the tool on G-code, and M3 in the tool off G-code.  That way it puts an M4 at the beginning of a block of G1 operations and an M3 after.  It'll pull the spindle direction line LOW for G0 moves.

The cabinet I have has two interlock switches; I'm going to use one to interrupt the TH line when the door is open to force the laser tube to turn OFF, and the other to hook to the HOLD and RESUME lines to pause the gantry when the lid is opened, and resume the gantry when the lid is closed again--a little like my dryer.

brakthehun, 2017-08-22:
I've used a laser power supply (LPS) on this and it works great.
Not all laser power supplies are the same.
LPS 5v - don't connect
LPS gnd - don't connect
LPS PWM(IN) - don't connect
LPS TH - Pin 11 Arduino
LPS TL - GND Arduino
LPS P - lid switch or ground.

Be very careful. I've seen the laser fire while the arduino is booting. Make sure to connect a 10K resistor from Arduino ground to LPS TH. The arduino boots with all pins as inputs and pin 11 can float up enough to activate the laser.

00alkskodi00, 2018-06-13:
Hi 

i am novice

can that working?

[image]

Takton, 2018-12-05:
I use arduino uno and CNC SHIELD.
GRBL 1.1 and the program Laserweb4.
I'm trying to connect the power supply, but I can not.
Help solve this problem please

[image]

wili18, 2019-02-09:
Hallo everyone, please explain me. From arduino is only 1 signal output (pwm from Z +) used to control co2 power supply. This is used to control power at IN input. Next is need to activate the laser fire to the L input by connecting it to the GND. How do I do it? From the arduin there is no further signal (I try spindle enabl.,spindl dir. ...) for it. If I connect it like Takton(up) it fire with full power.
 Is there any next output what is connected (or log1) when laser is fire? Or some add switch board?
Thank you

AdamMulica, 2019-03-26:
> I use arduino uno and CNC SHIELD.
> GRBL 1.1 and the program Laserweb4.
> I'm trying to connect the power supply, but I can not.
> Help solve this problem please
> 

Hello,
Have you found the solution?
Regards

Zulfikar-stark, 2019-06-19:
hi. please help me
Power supply recomended for laser tube Co2 40w. and wiring diagram for arduino without cnc shield but with TB6600 driver stepper motor. Thank's

AdamMulica, 2019-06-19:
Zulfikar-stark search on the google "arduino with connect TB6600"
I buyed in aliexpress: 40W CO2 Laser Power Supply MYJG-40T 110V 220V for CO2 Laser Engraving Cutting Machine 35-50W

Arduino PIN 11 and GND - control with program LasderGRBL- power supply - connect LOOK my photo ;) power strip P2 and pin IN to pin 11 on arduino and pin G to GND on arduino :] or connect Analog Signal Control P2 - L G IN

Remember , make a bridge power supply P2 - G and P together and G and L together, or make switches
on/off. If you do not do it, you will not run the laser from the program, eg LaserGRBL

Zulfikar-stark, 2019-06-20:
thank's AdamMulica.
what about using mach3 as control ?
what pin is used in mach3 for control laser ?

for driver motor stepper not problem, i have tried with arduino and tb6600.

enrique812, 2019-06-21:
> > I use arduino uno and CNC SHIELD.
> > GRBL 1.1 and the program Laserweb4.
> > I'm trying to connect the power supply, but I can not.
> > Help solve this problem please
> 
> Hello
> Here is the solution. Look photo.
> I am using the LaserGrbl program. Command M03 S1000 - 5V full laser power. M03 S0 - 0V laser off.
> Regards
> 
> [image]

Buenos días amigo Adam, aunque no fue mi pregunta te agradezco esta respuesta ya que me ha ayudado bastante, pero tengo unas preguntas mas.
Quiero controlar un láser de CO2 de 45w con Arduino Uno y CNC SHIELD, para lo cual quiero usar una fuente MYJG-50 (dejare el link de donde la he comprado y la foto al final) que según veo es ligeramente distinta a la que has indicado
1. Puedo hacer la misma coneccion para la fuente que he comprado?
2. Para que son esos puentes que haces entre los pines TH con TL  y de TH con WP?
3.  En la coneccion de la imagen la CNC shield controlara la potencia y el encendido y el apagado del tubo láser? 
de antemano muchas gracias por tu ayuda
[image]

https://es.aliexpress.com/item/32791184674.html?spm=a2g0s.9042311.0.0.274263c0ds14eq

enrique812, 2019-06-21:
> Hello
> Here is the solution. Look photo.
> I am using the LaserGrbl program. Command M03 S1000 - 5V full laser power. M03 S0 - 0V laser off.
> Regards
>
> [image]

Good morning friend Adam, although I did not have my question, I thank you for this answer and you have helped me a lot, but I have a few more questions. 
I want to control a 45w CO2 laser with Arduino Uno and CNC SHIELD, for which I want to use a MYJG-50 source (on the link where it is understood and in the photo at the end) indicated

1. Can I make the same connection for the font I bought?
2. What are those bridges you make between TH pins with TL and TH pins with WP?
3. In the connection of the image will the CNC shield control the power and on and off of the laser tube?
thank you very much for your help
[image]

https://es.aliexpress.com/item/32791184674.html?spm=a2g0s.9042311.0.0.274263c0ds14eq

AdamMulica, 2019-06-21:
I dont know i dont use mach3

enrique812, 2019-06-21:
> Hello
> Here is the solution. Look photo.
> I am using the LaserGrbl program. Command M03 S1000 - 5V full laser power. M03 S0 - 0V laser off.
> Regards
>
> [image]

Good morning friend AdamMulica, although I did not have my question, I thank you for this answer and you have helped me a lot, but I have a few more questions.
I want to control a 45w CO2 laser with Arduino Uno and CNC SHIELD, for which I want to use a MYJG-50 source (on the link where it is understood and in the photo at the end) indicated

1. Can I make the same connection for the font I bought?
2. What are those bridges you make between TH pins with TL and TH pins with WP?
3. In the connection of the image will the CNC shield control the power and on and off of the laser tube?
thank you very much for your help

[image]

https://es.aliexpress.com/item/32791184674.html?spm=a2g0s.9042311.0.0.274263c0rIusH6

AdamMulica, 2019-06-21:
Hello,
You must connect, in power supply G and P - bridge! and G and L - if L = low power LASER ON - bridge

Connect with arduino:
Power supply IN - to arduino shield Z+, and G to Z- look my photo

enrique812, 2019-06-22:
muchas gracias por tu respuesta amigo AdamMulica, de verdad te estoy muy agradecido, pero seria mucha molestia si te pido que por favor me indiques para que sirven los pines:
INPUT CONTROL SIGNAL (IN) , SIGNAL GROUND (G), WATER PROTECTION (P), L y H?, he buscado en internet y solo dice su nombre y una leve referencia de lo que son, pero no especifica cual es su función o para que sirven; se que el pin 5v se usa solo si harás uso de un potenciometro, por eso no pregunto por el.
En tu respuesta me estas diciendo que:
1. Haga un puente entre G (signal ground) y P (water protection)?
2. Ademas que haga otro puente entre G (signal ground) y L  para que el láser tenga baja potencia?
3. Ademas que conecte el pin IN(input control signal) de la fuente al pin Z+ de la cnc shield?
4. ademas que conecte el pin G (signal ground) al pin Z- de la cnc shield?
Un ultimo favor amigo, me puedes indicar las conexiones en la imagen que te estoy enviando?

[image]

Disculpa mi ignorancia amigo es que me he aventurado a hacer esto sin tener mucho conocimiento del tema.
De antemano muchísimas gracias por tu ayuda, espero me tengas paciencia.

AdamMulica, 2019-06-22:
[image]

I think so
I'm not sure because I have a different power supply
1 method
but if L and G = 0V then the laser will be ON or
2 method
If H and 5V = 5V then the laser will be ON
or you must have these two methods turned on together
Do it by trial and error

Z+ AND Z- = you control voltage from 0V to 5V from the LaserGRBL program

[image]

enrique812, 2019-06-23:
amigo AdamMulica discúlpame por ser un poco distraído; ahora veo que la fuente que tienes es distinta a la que yo uso y recién me he dado cuenta que tu pones un puente entre los pines G y P ,ademas pones otro puente entre los pines  G y L, ademas conectas G con Z- y IN con Z+.
tengo una ultima duda.
en la imagen que me has enviado hay una nota que dice "security  water circulation in the laser tube if there is no bridge, the laser will be turned off"
esa nota quiere decir que el puente entre G y P sirve para que si no hay agua enfriando el tubo láser, este tubo se apagara?
o quiere decir otra cosa?
de antemano te agradezco tu ayuda amigo AdamMulica

AdamMulica, 2019-06-23:
[image]

This is my power supply and how it is connected
If I do not have a bridge between G and P, the laser tube does not work
The point is that it's best to give a button there ON/OFF

[image]

1 button - connect to G and P
2 button - connect to the water pump

Then you turn on the 2 buttons and you are sure that you have the water pump turned on and the protection in the power supply

enrique812, 2019-06-23:
muchas gracias por tu ayuda amigo AdamMulica, sin tus consejos no hubiera podido entender como se conecta la fuente y el láser, estoy pensando en instalar en serie un termostato relé programable y un sensor de agua entre los pines G y P, para que si no hay flujo de agua no funcione, pero también si la temperatura del agua sobrepasa los 30 grados se apague, todo esto como doble medida de seguridad.
El interruptor de apagado y encendido lo colocare entre los pines G y L.

Y para finalizar dos preguntas mas:

1. ¿Se podrá adaptar una pantalla táctil compatible con arduino uno para controlar la potencia máxima del láser desde esa pantalla sin necesidad de ir hasta una computadora?

2. ¿te ha pasado que si enciendes la fuente antes de conectarte con el programa, el láser se enciende?, si esto te ha pasado ¿como lo solucionaste?

AdamMulica, 2019-06-24:
The problem I noticed is:
If I do not have a USB cable connected between the computer and arduino, the laser sends a tiny beam all the time. This is probably a GND issue but I have not solved this problem. You probably need to fumble at SHIELD V3.
As for the touch panel, I do not know, I do not use.
It's best to test yourself what to do and how to fix the problem.

Good luck in device development :)

enrique812, 2019-06-25:
otra ves gracias por tu respuesta amigo Adam Mulica.
A propósito de GRBL láser, he encontrado esto: https://github.com/gnea/grbl/blob/332acada8854538443d625c84b5346a34cabbf6a/doc/markdown/laser_mode.md
El GRBL láser que me has dicho que use es el mismo que hablan en esta publicación?

AdamMulica, 2019-06-25:
Hello,
First install driver ARDUINO
https://www.arduino.cc/en/Main/Software - Download the Arduino IDE

connect usb to arduino and:
You must installing a file named grbl_v1.1f.20170801.hex
to arduino 
https://github.com/gnea/grbl/releases - download file .hex
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino - installing .hex to arduino

The LaserGRBL program 
http://lasergrbl.com/en/download/ 
is for firing and engraving and it connects with arduino

enrique812, 2019-06-26:
muchas gracias amigo AdamMulica ya instale el grbl laser y estoy empezando con lo del grbl del arduino, pero mientras pensaba como comenzar, me vino una duda con respecto a los drivers que tengo que usar, mi duda esta en que tengo motores  Nema 17  1.2A  7.2V  los cuales no se que drivers ponerle.
Tengo drivers TB6600  4.5A 8-50V  y  drivers  DRV8825  1.5A  8.2-45V, pero como te abras dado cuenta, ninguno de estos dos drivers esta en el rango de voltaje de mis motores.
Tu crees que pueda usar alguno de esos 2 tipos de drivers o mejor seria que compre otros drivers que sean del voltaje de los motores que tengo (7.2v)?

AdamMulica, 2019-06-27:
Hello,
I used DRV8825 and i have Nema 17 1.7A (17HS8401) and everything it works good :)
pay attention to A and not to V
you can use TB6600 and DRV8825 :] what do you want

enrique812, 2019-06-28:
que bueno amigo AdamMulica gracias a ti me ahorre la molestia de comprar drivers nuevos.
Dime ¿hay alguna tarjeta alternativa a arduino uno y cnc shield que se pueda usar con grbl láser y los drivers DRV8825 a la cual se le pueda colocar una pantalla táctil?

AdamMulica, 2019-07-01:
As for the touch screen, I have no idea you need to look for answers on your own

Zulfikar-stark, 2019-09-11:
I have a problem with the accuracy of cutting acrylic using a co2 laser tube. how to set up a gcode file and sender what is compatible ?
help me please. thank's

AdamMulica, 2019-09-11:
I do not know what you have a problem with?

If you use the LaserGRBL program you must enter the arduino configuration and parameters:
$100, $101 $102 - set the appropriate values ​​- if you send the command take 50 steps then it should move to 5cm. Length the ruler and set the appropriate parameter

distebia, 2019-10-10:
Salve, ho letto la discussione ma non trovo un aiuto per il mio caso. mi spiego meglio: attualmente utilizzo arduino uno per pilotare un laser a diodo da 10w tramite pin 11 e gnd (PWM) e tutto funziona a meraviglia tramite Lasergrbl e GRBL 1.1f. Vorrei installare un laser co2 e non riesco a capire come da arduino uno uno vado a gestire l'alimentatore del laser. Qualcuno mi potrebbe aiutare anche con uno schema come posso fare? Grazie a tutti

AdamMulica, 2019-10-12:
> Salve, ho letto la discussione ma non trovo un aiuto per il mio caso. mi spiego meglio: attualmente utilizzo arduino uno per pilotare un laser a diodo da 10w tramite pin 11 e gnd (PWM) e tutto [...]

Hello,
Read my posts and you'll find the answer.

distebia, 2019-10-12:
Ho letto ma si parla di SHIELD V3 e DRV8825 soltanto

AdamMulica, 2019-10-12:
Arduino uno: 
PWM 0-5V - pin 11 = Z+
GND = Z-
;)

You need to buy a power supply for the laser tube co2 and connect in the same way as this diode laser - read the posts that are earlier carefully

Everything is described in the posts above

distebia, 2019-10-12:
il laser a DIODo lo controllo in pwm tramite arduino uno, quindi pin11 e gnd sulla scheda arduino, non ho  SHIELD V3. il problema è collegare arduino uno all'alimentatore del laser co2.
[image]

distebia, 2019-10-12:
[image]
come collego il mio arduino a questo alimentatore?

AdamMulica, 2019-10-13:
But what do you not understand? writes one last time. It doesn't matter if you don't have V3 shield. Arduino uno: 
PIN 11 - to the power supply for PIN IN
GND to the power supply for G
and must connect P to G in power supply and H or L to G - because the power supply does not start. These are safeguards. I've described it before.
FG to tube laser on MINUS - i thing so?

This is described in the last posts. If you do not read with understanding you will not run a co2 laser.

distebia, 2019-10-13:
ok, pin11 arduino a pin IN alimentatore, gnd arduino a G alimentatore, ma se voglio sfruttare la possibilità di avere un tasto "test a bassa potenza" sul pannello cosa devo collegare? Fungo di emergenza? Potenziometro? Per l'amperometro suppongo in serie alla linea di alimentazione del tubo.

AdamMulica, 2019-10-13:
For me it is like this: if I want to use the test button I have to disconnect the IN pin, I read somewhere that you need to disconnect the whole cube where there are pins H L P G IN 5V etc.
TRY different configurations.

distebia, 2019-10-13:
ma, escludendo arduino, come dovrei collegare il tutto per avere un pannello classico? sull'alimentatore potenziometro su uscita 5V per variare la tensione su IN? e test bassa potenza?

AdamMulica, 2019-10-13:
I am sorry but I will not do an electronic diagram for you, what it should be connected to. When building such a machine you must have an idea about electronics.

AdamMulica, 2019-10-21:
Zulfikar-stark - i have this power supply
[image]
If you have the same power supply then connect L- to the laser tube at minus and AC. 
The other pins do not connect - then you can use the TEST button

distebia, 2019-10-21:
> [image]
> 
> Penso quindi
> non sono sicuro perché ho un
> metodo di alimentazione 1 diverso
> ma se L e G = 0 V allora il laser sarà acceso o
> 2 metodo
> Se H e 5 V = 5 V allora il laser sarà acceso
> o è necessario disporre di questi due metodi attivati ​​insieme
> Fallo per tentativi ed errori
> 
> SEGNALE DI CONTROLLO INGRESSO (IN) - potenza laser dallo 0% al 100% di controllo dal programma!
> G (GND) e L - se non è acceso, il laser non funzionerà (non è potenza laser) - protezione
> o
> 5V e H - se non è acceso, il laser non funzionerà (non è potenza laser) - protezione
> G ( GND) e P - se non è acceso, il laser non funzionerà - sicurezza
> DEVI ACCENDERE due sicurezza
> 
> Z + AND Z- = si controlla la tensione da 0 V a 5 V dal programma LaserGRBL
> 
> [image]

Rieccomi, scusa la mia domanda ma vorrei capire meglio i collegamenti fatti da te nel connettore P2:
5V e H switch on/off or bridge (a che serve?)
Cosa comandano H e L?
L e G switch bridge (a che serve?)
P e G  switch on/off or bridge (serve a comunicare che la pompa dell'acqua è accesa?)
dopo, sempre se ho capito bene, hai collegato un potenziometro fra 5V/In/G per variare la tensione su In e regolare la potenza del laser, vero? 
In tutto questo (potenziometro) come colleghi il pin 11/Z+ (pwm) e gnd/Z- di arduino?
Anche io utilizzo Lasergrbl, grbl 1.1f, arduino uno, 3 tb6600 e un modulo laser 450nm 10W comandato in pwm e mi trovo benissimo.
Grazie

opserc20, 2019-11-16:
Guys, is there any clean tutorial to this upgrade, i have the same power supply as Adam

Zulfikar-stark, 2019-11-17:
please help
my fuse is explode. i haved repalced it and fuse explode again. what needs to be fixed ?
(Explode before installing to the tube)
like this
https://www.youtube.com/watch?v=QKgn-3pHl0E

Zulfikar-stark, 2019-12-16:
AdamMulica
do you have solving to my problem ?

AdamMulica, 2019-12-16:
Zulfikar-stark - To help you I need to have your laser with me. I can't help at a distance. You have to take the meter in your hand and look for a problem.

distebia, 2020-02-01:
Salve, ho quasi completato lo scheda di collegamento del mio laser co2 comandato da arduino uno. Allego 2 schemi, uno utilizza soltanto il segnale pwm da arduino su pin IN e l'altro utilizza sia Arduino che il potenziometro. quale è migliore? eventualmente cosa dovrei cambiare? So che Arduino accende e spegne il laser mediante PWM ma se arduino manda il max (255) posso correggere (abbassare) la potenza mediante potenziometro? 
Dimenticavo...in entrambi gli schemi utilizzo un w1209 per avere sotto controllo la temperatura dell'acqua. è corretto collegarlo su WP oppure dovrei collegarlo per interrompere direttamente l'alimentazione 220VAC dell'alimentatore laser? Non so se regge il carico.
[image]
[image]

distebia, 2020-02-01:
Dimenticavo...nell'immagine sono inserite le vere foto di quanto vado a montare.

AdamMulica, 2020-02-03:
For manual control, potentiometer

[image]

If you want to control with arduino, I advise you to disconnect the potentiometer, you can also connect the arduino and the potentiometer together, but for what?

W1209 can be connected under WP and GND and after exceeding the maximum temperature switches off the power supply. W1209 must be set to opening contact

distebia, 2020-02-03:
Infatti il dubbio era proprio questo: se invio un segnale pwm da arduino il laser lo esegue ma, metti caso mi accorgo che la potenza è troppo forte (ne basta molto meno) e da lasergrbl non riesco ad abbassare, attraverso il potenziometro potrei abbassare la potenza, ora mi rendo conto che non serve, o uno o l'altro. Lo schema di collegamento come da immagine senza potenziometro va bene? cambieresti qualche cosa? Il W1209 è messo in serie con il sensore di flusso proprio per controllare in fase di avvio e in fase di lavorazione sia la presenza di flusso dell'acqua sia la temperatura dell'acqua nella vaschetta. A proposito mi sapresti indicare fino a che temperatura posso impostare il termostato?  
Perché ci si collega da arduino PIN11 a TH nell'alimentatore? Se guardi l'immagine senza potenziometro collego il Pin11 a IN e gnd a G.
Test pulsante TL e GND - hmmmm? Funzionerà se si imposta la tensione minima tramite il potenziometro. Se tagli, tieni premuto il pulsante tutto il tempo? Se guardi l'immagine senza potenziometro (la prima) ho collegato TH a 5V tramite pulsante (test laser)
distebia@gmail.com e grazie per l'aiuto

Alexandermolinagonzalez, 2020-02-19:
AdamMulica como vas , tengo una laser con tubo de 50 w y fuente ( power supply 50w- g) he intentado muchas conecciones y no logro configurar para que me funcione el sistema pwm para grabado de fotos o imagenes con escala de grises,  le tengo instalado un potenciometro y opte por comprar un modulo pwm mosfet: mi pregunta es si es necesario usar el modulo o no ? O si tienes algun esquema de coneccione y me puedas ayudar te agradezco

Alexandermolinagonzalez, 2020-02-19:
Mi correo   alexmg0310@gmail.com

johnny3001, 2020-03-08:
> [image]
> 
> This is my power supply and how it is connected
> If I do not have a bridge between G and P, the laser tube does not work
> The point is that it's to give a button there ON/OFF
> 
> [image]
> 
> 1 button - connect to G and P
> 2 button - connect to the water pump
> 
> Then you turn on the 2 buttons and you are sure that you have the water pump turned on and the protection in the power supply
> 
> or
> 
> buy
> W1209 Red LED Digital Thermostat Temperature Control Thermometer Thermo Controller Switch Module DC 12V Waterproof with Case Box on aliexpress and control pin P (water protection)
> 
> If you exceed the water temperature in the 30-degree tank, the relay in the device will take a break and the laser will not work
> 
> It depends on you, how want to protect your equipment!
> 
> I use laser from time to time, so i do not overheat water in the laser tube, but also check the temperature in the tank. If I use it all the time, I will mount the device
> W1209 Red temperature control of the LED thermostat

Hello,

I have my connections just lik yours, but my Y motors start to loose steps as I increase laser potency on laserweb.
Actually they start to act crazy.

Zulfikar-stark, 2020-05-16:
[image]

i have a problem like that. iam not used cnc shield.
P2 MYJG 40w wiring : G to P . L to P . IN to pin D11 . G to gnd arduino. What's wrong with that?

AdamMulica, 2020-05-20:
Hello. I have no idea. After the message you can see that there is a problem with arduino. Maybe you need to upload a new program or update the GRBL program because it is already 3.4

makerpolaroid, 2020-07-02:
Hey guys ! I read your comments and I m a little bit confused. I draw a « wiring diagram » of what I understood. Could you tell me if i understood it well ? 
I’m using a cnc shield and a my laser PSU is a MYJG-40MG
[Wiring+diagram.pdf](https://github.com/grbl/grbl/files/4865591/Wiring%2Bdiagram.pdf)

AdamMulica, 2020-07-02:
makerpolaroid - According to your diagram, almost everything is correct. But.
1. MYJG-40MG - On the back of the power supply you have a thick red wire that you connect to the laser tube.
2. Improved assembly diagram 

[image]

makerpolaroid, 2020-07-02:
Oh alright i see (i don’t receive the laser PSU yet that’s why i don’t see this red wire).
Thank you so much !!

makerpolaroid, 2020-07-02:
@AdamMulica Sorry about disturbing you one more time but is it possible to engrave materials this way ? I mean not only cut ? I saw on several forums it was all or nothing with the signal wire  (the pwm one)...

AdamMulica, 2020-07-02:
FG - it's nothing but ground (earth) - well you have it connected.
You can engrave but there is a problem. The laser is activated from 0.3V, i.e. there is a gap between 0 and 0.3V. This causes you to be limited in engraving. I mean, you can't make a very thin engraving. As for the inscriptions and those things that are constant, i.e. have a constant depth, you can engrave.

[image]
[image]

makerpolaroid, 2020-07-02:
Thanks for the wiring, I will link FG to my wall plug. 
I will also try to find a solution for the 0 to 0.3v gap... I ll let you know if I a suceed ! 
Do you cut sometimes at 5v or do you use a lower voltage pwm ?

[22 more replies]