grbl/grbl #1264: Candle_1.1.7 hightmap not probing well

migimantas, 2017-07-20:
Hello,
so I am having this issue I use grbl 1.1e with arduino mega 2560. For cnc milling I use Candle_1.1.7 and it has hightmap function for pcb milling. The problem is that when the probe touches plate machine stops and I get ERROR:9 
The probe itself works fine I guess, when doing regular probing I do not get errors. 
This is example video how it should work
[https://youtu.be/tK5Amg-Ye58](url)

And this is how it looks for me when the probe touches plate.
[image]
I do homing cycle but still the same error.
I disable homing still the same error.

chamnit, 2017-07-20:
Grbl is getting placed into an ALARM state for some reason. According to the wiki and other documentation, ALARM:3 is caused by a 'Reset while in motion'. Meaning your system is being reset for some reason. Maybe you have a reset button and its poorly isolated. error:9 is just the standard Grbl response for trying to command something in an ALARM state. It blocks all gcode.

migimantas, 2017-07-20:
Somehow I only get this reset error when I am probing. The probe touches plate, it stops and the system shows error:3 and resets. 
Probe setting is
$6=0
Probe circuit is
[image]
Also I have noticed that when I do not engage Z in a movement but just touche probe contacts system just resets and no error:3. 
[image]

chamnit, 2017-07-20:
Please provide your $$ settings and your $I build info string. Also which pin is your probe connected to?

migimantas, 2017-07-20:
I apologize for my stupidity. I have just noticed that  probe pin is MEGA2560 Analog Pin 15 and I have connected my probe pin to Analog Pin 11 and all the time I was shorting pin to GND and resting processor

closed 2017-07-21 by chamnit