grbl/grbl #1266: Y motor only moving 1 direction

scarypatrick, 2017-07-24:
Hi,
I'm building an cnc laser cutter with an arduino uno with an cnc shield v3. But I have a problem with the y motor direction. It only moves in one way while the x motor is moving both ways. When I just installed everything is worked, but after a view tries it doesn't anymore. I viewed multiple topics on this problem but nothing is working. I tried to switch the stepper drives but that didn't make a difference. Switched the stepper motors and then the x motor moved in one way so it isn't the stepper motor. Tried cloning the y controller to the 4th connection. Also didn't work. Tried multiple grbl hex files, all the same. I also tried multiple gcode senders but is just doesn't make a difference. Is there anything else that I can trie to help this problem? Or shall I order an new shield to test if this shield is broken.

Protoneer, 2017-07-24:
Its likely a dry joint or bad connector on the board. The manufacturers
dont spend a lot of time on quality checks at that prices they charge.

Elecrow.com sells the latest version of the boards for $15...

algo1980, 2017-07-24:
may be a crap hardware. try checking your connectors or replacing your motors.

vMeph, 2017-07-24:
what version grbl you using?

algo1980, 2017-07-24:
no version acts this way vmemph ive checked em all ;)

vMeph, 2017-07-24:
@algo1980 just was asking cause u saied you using the cnc shield v3 and that shield if you look at pinouts was made for GRBL 0.8 versions below, on new versions GRBL 0.9 and above the pins 11 and 12 on the arduino was swaped, eaven knowing that your problem might not be do to this cause pins that was swaped was spindle enable and limit Z axis,  just wanted to warn you that you might want to take a look on it, there is changes that can be done like soldering 2 wires on arduino bottom and then bend the pins on the shield and swap the pins

that if your shield is the V3 version, like this one
http://www.ebay.com/itm/V3-Engraver-3D-Printer-CNC-Shield-Expansion-Board-A4988-Driver-Arduino-/191736127011?epid=622498425&hash=item2ca45d1e23:g:hFUAAOSw1S9WciGu

you might want to take a look on GRBL wiki page 

about your problem i think that can look like a damage driver, you can test that by changing the driver that is working X and put it on the Y socket and put the Y that isnt working on the X to see if that might be the case

redFrik, 2017-08-02:
try with another arduino. i had the exact same issue today and after trying out different cables, controllers, grbls, etc i finally replaced the uno with another one (same model) and then it worked (i.e. y axis now move both back and forth).

scarypatrick, 2017-08-03:
Exactly. I replaced the arduino and everything is working now.

algo1980, 2017-08-03:
You could try changing pin connections from config.h

DjamesSuhanko, 2017-09-17:
Hello, everyone!
I'm facing the same problem related to Y axis direction in CNC shield v3. I changed drivers X with Y to test it. X is working with anyone; Y is not changing direction with none.
The other tip that I could see above is to use 0.8 version, but I couldn't found it ( @vMeph ). I  think it's the simplest and faster test that I can perform now. Does someone know where I can download the 0.8 version?
And, if this version does not work too, I will buy another Arduino UNO to test again.
To  finish, I saw above that I can change pin setup in config.h ( @algo1980 ) , but I saw no traces about how to do that. Can someone help me realize how to do that?
Thanks!

vMeph, 2017-09-17:
version grbl 0.8 https://github.com/grbl/grbl/tree/v0_8

LucianIonut, 2018-06-25:
I haved same problem with X and Y axis, in my case was a faulty Arduino Uno (Analog 1 to 5) wich doesn't change their state. I atach a simple diagnostic file for all pins, you onley must measure all of them and see if they ar changing the state (blink mode for 3 sec).
    Good Luck !
[Diagnostic_Arduino_UNO_PinOuts.zip](https://github.com/grbl/grbl/files/2134286/Diagnostic_Arduino_UNO_PinOuts.zip)

DjamesSuhanko, 2018-06-25:
Thanks, @LucianIonut !

LucianIonut, 2018-06-27:
I have updaate the .ino file, i meke it in hurry and something was wrong. Sorry !