grbl/grbl #1267: Check door

Davida54, 2017-07-25:
How do you defeat this when there is no door

109JB, 2017-07-25:
Did you enable the safety door in the config.h file? It is this line (Line 137 in the current version):

// #define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. Uncomment to enable.

If you don't have a safety door it should have the two slashes at the front of the line to disable it as above.

Davida54, 2017-07-29:
Thank you 109JB, I am new at this and am in need of some more help. I can not find the configure.h file you are referring me to. Every thing else is working as it should. Just when I send my own file projects I get this error  Check Door.

closed 2017-07-29 by Davida54

chamnit, 2017-07-29:
@Davida54 : What version of Grbl are you using?  If you are using v1.1, please provide your $I build info output. 

Check door should only be available if you or your CNC manufacturer has enabled it. The default is disabled.

GuiRitter, 2019-04-11:
Me too. Happened during a program, after executing a few hundred G1s successfully.

`$I` is `[0.9j.20160726:]`.

I'm not certain, but most likely `ENABLE_SAFETY_DOOR_INPUT_PIN` wasn't uncommented before downloading the code to Arduino.

Feed hold pin is floating.