grbl/grbl #1268: No arcs

pettuma, 2017-08-04:
Whenever G02/G03 is used the machine draws the line instead of expected arc. The target point is reached but the trajectory is not curve but linear. UNO with grbl 0.9, 320 steps/mm, various gcode senders. What do I do wrong? Thanks.

mayhem2408, 2017-08-04:
@pettuma Can you share some Gcode that is causing the problem so we are test it.

X3msnake, 2017-08-04:
> Hi John,
>
> ... thank you very much for interest in my problem. To be as simple as
> possible I do not use any gcode sender now. I communicate with Arduino
> only via pure ASCII terminal (Termite.exe, COM3, 115.200 kBd) and so I
> can only send one gcode at one time (no program) as a command. G00, G01
> work very well. G02 and G03 do not work well (they work like G01, i.e.
> machine draws line between actual and target points. It behaves like the
> radius is infinite).

have you tried IJK instead?

pettuma, 2017-08-04:
Yes I did, the same result.

chamnit, 2017-08-04:
Please post your Grbl $$ settings and your $I build info strings.

pettuma, 2017-08-04:
ok
$$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=6 (dir port invert mask:00000110)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=18 (status report mask:00010010)
$11=0.000 (junction deviation, mm)
$12=0.000 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=208 (homing dir invert mask:11010000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=320.000 (x, step/mm)
$101=320.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=2000.000 (x max rate, mm/min)
$111=1000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=30.000 (x accel, mm/sec^2)
$121=20.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=290.000 (x max travel, mm)
$131=90.000 (y max travel, mm)
$132=0.000 (z max travel, mm)
ok

$I
[0.9j.20160316:]
ok

109JB, 2017-08-04:
Arc tolerance and junction deviation are both set to 0.  Why?

pettuma, 2017-08-04:
It is default value. I changed it now: 
$12=0.5 
ok 
$11=0.5 
ok. 

I tested again:
g0 x0 y0            Machine set the position x0 y0 
g1 x50 y50        Machine draws the line from point X0 y0 to point x50 y50 
g2 x0 y0 r40     It WORKS !!! I can see the arc !!!

Thanks !!!

X3msnake, 2017-08-04:
you should drop the value to something like 0.01 mm

109JB, 2017-08-04:
The generic defaults are as follows:
#define DEFAULT_JUNCTION_DEVIATION 0.01 // mm
#define DEFAULT_ARC_TOLERANCE 0.002 // mm

chamnit, 2017-08-04:
Default settings for $11=0.01 and $12=0.002

closed 2017-08-04 by chamnit

pettuma, 2017-08-04:
OK, I will tune it. Thank you a lot for your advice!