grbl/grbl #1269: Timeout Serial Grbl v0.9j da Arduino IDE

Pante90, 2017-08-06:
Good morning,
I would like to ask why, if I load through the Xloader the arduino grbl v0.9j.hex file works correctly in everything; If I instead compile the version 0.9j socket with Arduiono IDE by changing the variable_spindle value from 1000 to 255, once loaded into the universal gcode sender program, it crashes to the connection and returns the error "grbl v0.9 [$ Help] time out "and I can not connect it. why?

chamnit, 2017-08-06:
There is no specific reason that Grbl would lock up by compiling via the IDE, since the hex is also built by the IDE tools. Try connecting to Grbl through a serial terminal to see what Grbl is returning to UGS.

Pante90, 2017-08-06:
It also gave the same problem to the serial monitor via arduino IDE. Anyway I managed to fix, loading with the Xloader the .hex file created by the arduino ID. Now works. Thank's you so much...

closed 2017-08-06 by chamnit