grbl/grbl #1271: Homing without z-axis

pettuma, 2017-08-09:
Temporarily I have only xy machine with limit switches. I connected one manual switch to simulate Z limit switch and started the standard homing cycle $H. The homing cycle somehow goes until I trigger the manual swith. After this immediately "ALARM: Homing fail" appears. 
 
I start with this grbl state:  <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000,Lim:000>
Setting:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=6 (dir port invert mask:00000110)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=19 (status report mask:00010011)
$11=0.010 (junction deviation, mm)
$12=0.010 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=1000.000 (homing feed, mm/min)
$25=200.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=320.000 (x, step/mm)
$101=320.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=2000.000 (x max rate, mm/min)
$111=2000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=30.000 (x accel, mm/sec^2)
$121=30.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=290.000 (x max travel, mm)
$131=90.000 (y max travel, mm)
$132=100.000 (z max travel, mm)
Version: [0.9j.20160316:]
Is there any advice why the homing cycle cannot be completed? Thanks.

lalo-uy, 2017-08-09:
In the homing sequence the switch goes on, then the direction is reversed
and goes off, then on again slowly and finally off.
You will have to mimic all this steps to avoid the ALARM.

Much better comment off the Z homing cycle in config.h and recompile.

109JB, 2017-08-09:
Why not just turn off Z axis homing in the config.h file?

You should be able to change these 2 lines

#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.

to:

//#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.

Once done the machine will home only on x and y

pettuma, 2017-08-09:
Ok, I will do it this way. I wanted to test it as I prefer to work with 
wires.

Dne 09.08.2017 v 16:34 109JB napsal(a):

closed 2017-08-09 by chamnit