grbl/grbl #1275: Extending GRBL with digital and analog inputs

dastultz, 2017-08-18:
Hello,

I'm seriously spinning my wheels here and hoping I can get some help. I am trying to customize GRBL for a Sanguino/644p with a number of buttons, lights, and a potentiometer. (I am replacing my home-grown motion controller with GRBL, all this is for jog/override control.)

I'm having a very odd problem with some of my code. I have added a call to my code from the main protocol loop, simplified here to focus on the problem:

```
void loop() {
  input_service();
  
  if (input_edgeHigh(6)) {
    println(input_edgeHigh(6), 8);
  }

  uint8_t high = input_edgeHigh(6);
  if (high) {
    println(high, 8);
  }
}
```

The `input_service()` function reads digital inputs and one analog. The digital input states are stored in a byte array. I track the current and previous states so I can respond to rising and falling edges.

The function `input_edgeHigh` returns a non-zero value when digital input 6 goes high. The 2 "if" blocks have different behavior. Both the "if" blocks print out the value returned by the `input_edgeHigh` function call. The first "if" block prints "1" the first time but then prints "0" a bunch of times (so long as the pin stays high) (`println` writes to serial as binary string). The second "if" block prints "1" just once. This seems terribly wrong (the first block printing "0"), since the block shouldn't be entered if the function returns zero. The second "if" block behaves as expected, somehow assigning the result to the local variable cures the woes.

If you're still with me, here's the relevant code, simplified. I store the current and previous state of the pins in the right-most 2 bits of each entry in buttonStates[].

```
#define DIN_STATE_CURR 0b001
#define DIN_STATE_PREV 0b010

uint8_t buttonStates[] = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };

uint8_t input_edgeHigh(uint8_t dinIndex) {
  return (buttonStates[dinIndex] & DIN_STATE_CURR)
      && !(buttonStates[dinIndex] & DIN_STATE_PREV);
}
```

Interestingly, I've got an `input_edgeLow` function that is working just fine...

My bigger problem, since I don't have a work around, is that reading an analog input always returns 0. I've tested all of this logic in an Arduino sketch (not using Arduino.h stuff except Serial) and it all works flawlessly.

I've gone through the GRBL code and commented out everything related to IO except steppers with no change in behavior.

Can anybody explain what is going on, or give me some way to read an analog input? Is there something an Arduino sketch does that GRBL does not or something GRBL is doing the Arduino sketch does not?

Thanks.

/Daryl

dastultz, 2017-08-20:
Hello, I have solved both the logic problem and the analog read problem though I don't really understand what's going on.

`analogRead()` implemented in `wiring_analog.c` makes a call to `delay(1)`. This is problematic in some way. Putting `delay(1)` directly in my code fails to compile. Something about `vector_18` in `stepper.c`.

The function `input_edgeHigh` is declared to return `uint8_t` but it actually returns more than 8 bits. In a simpler scenario, say comparing 2 `uint8_t` values of 3 and 3, the return value is actually 256. I have no explanation for the 9th bit but I know how to work around it. This behavior does not appear in C++ if I recall my troubleshooting correctly.

I'm leaving this open hoping someone can educate me. Go ahead and close it if you feel it's off topic.

Thanks.

csdexter, 2017-08-20:
The Arduino API and Grbl don't mix. Really, save yourself the trouble and don't even try it.

The Arduino `analogRead()` uses a convoluted (and very badly designed, IMHO) way to read the ADC without actually using interrupts (i.e. the way it's supposed to be done). It makes assumptions about when the conversion starts and how long it's going to take and it hides all that by calling `delay()` to stall the CPU while the ADC does its thing.
The Arduino `delay()`, in turn, also makes assumptions about what timers are or are not available, which assumption is rendered void by the fact that Grbl *does* use timers without any regard or allowance for the Arduino API (since they're not supposed to coexist). This is what's generating the vector_18() error.

The proper way to read an analog input from Grbl would be something like this:
1) Write an ISR for the ADC, as simple and as short as you can: basically just read the converted value off the ADC and store it in a global volatile variable. Along with the value, also set a boolean flag that indicates the value was [just] updated.
2) In the generic initialization code, configure the ADC and enable its "conversion complete" interrupt.
3) At the call site for Arduino's `analogRead()`, set the boolean flag to false and start the ADC conversion.
4) Further down, when you'd actually use the value, insert a check for the flag: if it's true, the stored value is available. If not, the ADC is not done yet and you should "come back later". Depending on how your code is designed, this may literally mean "try again later, on a future iteration" or busy-loop. Bear in mind that Grbl runs in real time and thus busy-looping for any amount of time may cause it to lose performance.

chamnit, 2017-08-20:
Grbl does not use any arduino code and will break if you do. It just uses the arduino IDE, as a vehicle to easily compile and flash Grbl to Arduino compatible hardware.

closed 2017-08-20 by chamnit

dastultz, 2017-08-20:
Yes, I was aware of the incompatibility with Arduino. The path I described helped me figure it all out, but I didn't intend to keep using Arduino libraries or fault GRBL for the problem. GRBL is great. I share my findings in the event someone else finds it useful.

Thanks for the explanation and a great project.