grbl/grbl #1276: GRBL 1.1 is biggetst shit on tne world

eddie99, 2017-08-19:
You DEV has completely failed on this version 0.9 was working as excepted  but v1.1 is a mess!!!!!

jahnj0584, 2017-08-19:
Want to give an actual reason or are you just mad you can't get it set up

tklus, 2017-08-19:
Good luck getting help with that attitude...

V1.1 has many big improvements abd does have differences than 0.9x...

109JB, 2017-08-20:
If you expect V1.1 to perform like 0.9 then you obviously didn't read the WIKI.  Why not try that first!!

chamnit, 2017-08-20:
This guy likely didn't realize that the communication protocol changed due to having a lot more realtime data and that it requires a GUI that supports it. Either way, this topic is clearly designed to be inflammatory without any respect for others. No reason to continue this. Closing and blocking.

closed 2017-08-20 by chamnit