grbl/grbl #1277: Negative Coordinate Problem

bashkan, 2017-08-21:
I have compiled GRBL 0.9 for my 2 axis machine with the following options:
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)) -> for 2 axis
 #define HOMING_FORCE_SET_ORIGIN -> so that when machine homes, it is at 0,0 (and in theory the coordinate system should become positive but it doesn't)
//#define HOMING_INIT_LOCK -> commented this one out so that the machine will not be in alarm state when it boots up.

The homing works perfectly and moves the tool to lower left corner. However after homing, both X and Y coordinates expect negative values. For example, If I send X10 Y10, it hits the limit switches. If I send X-10 Y-10 then the tool moves.
I have over 200 g-code files which all consist of positive values. Since converting them is not an option (I do not know if it is even possible) for me, I am stuck.  I have read the guide (https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9) several times but could not find an answer.
Can you please help me?
Koray.

langwadt, 2017-08-21:
sounds like you have directions inverted ($3), afaiu to get the homing you want you need to use ($23)

and you shouldn't need to change gcode just because it is positive values regardless of how you home,
you just set the work coordinates to match what you need

bashkan, 2017-08-22:
Dear langwadt,
Thank you for your reply. I am afraid I do not fully understand what you mean by just setting the work coordinates. I am a newbie :)
For reference here is my settings:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=100.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=400.000 (x, step/mm)
$101=400.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=100.000 (x max rate, mm/min)
$111=100.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=450.000 (x max travel, mm)
$131=450.000 (y max travel, mm)
$132=450.000 (z max travel, mm)

langwadt, 2017-08-28:
here's a quick intro from "this old tony", the meat of it at around 3:30

bashkan, 2017-09-03:
Dear langwadt, 
I am guessing you intended to send a link (probably to a tutorial video) but somehow forgot to put the actual link? If that is the case, can you please kindly post again? I am desperately in need of help.
Best Regards.

jahnj0584, 2017-09-03:
If you use the jog buttons does it move correctly?

https://youtu.be/BiuD6li6h8A

bashkan, 2017-09-04:
Dear jahnj0584,
I am using Universal Gcode Sender. After homing, when I click on for example X+, the tool hits the limit switch and goes into alarm state. I home again and click on X-, then the tool moves. I am not sure if this means 'correct move'. But it is definitely not the move I desire. Because after homing, any g-code file I send has positive values in it.
Koray

jahnj0584, 2017-09-04:
If you're homing in the lower left, X+ is to the right and Y+ is away from
you. Sounds like you have the directions inverted. I believe it's $3 you
edit to fix it. Check the wiki

bashkan, 2017-09-05:
@jahnj0584 
I will try changing $3 and will post results. Currently homing is lower left and as far as I remember X+ is to the left and Y+ is toward me (hence hitting the limit switches). Can this be related to my erroneously connecting the 4 cables from the steppers in opposite direction on the cnc shield? (i.e. should be 4-2-3-1 instead of 1-2-3-4)

chamnit, 2017-09-05:
Please read this [wiki page](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#quick-guide-to-setting-up-your-machine-for-the-first-time) on setting up your machine. You have your coordinate system backwards.

closed 2017-09-05 by chamnit

bashkan, 2017-09-09:
Setting $3=7 and $23=7 solved the problem. Thank you all very much for your help, it is appreciated.