grbl/grbl #1278: Stepper motors fail to change direction in time

Daumaz, 2017-08-27:
Hello everyone, I am trying to build my first cnc machine and I am facing this issue. My GRBL controlled cnc would sometimes fail to change the stepper direction in time for next set of steps, so it starts moving the wrong way and then quickly switches to going the right way again offsetting the whole operation from time to time. I noticed that it struggles to switch between right and left (X axis) and between back to forward (Y axis). I do not have much information on the Z axis since it is not moving a lot and is more difficult to observe (haven’t had the machine to slam in to a work piece or move up unintentionally, it could be fine). The X and Y axis are operated by identical Nema 17 0.6Nm 1.5A motors and the Z axis uses Nema 17 0.45Nm 1.7A. I use DRV8825 stepper drivers running on 24V pc power supply. The failures occur on more difficult paths (did not have any problems with easels basic shapes) and also while jogging, during the time cnc moves the wrong way it seems to be going a bit faster. Here is a video of one of operations: https://www.youtube.com/watch?v=cRGsGR6I628&feature=youtu.be

Things I tried:
Reducing the feed rate
Shielding high current cables
Moving the drivers back to breadboard (I was using my homemade board)
Inverting direction pins
Inverting step pins
Increasing step idle delay ($1) that or the feed rate might have reduced the offsets, increasing this value further has no effect
Sending +5V to all endswitch pins
Using external power supply for the Arduino
Adding pull down resistors on the direction pins
Reducing steps/mm of X and Y (that fixed a different problem I was having)	
Using different Arduino board
Changing the signal wires (when moving them I am not able to make the problem happen)

I am out of ideas, could not find anything more on the google, probably did not mention all of the things I tried. Anyone have ideas? Help would be very appreciated.

X3msnake, 2017-08-27:
> v=cRGsGR6I628&feature=youtu.be

Please also post your full GRBL definitions.

X3msnake, 2017-08-27:
Looks like you are losing steps.

What microstepping setting are you using on your drivers?

Daumaz, 2017-08-27:
I really do not think that it is losing steps, if I were to add the wrong way drawn part of the path in reverse and say continue from there it makes up the correct shape. I do not use microstepping, my GRBL settings:
$0=10
$1=50
$2=0
$3=0
$4=0
$5=0
$6=0
$10=115
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=350.000
$111=350.000
$112=450.000
$120=10.000
$121=10.000
$122=100.000
$130=200.000
$131=200.000
$132=200.000
$0=10
$1=50
$2=0
$3=0
$4=0
$5=0
$6=0
$10=115
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$110=350.000
$111=350.000
$112=450.000
$120=10.000
$121=10.000
$122=100.000
$130=200.000
$131=200.000
$132=200.000

jahnj0584, 2017-08-27:
It's backlash

Daumaz, 2017-08-27:
At first I thought so too, but then I saw it moving up to 10mm, then you can really hear and see the motors moving it

vMeph, 2017-08-27:
try keep motors always enable to see 
$1 = 255    (Step idle delay, milliseconds)

Daumaz, 2017-08-27:
Tried that, no change

X3msnake, 2017-08-27:
Can you post your gcode

What are you using to post the code ?

Daumaz, 2017-08-27:
I am using Easel. The problem also happens while jogging, do not think code has anything to do with it.
[TheCode.txt](https://github.com/grbl/grbl/files/1254418/TheCode.txt)

X3msnake, 2017-08-27:
Changing the signal wires (when moving them I am not able to make the
problem happen)

What do you mean here?

Daumaz, 2017-08-27:
I moved the step and direction signal wires (going from Arduino to the drivers) to maybe make them lose contact, but the cnc kept going the right way

X3msnake, 2017-08-27:
Can you pinpoint the lines that this happens?

Have you tried using other sender?

Daumaz, 2017-08-27:
Its always different, no, I have not used any other senders

X3msnake, 2017-08-27:
Just to exclude the sender can you try the same code with a different one
like ugcs

Daumaz, 2017-08-27:
I have UGCS, will try it.

Daumaz, 2017-08-27:
Just ran the same code trough UGCS, the problem is still there

Norwegian-Gadgetman, 2017-08-27:
Have you verified that the cause isn't mechanical?
(Tighten the setscrews holding the pulleys on the driveshafts. Maybe even filing a flat spot on the driveshaft for the screw to bite into)
Or that the steppers get enough power(Amps) to fight the inertia?

Daumaz, 2017-08-27:
Yes I have, as I mentioned before, it would sometimes jump quiet a bit further, then I can clearly hear the motors reversing quickly. I have set the current limit to mach my motors.

Daumaz, 2017-08-27:
Update: I disconnected the stepper wiring shield and lifted the cables of of my metal work table surface. Now the X axis jumping randomly both directions whenever it is powered on, also it reacts to my touch (on the shielding). Might be on to something here...

X3msnake, 2017-08-27:
Can you take some pics of your electronics

electrokean, 2017-08-27:
Are you sure you've got a common ground between all the components?

Daumaz, 2017-08-27:
Pics coming, I have got Arduino and all all of drivers grounds connected

jahnj0584, 2017-08-27:
Electrical is fine. This looks and sounds like backlash to me. Check your
couplers. Put a dial indicator on and jog back and forth a few dozen times
and see where it returns

Daumaz, 2017-08-27:
Here is what is left my wiring
[image]
Arduino (originally UNO)
[image]
The drivers (taken back the breadboard)
[image]
My original board, now I am only passing 24V trough it because it has the capacitors on it
[image]
Power boosters 12V to 24V, 2 in parallel for more current
[image]
Z axis motor wires and what used to be spindle power (now disabled)

Daumaz, 2017-08-27:
I really do not think it is backlash, its just going the wrong way
[image]

jahnj0584, 2017-08-27:
It looks like you are using hot glue as a coupler lol. What ballnut do you
have?

Wiring either works or it doesn't. Lead screws have lots of backlash with
regular nuts. Openbuilds sells a good anti backlash one you should invest
in.

Daumaz, 2017-08-27:
And the twisted yellow and black wires coming of of spindle power is +12V for brushless motor cooling fan which is now cooling the drivers

chamnit, 2017-08-27:
This really looks more like lost steps to me and the motors are stalling occasionally. In your video, you state that the loud rattling in your machine is from eBay bearings. I'm not sure if that is really the case, but I would suspect that your machine is prone to binding randomly due to the rattling. This binding is causing your steppers to lose steps.

I don't think this is backlash, because it makes a decent circle pretty consistently. I don't think it's something electrical because it would happen much more often than what is in the video.

Daumaz, 2017-08-27:
Those are plywood discs held together by glue and hot glue to give some flex. The screw side has a nut inside and the metal plate for pressing the last disc to the screw. Yeah sketchy :), but works. The whole screw system is temporary, I will get the 8mm diameter 2mm pitch ones with good ball bearings later, just what to see it move correctly.

Daumaz, 2017-08-27:
I had problems with slipping at first chamnit, I have heard the sound it makes. My video did not catch any big ones, but when it moves extra 5mm you can really hear the motors give the extra effort to do so. The only thing about possible slip would be the Z axis while its accelerating, it just does not sound completely right, steps are I would say more violent. That is why I increased the acceleration of the Z axis a lot, for the weight it caries it does not have any issues working even without acceleration (full speed always)

chamnit, 2017-08-27:
Lost steps isn't slipping. It's your motor failing to move. If it's doing it during a change in direction, it can be instantaneous and hard to hear. Your problem doesn't have anything to do with the Z-axis, it's the X and Y during the job. And you should be **decreasing** acceleration, rather than increasing it (`F=ma`), because you want to reduce the force the motor has to drive.

Daumaz, 2017-08-27:
jahnj0584 i use 2 regular nuts spaced from one another by plywood plates and sheets of cardboard for final adjustment

Daumaz, 2017-08-27:
I understand that I am making things more difficult for the Z axis and it has nothing to do with the main problem, just thought I mention it. I always thought lost steps and slipping are the same, now i see what you mean, I will try less acceleration then.

chamnit, 2017-08-27:
I meant less acceleration for Z. If you are experiencing lost steps at 10mm/s^2 already, reducing it further won't really help that much. I would really concentrate on reducing friction and binding on your machine. If you think it's something else, you can detach your steppers from your machine and come up with a way to check if any of your steppers are going the wrong direction. Anyhow, this issue is closed, since this isn't a problem with Grbl itself.

closed 2017-08-27 by chamnit

electrokean, 2017-08-27:
Two suggestions from the electronics perspective
1. Make sure your ground wires connecting the digital logic sections are not carrying the motor currents, and that there are no ground loops.  A star grounding configuration is often best.  With the motors drawing several amps through thin wires you could see significant voltage differentials across the wiring.
2. Try running the motors at 12V, eliminating the 24V boost circuits which could be struggling or inducing noise.  You may need to reduce your speeds/accelerations with the lower voltage