grbl/grbl #1279: How to hack overrides

dastultz, 2017-08-27:
Hello, I am customizing GRBL for my hardware, a Sanguino/644p CNC machine. I have a number of buttons and a slider (pot) for jogging and override. I have tapped into `protocol_main_loop` to run my code. I have jogging working great. I'm trying to issue override commands at the push of a button. I thought it would be pretty simple. I see in the ISR for `serial.c` that the incoming RT commands result in calls to `system_set_exec_state_flag`. So I tried that but it doesn't seem to be working. I thought perhaps things might get reset at the top of the next loop, so I tried calling `protocol_execute_realtime` after with no change in behavior. Consider this:

```
printString("feed coarse plus\r\n");
system_set_exec_motion_override_flag(EXEC_FEED_OVR_COARSE_PLUS);
protocol_execute_realtime();
```
The print statement is visible in the console, but the override does not take place.

What is the best way to trigger rapid/feed override from within GRBL?

Thanks.

109JB, 2017-08-27:
> What is the best way to trigger rapid/feed override from within GRBL?

Upgrade to V1.1 of grbl which has overrides already implemented

dastultz, 2017-08-28:
Yes, I'm running 1.1f of GRBL. I see that you can command overrides over serial. I'm asking how it can be done within GRBL. What functions do I need to call? I'm coding "pendant" controls within one microcontroller running GRBL. So I'm looking at digital inputs (buttons) and I want to call functions to execute the overrides. I am not asking how to command overrides over a serial connection. Thank you.

electrokean, 2017-08-28:
You should be asking at https://github.com/gnea/grbl/ then - this repo is for grbl 0.9

dastultz, 2017-08-28:
Interesting. I don't see any evidence of that, but very well, I'll move along.

closed 2017-08-28 by dastultz

electrokean, 2017-08-28:
No worries, just didn't want you wasting more time because your issue was posted in the wrong place.
Just for reference, there is a bold note at the top of this README.md that points you to the new repo for 1.1f and hinting that 0.9 is deprecated.  You must have gone to the new repo if you're working with the 1.1f source.