grbl/grbl #1280: Led Lighting

CLMNTDSGNR, 2017-08-30:
Hello everyone,

This is not really a problem but rather whether it is feasible.
I would like to put leds on my machine (RGB tape). I would like to provide them with a different color depending on the state. Red = Error / Cycle Stop, Orange = Feed Hold, Green then white, Start of cycle.

Aside from sending a command to an arduino slave? But no idea if this is feasible given the limitation of the Arduino.

What type of buttons are used for grbl for the Abort / Start / Feed Hold functions?
With hold? If so, it would be possible to use the button lighting as "signal" that would be sent to another arduino to manage the lighting?

Thank you

jahnj0584, 2017-08-30:
They are on the A pins. You would share their ground, as the switch isn't
activated until the circuit is closed.

RabbitMountain, 2021-08-08:
I would be curious to know if you ever implemented this strategy. I think having two arduinos communicating through the serial port is a possible way to increase the available I/O ports that could be used for many LEDs.

But also see https://github.com/grbl/grbl/issues/628

I just figured out how to hack GRBL to turn on only one diagnostic LED attached to Analog4. I have it set to turn on when an alarm state is active (such as failure to home the machine).