grbl/grbl #1281: laser issue

Davida54, 2017-08-31:
when using laser, the laser is on when moving between positions and off when it should be burning. In other words its backwards.

dutchpatriot, 2017-09-02:
Could you drop the gcode file here?

daniele-salvagni, 2017-09-05:
The laser will fire when the output is low and not high, see here: https://youtu.be/Aj-kghLsdEI?t=310

chamnit, 2017-09-05:
Grbl PWM output for lasers requires low for off and high for on. If your system needs the opposite, you will need to alter the spindle control source code to work with your system. 

Inverted PWM output is slated to be added at some point but I have zero time right now to work on it.  FWIW, it's not that difficult to alter the source.

closed 2017-09-05 by chamnit

daniele-salvagni, 2017-09-05:
Wouldn't an hardware inverter be safer? I fear that the pin could be set low during initialization and fire the laser.

chamnit, 2017-09-05:
Sure. Do whatever is appropriate for your machine. There are an infinite number of ways to deal with the initialization problem. That would be one.

Davida54, 2017-09-07:
I got it. Thanks for help

David