grbl/grbl #1284: bursts of speed

CliveMcCarthy, 2017-09-08:
I'm writing a GUI for Grbl in C with GTK+3.0 on Linux. It is a simpler interface than other GUIs because our work is 2D with modest Z demands (large art drawings on a plotter). I'm using a serial port with the canonical protocol and simple send and acknowledge interface to Grbl 0.9j

Most of the G-Code instructions have VERY short distances ~1mm and there a lot of them. I find that most G-Code lines execute very fast (in less than 100ms) but some, particularly at start-up, take several seconds. At times it can take Grbl 12 seconds to respond to a G-Code line.

Is this to be expected or do I have a bug?

chamnit, 2017-09-08:
You likely have a bug. Grbl should execute any command it received almost immediately. Grbl will slow down if it has to execute lots of very short lines rapidly. This is due to the planner not having a significant distance to accelerate over. Look at the stream.py scripts in the repo to see how to write a simple streamer. Also, test against another GUI like UGS or bCNC to test performance.

closed 2017-09-08 by chamnit

CliveMcCarthy, 2017-09-08:
Things work fine in $C check mode. I get over 400 lines per second in that mode with no error or delays.

chamnit, 2017-09-08:
@CliveMcCarthy : Yes, because check mode doesn't plan anything. Just parse.

CliveMcCarthy, 2017-09-09:
It seems that, at maximum, I can pump about 35
          lines of 26 bytes per second from my code to Grbl with the
          planner (presumably) occasionally holding things up for a few
          tenths of a second before continuing.
        
      Is this
              how things should be?
              
            
          I have run 50,000 lines of G-Code and the initial erratic
          behavior in the run seems to settle down. Does the planner
          'learn' from the 'style' of code it is given?
        
      
      
      
    
    On 09/08/2017 04:09 PM, Sonny Jeon
      wrote:
    
    
      @CliveMcCarthy : Yes, because check
        mode doesn't plan anything. Just parse.

CliveMcCarthy, 2017-09-09:
After more testing it looks as if I can send an average of 40 G-Code lines per second which, at 1mm per line, is 2,400 mm/min. There are repeated periodic delays up to 12 seconds.

CliveMcCarthy, 2017-09-09:
When Grbl is in check mode data transfers at 400 lines per second, so I know the serial link is good. Do you have any ideas for me, to speed up the planner/execution process on the Arduino?

[image]

CliveMcCarthy, 2017-09-09:
Looks as
        if I shall have to wait for the ARM version:
      
    There not much of a "team" Grbl, as it's
      just been myself for a very long time. 8-bit support will be
      dropped soon after v1.0 is released. I will be moving on to
      investing all my time to continuing to develop the ARM version,
      which is a complete re-write of the Grbl base code. Grbl itself is
      designed specifically for the 8-bit AVR and has a lot of
      "unsavory" code that was written for the interest of maximizing
      performance and reducing compiled flash size. Mainly the latter.
      The ARM version has no such restriction, and there are a number of
      algorithmic improvements that will make it better on a fundamental
      level. I won't divulge what they are, but they will be awesome.
      
    
    On 09/08/2017 04:09 PM, Sonny Jeon
      wrote:
    
    
      @CliveMcCarthy : Yes, because check
        mode doesn't plan anything. Just parse.

chamnit, 2017-09-10:
Did you do what I asked and test against a reputable GUI?

CliveMcCarthy, 2017-09-11:
I did a quick test with UGS and found it could run
          my test G-Code, which has about 140,000 lines, in about an
          hour. Quite satisfactory.
      
      A
          typical write of a typical G-Code line of mine (30 bytes)
          takes less than 10 microseconds from my code.
          
        If I run my test G-Code,
          with my program and with check mode on, the response from Grbl
          mostly takes less than 10 microseconds but quite commonly
          4,000 microseconds too. Odd really because the lines are very
          similar and one would expect a near constant time for parsing.
          The write-and-acknowledge cycle time is fast enough and it
          convinces me that the hardware/software protocol between my
          computer and the Arduino is running properly.
          
          If I run it in operational mode things slow down badly. The
          time for the 3 byte read response (Grbl's 'ok'
          acknowledgement) varies a great deal from returning as fast as
          5 milliseconds and commonly 20 milliseconds but
              often 100 milliseconds and periodically many
              seconds.
              
              I've tried adding a delay after the write before the read
              takes place but it does not improve the slow response. If
              the write-and-acknowledge
                  cycle time were a consistent 25 milliseconds it would
                  approximately equal the cycle time with UGS.
                  
                It is very odd.
            
    
    On 09/09/2017 09:52 PM, Sonny Jeon
      wrote:
    
    
      Did you do what I asked and test against a reputable GUI?

CliveMcCarthy, 2017-09-11:
I'm re-running the same G-Code on UGS and it looks
          as if it is sending a G-Code line in 30ms. This limits our
          drawing rate to about 2000mm/min which is much lower than our
          machine can handle. It is clearly good enough for most
          machining tasks where the movements are usually much larger
          and the processing overhead much smaller. In our case we have
          ~1mm movements and a G-Code line for each and as many as a
          million lines of G-Code. The processing in Grbl dominates the
          speed of execution, not the CNC machine itself. I thought that
          my own driver at 25ms/line was a problem.
      
      Our
          present driver system produces noticeable faster work when
          running at 6,000 versus 7,000 mm/min. It seems Grbl won't be
          able to do that if the G-Code processing/planning rate is
          limited to 25 to 30ms per line. I had noticed, on a much small
          machine, that the F6000 rate had no effect. Now I know why.
          
          When I read the Wiki on Grbl I thought that keeping the input
          buffer full would be the limiting factor. It appears not to be
          the case. I presume that UGS and bCNC both use the smart
          character counting method.
          
          For your ARM version you might consider a target of 10ms per
          line of G-Code.
        
    
    On 09/09/2017 09:52 PM, Sonny Jeon
      wrote:
    
    
      Did you do what I asked and test against a reputable GUI?

CliveMcCarthy, 2017-09-11:
This screen shot illustrates the issues well. At the end of the
    sequence there is an M08 instruction and Grbl takes 24 seconds to
    respond.
    
    
    
    
    On 09/09/2017 09:52 PM, Sonny Jeon
      wrote:
    
    
      Did you do what I asked and test against a reputable GUI?

CliveMcCarthy, 2017-09-11:
[image]

chamnit, 2017-09-11:
Set your $1 setting to $1=255. Grbl can handle up to 400 lines per second. That's 2.5 ms per line. Not 25. The machine is likely stopping, enforcing the step delay, then resuming. The default is about 25 ms, which would explain your problem

chamnit, 2017-09-11:
The longer delays you are seeing are from your g-code. Grbl has to execute all buffered motions prior to executing a line that requires a sync with motion. M8 M9 M30, etc all qualify as a sync command.

CliveMcCarthy, 2017-09-11:
That seems
        to help a lot. Does a feed rate change also count as a 'sync
        with motion' because each feed rate change seems to cause a
        'stall' too? There are other 'stalls' seemingly unrelated to
        anything...
        
        BTW, thank for you assistance with this.
      
    On 09/11/2017 08:52 AM, Sonny Jeon
      wrote:
    
    
      Set your $1 setting to $1=255. Grbl can handle up to 400 lines
        per second. That's 2.5 ms per line. Not 25. The machine is
        likely stopping, enforcing the step delay, then resuming. The
        default is about 25 ms, which would explain your problem
      —
      
        
          
          
        
        
      
 t>
</body>
</html>
</html></blockquote>
    <br>
  </body>
</html>

CliveMcCarthy, 2017-09-11:
Do you
        have any other Grbl suggestions? I'm still getting just a little
        less than 20ms per line.
    
    On 09/11/2017 08:52 AM, Sonny Jeon
      wrote:
    
    
      Set your $1 setting to $1=255. Grbl can handle up to 400 lines
        per second. That's 2.5 ms per line. Not 25. The machine is
        likely stopping, enforcing the step delay, then resuming. The
        default is about 25 ms, which would explain your problem
      —
      
        
          
          
        
        
      
 t>
</body>
</html>
</html></blockquote>
    <br>
  </body>
</html>

CliveMcCarthy, 2017-09-11:
I have my
        code calculating a running average of the response time. After
        60,000 lines the average is 10.8ms.
    
    On 09/11/2017 09:00 AM, Sonny Jeon
      wrote:
    
    
      The longer delays you are seeing are from your g-code. Grbl has
        to execute all buffered motions prior to executing a line that
        requires a sync with motion. M8 M9 M30, etc all qualify as a
        sync command.

chamnit, 2017-09-11:
Are you still using the send response protocol? If so, test again with UGS or the GRBL stream.py script. They use the character counting protocol.

CliveMcCarthy, 2017-09-11:
Yes, I'm
        still using the send-response protocol. My latest run with
        $1=255 has improved things a good deal.
        
        Previously ~140,000 lines took 80 minutes on UGS or my
        code, they were the same. Now the job runs in 50 minutes
        on my code, however, the average response time from Grbl is
        13.6ms.
        
        My code sends a line of code in ~5 microseconds so as soon as
        Grbl wants another line, its there (the C code will be much
        faster that Python or Java). All my G-Code lines are very close
        to the same size: 27 to 30 bytes. I guess I could pre-load the
        127 byte input buffer with three or four lines and not wait for
        an acknowledgement. Then proceed a line at a time.
        
        Does Grbl send an acknowledgement when the planner has finished
        a line or does the input buffer acknowledge the receipt of a
        line?
      
    On 09/11/2017 11:21 AM, Sonny Jeon
      wrote:
    
    
      Are you still using the send response protocol? If so, test
        again with UGS or the GRBL stream.py script. They use the
        character counting protocol.

CliveMcCarthy, 2017-09-11:
I took my 140,000 line test and ran it on UGS with
          $1=255. The run time was 50 minutes. The run time on my code
          was 49.8 minutes. I conclude that the character counting helps
          the Java sender, however, my dumb send-and-acknowledge is
          probably compensated for by the faster feed from the C program
          which keeps the pipeline full.
          
          UGS also experiences some stalling but, of course, I don't
          have numbers, just a visual impression.
          
          In another week I shall be able to run this 140,000 line test
          on our CNC machine with its current driver and software. Then
          I will know if I can cut over the hardware to the Arduino and
          Grbl. I hope I can, because I dislike using XP and with a
          legacy parallel port interface.
          
        
    
    On 09/11/2017 11:21 AM, Sonny Jeon
      wrote:
    
    
      Are you still using the send response protocol? If so, test
        again with UGS or the GRBL stream.py script. They use the
        character counting protocol.

CliveMcCarthy, 2017-09-12:
I have my code driving Grbl in a stable condition with a very big G-Code file. Its been running consistently for more than eight hours and has six hours to go.

It seems that Grbl stalls every 100~120 lines or so. My code sends a '?' status request every twenty lines and the planner buffer, with few exceptions, reports 17 -- it is always full -- even immediately  prior to a stall.

The stalls are characterized by read responses that take more than a second. Normal responses take less than 20ms.

CliveMcCarthy, 2017-09-12:
My first experiment this morning has been to increase the junction deviation to 1mm from 0.02mm this seems to decrease the number stalls. I now count as a stall as any response that is 100ms or greater. The vast majority are under 20ms.

chamnit, 2017-09-12:
It's not secret that compiled C executables are faster than Python or Java. I'm not surprised that the latency of your GUI is low. 

I'm starting to feel like I'm repeating myself over and over again. It's a waste of time and energy. Please consider the character counting protocol. As the wiki interface page states, it's a lot more efficient, will speed up the job, and is recommended for more demanding jobs. The main reason being is that you haven't considered the time it takes to transmit a serial message to Grbl. For 25 characters at 115200 baud, that's about 2.2 milliseconds. Grbl has to wait until it has received the complete message before parsing it and executing it. Also, consider that the time that it takes for Grbl to transmit an 'ok' back to your GUI. Albeit in less time, Grbl can immediately start executing the next line, before it transmits the 'ok' and the latency time of the USB protocol + your host computer.

Please try to understand how Grbl works before placing a 'performance' time on it. Grbl has a planner of 15 blocks or so. It plans for optimal acceleration and execution time within the total distance inside the planner. If the blocks are all very short lines, then it doesn't have that much distance or time inside it. If your acceleration is low, then it can't accelerate to a high velocity before it has to decelerate to a stop at the end of the buffer to always ensure the motions are always managed. These factors, including junction deviation (alters how fast it can move through a junction), all effect Grbl's performance, but do not have much to do with communication. It's a limit of your settings, planner size, and planner assumptions.

As far as your 1 second read response issue, it's very hard to tell what you are defining as 1 second. I think you need to share your $$ settings, your $I build info strings, and your test program. So I can see what you are trying to do and help identify what the primary issue is.

CliveMcCarthy, 2017-09-12:
I just
        tried to send you the test code but the file is 7.5MB and it
        exceeds the Github email server's limit.
        
      
    The response from the remote server was:
    
      552 5.3.4 Message size exceeds fixed limit
    
    On 09/12/2017 08:16 AM, Sonny Jeon
      wrote:
    
    
      It's not secret that compiled C executables are faster than
        Python or Java. I'm not surprised that the latency of your GUI
        is low.
      I'm starting to feel like I'm repeating myself over and over
        again. It's a waste of time and energy. Please consider the
        character counting protocol. As the wiki interface page states,
        it's a lot more efficient, will speed up the job, and is
        recommended for more demanding jobs. The main reason being is
        that you haven't considered the time it takes to transmit a
        serial message to Grbl. For 25 characters at 115200 baud, that's
        about 2.2 milliseconds. Grbl has to wait until it has received
        the complete message before parsing it and executing it. Also,
        consider that the time that it takes for Grbl to transmit an
        'ok' back to your GUI. Albeit in less time, Grbl can immediately
        start executing the next line, before it transmits the 'ok' and
        the latency time of the USB protocol + your host computer.
      Please try to understand how Grbl works before placing a
        'performance' time on it. Grbl has a planner of 15 blocks or so.
        It plans for optimal acceleration and execution time within the
        total distance inside the planner. If the blocks are all very
        short lines, then it doesn't have that much distance or time
        inside it. If your acceleration is low, then it can't accelerate
        to a high velocity before it has to decelerate to a stop at the
        end of the buffer to always ensure the motions are always
        managed. These factors, including junction deviation (alters how
        fast it can move through a junction), all effect Grbl's
        performance, but do not have much to do with communication. It's
        a limit of your settings, planner size, and planner assumptions.
      As far as your 1 second read response issue, it's very hard to
        tell what you are defining as 1 second. I think you need to
        share your $$ settings, your $I build info strings, and your
        test program. So I can see what you are trying to do and help
        identify what the primary issue is.

CliveMcCarthy, 2017-09-12:
$1=255 (step idle delay, msec)
	$2=0 (step port invert mask:00000000)
	$3=0 (dir port invert mask:00000000)
	$4=0 (step enable invert, bool)
	$5=0 (limit pins invert, bool)
	$6=0 (probe pin invert, bool)
	$10=12 (status report mask:00001100)
	$11=1.000 (junction deviation, mm)
	$12=1.000 (arc tolerance, mm)
	$13=0 (report inches, bool)
	$20=0 (soft limits, bool)
	$21=0 (hard limits, bool)
	$22=0 (homing cycle, bool)
	$23=0 (homing dir invert mask:00000000)
	$24=25.000 (homing feed, mm/min)
	$25=500.000 (homing seek, mm/min)
	$26=250 (homing debounce, msec)
	$27=1.000 (homing pull-off, mm)
	$100=250.000 (x, step/mm)
	$101=250.000 (y, step/mm)
	$102=250.000 (z, step/mm)
	$110=15000.000 (x max rate, mm/min)
	$111=15000.000 (y max rate, mm/min)
	$112=4000.000 (z max rate, mm/min)
	$120=10000.000 (x accel, mm/sec^2)
	$121=10000.000 (y accel, mm/sec^2)
	$122=10.000 (z accel, mm/sec^2)
	$130=1200.000 (x max travel, mm)
	$131=1000.000 (y max travel, mm)
	$132=200.000 (z max travel, mm)
	ok

chamnit, 2017-09-12:
You are exceeding Grbl's max supported 30kHz step rate. (15,000 mm/min  *250 step/mm)/60 sec/min -> 62.5 kHz. Grbl doesn't actively perform a check for this, since there isn't any flash space left in v1.1. All remaining bytes are reserved for bug fixes.

CliveMcCarthy, 2017-09-12:
Really
        small
    
    On 09/12/2017 08:16 AM, Sonny Jeon
      wrote:
    
    
      It's not secret that compiled C executables are faster than
        Python or Java. I'm not surprised that the latency of your GUI
        is low.
      I'm starting to feel like I'm repeating myself over and over
        again. It's a waste of time and energy. Please consider the
        character counting protocol. As the wiki interface page states,
        it's a lot more efficient, will speed up the job, and is
        recommended for more demanding jobs. The main reason being is
        that you haven't considered the time it takes to transmit a
        serial message to Grbl. For 25 characters at 115200 baud, that's
        about 2.2 milliseconds. Grbl has to wait until it has received
        the complete message before parsing it and executing it. Also,
        consider that the time that it takes for Grbl to transmit an
        'ok' back to your GUI. Albeit in less time, Grbl can immediately
        start executing the next line, before it transmits the 'ok' and
        the latency time of the USB protocol + your host computer.
      Please try to understand how Grbl works before placing a
        'performance' time on it. Grbl has a planner of 15 blocks or so.
        It plans for optimal acceleration and execution time within the
        total distance inside the planner. If the blocks are all very
        short lines, then it doesn't have that much distance or time
        inside it. If your acceleration is low, then it can't accelerate
        to a high velocity before it has to decelerate to a stop at the
        end of the buffer to always ensure the motions are always
        managed. These factors, including junction deviation (alters how
        fast it can move through a junction), all effect Grbl's
        performance, but do not have much to do with communication. It's
        a limit of your settings, planner size, and planner assumptions.
      As far as your 1 second read response issue, it's very hard to
        tell what you are defining as 1 second. I think you need to
        share your $$ settings, your $I build info strings, and your
        test program. So I can see what you are trying to do and help
        identify what the primary issue is.

CliveMcCarthy, 2017-09-12:
The
        character count protocol is used in UGS, right? When I run my
        test code on UGS it finishes the job in the same time
        as my code does. The character counting won't help.
    
    On 09/12/2017 08:16 AM, Sonny Jeon
      wrote:
    
    
      It's not secret that compiled C executables are faster than
        Python or Java. I'm not surprised that the latency of your GUI
        is low.
      I'm starting to feel like I'm repeating myself over and over
        again. It's a waste of time and energy. Please consider the
        character counting protocol. As the wiki interface page states,
        it's a lot more efficient, will speed up the job, and is
        recommended for more demanding jobs. The main reason being is
        that you haven't considered the time it takes to transmit a
        serial message to Grbl. For 25 characters at 115200 baud, that's
        about 2.2 milliseconds. Grbl has to wait until it has received
        the complete message before parsing it and executing it. Also,
        consider that the time that it takes for Grbl to transmit an
        'ok' back to your GUI. Albeit in less time, Grbl can immediately
        start executing the next line, before it transmits the 'ok' and
        the latency time of the USB protocol + your host computer.
      Please try to understand how Grbl works before placing a
        'performance' time on it. Grbl has a planner of 15 blocks or so.
        It plans for optimal acceleration and execution time within the
        total distance inside the planner. If the blocks are all very
        short lines, then it doesn't have that much distance or time
        inside it. If your acceleration is low, then it can't accelerate
        to a high velocity before it has to decelerate to a stop at the
        end of the buffer to always ensure the motions are always
        managed. These factors, including junction deviation (alters how
        fast it can move through a junction), all effect Grbl's
        performance, but do not have much to do with communication. It's
        a limit of your settings, planner size, and planner assumptions.
      As far as your 1 second read response issue, it's very hard to
        tell what you are defining as 1 second. I think you need to
        share your $$ settings, your $I build info strings, and your
        test program. So I can see what you are trying to do and help
        identify what the primary issue is.

CliveMcCarthy, 2017-09-12:
This is what my collaborator and I are working on. The work you see on the wall is made from four 1200mm x 800mm ink drawings on paper.

[image]

chamnit, 2017-09-12:
Not sure if you got my response or not, but the problem is you are exceeding Grbl's 30kHz max supported step rate. This is causing your Arduino to freeze up and likely crash. All of the Arduino's time is spent on servicing the step pulse and not planning. Reduce your step/mm to something more realistic like 40step/mm to 80 step/mm.

CliveMcCarthy, 2017-09-12:
$1=255 (step idle delay, msec)
	$2=0 (step port invert mask:00000000)
	$3=0 (dir port invert mask:00000000)
	$4=0 (step enable invert, bool)
	$5=0 (limit pins invert, bool)
	$6=0 (probe pin invert, bool)
	$10=12 (status report mask:00001100)
	$11=1.000 (junction deviation, mm)
	$12=1.000 (arc tolerance, mm)
	$13=0 (report inches, bool)
	$20=0 (soft limits, bool)
	$21=0 (hard limits, bool)
	$22=0 (homing cycle, bool)
	$23=0 (homing dir invert mask:00000000)
	$24=25.000 (homing feed, mm/min)
	$25=500.000 (homing seek, mm/min)
	$26=250 (homing debounce, msec)
	$27=1.000 (homing pull-off, mm)
	$100=50.000 (x, step/mm)
	$101=50.000 (y, step/mm)
	$102=250.000 (z, step/mm)
	$110=5000.000 (x max rate, mm/min)
	$111=5000.000 (y max rate, mm/min)
	$112=4000.000 (z max rate, mm/min)
	$120=500.000 (x accel, mm/sec^2)
	$121=500.000 (y accel, mm/sec^2)
	$122=10.000 (z accel, mm/sec^2)
	$130=1200.000 (x max travel, mm)
	$131=1000.000 (y max travel, mm)
	$132=200.000 (z max travel, mm)
	ok

CliveMcCarthy, 2017-09-12:
I just
        did. I set it to 50steps/mm. The Arduino didn't crash. I'm still
        getting stalls from Grbl, but fewer.
        
        Is there anything else I should loosen up?
        
          $1=255 (step idle delay, msec)
        $2=0 (step port invert mask:00000000)
        $3=0 (dir port invert mask:00000000)
        $4=0 (step enable invert, bool)
        $5=0 (limit pins invert, bool)
        $6=0 (probe pin invert, bool)
        $10=12 (status report mask:00001100)
        $11=1.000 (junction deviation, mm)
        $12=1.000 (arc tolerance, mm)
        $13=0 (report inches, bool)
        $20=0 (soft limits, bool)
        $21=0 (hard limits, bool)
        $22=0 (homing cycle, bool)
        $23=0 (homing dir invert mask:00000000)
        $24=25.000 (homing feed, mm/min)
        $25=500.000 (homing seek, mm/min)
        $26=250 (homing debounce, msec)
        $27=1.000 (homing pull-off, mm)
        $100=50.000 (x, step/mm)
        $101=50.000 (y, step/mm)
        $102=250.000 (z, step/mm)
        $110=5000.000 (x max rate, mm/min)
        $111=5000.000 (y max rate, mm/min)
        $112=4000.000 (z max rate, mm/min)
        $120=500.000 (x accel, mm/sec^2)
        $121=500.000 (y accel, mm/sec^2)
        $122=10.000 (z accel, mm/sec^2)
        $130=1200.000 (x max travel, mm)
        $131=1000.000 (y max travel, mm)
        $132=200.000 (z max travel, mm)
        ok
    
    On 09/12/2017 09:20 AM, Sonny Jeon
      wrote:
    
    
      Not sure if you got my response or not, but the problem is you
        are exceeding Grbl's 30kHz max supported step rate. This is
        causing your Arduino to freeze up and likely crash. All of the
        Arduino's time is spent on servicing the step pulse and not
        planning. Reduce your step/mm to something more realistic like
        40step/mm to 80 step/mm.

chamnit, 2017-09-12:
Re-test UGS.

CliveMcCarthy, 2017-09-12:
Will do.
        
        I just ran the file I sent you named hello_image_small.cnc
        -- it ran on my code with those Grbl settings in 14.12 minutes.
        I had to reduce the drawing size a lot to get the file size down
        so that I could send it to you.
      
    On 09/12/2017 09:43 AM, Sonny Jeon
      wrote:
    
    
      Re-test UGS.

CliveMcCarthy, 2017-09-12:
The UGS
        result was 14.5 minutes. So the results are the same. The stalls
        are quite noticeable on UGS too.
    
    On 09/12/2017 09:43 AM, Sonny Jeon
      wrote:
    
    
      Re-test UGS.

usbcnc, 2017-09-12:
@CliveMcCarthy Interesting to see your drawing. Do you mind send me your file? I can try it on STM32 port as well as on my Windows simulation to see if I can find something.

CliveMcCarthy, 2017-09-12:
You may
        find this low-level C code useful for testing future Grbl
        releases.
    
    On 09/12/2017 09:43 AM, Sonny Jeon
      wrote:
    
    
      Re-test UGS.

CliveMcCarthy, 2017-09-12:
I'm afraid
        the G-Code files for our  actual drawings are
        too big to send via Github. The small test file I sent to Sonny
        exhibits the issues I'm having with Grbl on the Arduino.
        
        Is your STM32 port compatible with Grbl on the Arduino?
      
    On 09/12/2017 10:14 AM, usbcnc wrote:
    
    
      @CliveMcCarthy Interesting to see
        your drawing. Do you mind send me your file? I can try it on
        STM32 port as well as on my Windows simulation to see if I can
        find something.

usbcnc, 2017-09-12:
@CliveMcCarthy You cannot zip the file? The STM32 port is on ARM and it can easily get 180KHz step clock. https://www.youtube.com/watch?v=5DBtbAYmfEE&t=2s

chamnit, 2017-09-12:
I have not received any file. Rename to .txt and drag and drop into your github response.

CliveMcCarthy, 2017-09-12:
Will do in the early afternoon.

CliveMcCarthy, 2017-09-12:
I can send you other drawing files. The one you saw is part of a limited
edition.

chamnit, 2017-09-12:
If you don't want to publicly share anything, you can email me.

CliveMcCarthy, 2017-09-12:
This has my low-level interface to Grbl. Nothing special except it does attempt to measure the speed of transfers. It probably would need some work to run on Windows but will be fine for Linux.

[Arduino_serial_interface.txt](https://github.com/grbl/grbl/files/1297140/Arduino_serial_interface.txt)

chamnit, 2017-09-12:
No I meant to share your g-code file you are testing with. That's all I need to see where the problem is related to Grbl.

CliveMcCarthy, 2017-09-12:
Oh sure. This is the file that runs in 14 minutes.
[hello_image_small.txt](https://github.com/grbl/grbl/files/1297166/hello_image_small.txt)

CliveMcCarthy, 2017-09-12:
Here is a large drawing that USBCNC can try. [angel.cnc.gz](https://github.com/grbl/grbl/files/1297168/angel.cnc.gz)

CliveMcCarthy, 2017-09-12:
There is
        nothing at all secret about the test file. If you run it on UGS
        with visualization on you will see why!
    
    On 09/12/2017 12:35 PM, Sonny Jeon
      wrote:
    
    
      No I meant to share your g-code file you are testing with.
        That's all I need to see where the problem is related to Grbl.

chamnit, 2017-09-12:
@CliveMcCarthy : There are NO performance issues anywhere in your g-code program. I just tested it with the stream.py script and Grbl v1.1 (essentially the same as v0.9 in terms of performance). Grbl runs it exactly as it should. Going to ARM will make no difference. Let me tell you why. 

You have combined 3-axis motions programmed throughout your entire program. You have a very tiny Z-axis acceleration at $122=10.00, compared to your x and y axes at 10000mm/s^2 and 500mm/s^2 (in your reduced settings). What you think is stalling, isn't stalling. It's Grbl intentionally slowing down and making sure that the z-axis acceleration limit isn't exceeded.

The reason why UGS and your GUI program resulted in the same time is that this program isn't very demanding. When I say demanding, is when there are 400 lines per second and each line is about 0.1mm long. This is a common issue with laser engraving (note the Grbl LPC port can do about 1300 lines per second).

If you want me to test against your first test, with faster feed rates, you'll need to upload that file or at least a similar file that i can compare execution times (with your old settings except your step/mm at all 50.0).

CliveMcCarthy, 2017-09-12:
Ahh! Good!
        I'm running my tests with all three axes set the same. I won't
        know what my machine can  actually do
        until next week.
    
    On 09/12/2017 01:04 PM, Sonny Jeon
      wrote:
    
    
      @CliveMcCarthy : There are NO
        performance issues anywhere in your g-code program. I just
        tested it with the stream.py script and Grbl v1.1 (essentially
        the same as v0.9 in terms of performance). Grbl runs it exactly
        as it should. Going to ARM will make no difference. Let me tell
        you why.
      You have combined 3-axis motions programmed throughout your
        entire program. You have a very tiny Z-axis acceleration at
        $122=10.00, compared to your x and y axes at 10000mm/s^2 and
        500mm/s^2 (in your reduced settings). What you think is
        stalling, isn't stalling. It's Grbl intentionally slowing down
        and making sure that the z-axis acceleration limit isn't
        exceeded.
      The reason why UGS and your GUI program resulted in the same
        time is that this program isn't very demanding. When I say
        demanding, is when there are 400 lines per second and each line
        is about 0.1mm long. This is a common issue with laser engraving
        (note the Grbl LPC port can do about 1300 lines per second).
      If you want me to test against your first test, with faster
        feed rates, you'll need to upload that file or at least a
        similar file that i can compare execution times (with your old
        settings except your step/mm at all 50.0).

CliveMcCarthy, 2017-09-13:
BTW , thank you 😊

usbcnc, 2017-09-13:
@CliveMcCarthy 
At following setting.
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 15000.000    (X-axis maximum rate, mm/min)
$111 = 15000.000    (Y-axis maximum rate, mm/min)
$112 = 4000.000    (Z-axis maximum rate, mm/min)
$120 = 10000.000    (X-axis acceleration, mm/sec^2)
$121 = 10000.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)

I run angel.cnc on STM32F103 GRBL for 3:19 and no issue at all. (18xxxx lines).
Run on Win32 simulation for 2:30 (Win32 speed is not accurate) . I can confirm there is no issue in GRBL logic, Most likely related to your file has reach the max clock rate for Arduino.