grbl/grbl #1285: Is there a command that stops all motors.

StuartB4, 2017-09-17:
I've noticed people talking about using M18 on some machines to stop all motors so that no current is present and they can be moved freely. Can GRBL do it with a different command because M18 is not recognised as a valid command.

Thanks.

jahnj0584, 2017-09-17:
Setting $1 to anything below 255 will stop the flow of current after a move
is made

StuartB4, 2017-09-17:
Ok jahnj0584, thanks very much.

PicEngraver, 2017-09-22:
Stopping current to the motors means risking they may shift position slightly when the current resumes.   I, for one, choose to not risk that, and I maintain $1=255 all the time.

StuartB4, 2017-09-22:
I'm not bothered about it shifting position as I only want it to release the motors after finishing a job.

chamnit, 2017-09-22:
There is a sleep command that will do what you want. See the wiki commands page.

chamnit, 2017-09-22:
It is only available on Grbl v1.1.

closed 2017-09-22 by chamnit

StuartB4, 2017-09-22:
Ok chamnit. Thanks very much.