grbl/grbl #1286: No PWM output on pin 11

Mark-81, 2017-09-22:
I'm reading the documentation (https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode) but on pin 11 the voltage is always 0V.

Here my configuration:

`
$0=10 (Step pulse time)
$1=250 (Step idle delay)
$2=0 (Step pulse invert)
$3=4 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=1 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=7 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=500.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=5.000 (Homing switch pull-off distance)
$30=255 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=80.000 (X-axis travel resolution)
$101=80.000 (Y-axis travel resolution)
$102=200.000 (Z-axis travel resolution)
$110=10000.000 (X-axis maximum rate)
$111=10000.000 (Y-axis maximum rate)
$112=5000.000 (Z-axis maximum rate)
$120=1000.000 (X-axis acceleration)
$121=1000.000 (Y-axis acceleration)
$122=100.000 (Z-axis acceleration)
$130=600.000 (X-axis maximum travel)
$131=400.000 (Y-axis maximum travel)
$132=80.000 (Z-axis maximum travel)
`

After unlocking the machine I issues M3 S128 (or any other S value) but the voltage on pin 11 is always 0V.
Is there something else to check?

vMeph, 2017-09-22:
are you using CNC Shield V3?

Mark-81, 2017-09-22:
I'm using the Synthetos shield (https://www.synthetos.com/project/grblshield/)

chamnit, 2017-09-22:
Read the laser mode wiki page carefully. It explains the safety measures that keep the laser disabled in certain conditions.

closed 2017-09-22 by chamnit