grbl/grbl #1288: weird home switch behaviour

stillcool40, 2017-09-23:
I am trying to wire up homing switches and am getting the strangest behaviour.

using the arduino serial monitor and the grbl ? status you manualy trigger the z it shows up fine.

neither the x or y show up on their own UNLESS the z is also triggered.......driving me nuts

i have simplified the wiring to ground and switch straight to the pins

jahnj0584, 2017-09-23:
Did you read through the config.h file? It's explained there

stillcool40, 2017-09-23:
well i read through it but am failing to understand why  the z needs to be triggered as the homing cycle backs off the z switch to untrigger it before starting the other axis, which, because z is not triggered, just smash into the endstops
. im new to all this , can someone explain it simply and give hint as to how to fix it

vMeph, 2017-09-23:
What cnc shield you using and grbl version?

stillcool40, 2017-09-23:
its a very simple prototype shield nothing clever
i think its grbl9

carneeki, 2017-11-06:
@stillcool40 on most machines, the Z will retract up to home first. This is because if grbl is restarting after a power failure / software crash / similar, we don't want to move X/Y as it may cause a cutter to crash into a part. Retracting Z usually prevents that. Once Z is triggered, it's considered homed, will back off, and the other axes can home now.