grbl/grbl #1289: G Code causing z axis to alarm hard limit.

Preston4, 2017-09-25:
I am brand new to this whole CNC environment but am setting up my school CNC router. We are using the hx8410. I am using fusion 360 to export g code. when I try to send the code to the CNC, it says invalid command: and lists a few lines. It specifically says: 
Skipped Command #1
Skipped Command #2 
Skipped Command #3 
Skipped Command #9 
Skipped Command #6952

error: Unsupported command
>>> Z5
>>> Z2.976
>>>G1Z2.5F1000 

the g code that exports start with : 
%
(TEST 3)
(T2  D=3.175 CR=0 - ZMIN=-20 - flat end mill)
G90 G94
G17
G21
G28 G91 Z0
G90

(2D Adaptive2)
M9
T2 M6
S10000 M3
G54
M9
G0 X36.503 Y-36.369
Z15
Z5
Z2.818
G1 Z2.5 F762
G3 X38.497 Y-38.631 Z2.335 I0.997 J-1.131

If anyone could help, it would be massively appreciated.

X3msnake, 2017-09-25:
What post processor are you using?

Preston4, 2017-09-25:
I'm just using the built-in post processor in fusion 360. I don't know what its called.

X3msnake, 2017-09-25:
You must use the correct post processor you can try the one that ships with
fusion but it must say grbl or shapeoko

Preston4, 2017-09-25:
[image]

in the post process, it says I can use generic grbl

X3msnake, 2017-09-25:
what sender are you using?

Preston4, 2017-09-25:
universal g code sender

X3msnake, 2017-09-25:
what is the problem?

the skipped comands ate the % and the (commented text) so no fault here

when does the unsupported command appears and what dos it have to do with
the hard limit that you post on the title?

Preston4, 2017-09-25:
I think the wrong command is the z5 and stuff like that when I send the file, the z-axis goes all the way to the top and nothing happens

X3msnake, 2017-09-25:
can you control the machine manually within UGS ?

have you set the zero point on the part you want to machine and done a
reset of the xyz?

Preston4, 2017-09-25:
I can control the machine fine and can set zeros, I just can't use cut the shape I want.

X3msnake, 2017-09-25:
Try the Carbide 3D (grbl) post processor on the dropdown list

X3msnake, 2017-09-25:
also try manually removing from file

 G28 G91 Z0

save and run the code again

2017-09-26 0:38 GMT+01:00 Vinicius Silva <x3msnake@gmail.com>:

> Try the Carbide 3D (grbl) post processor on the dropdown list
>
>
>> I can control the machine fine and can set zeros, I just can't use cut
>> the shape I want.
>>
>> <https://github.com/grbl/grbl/issues/1289#issuecomment-332042595>, or mute
>> the thread
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

Preston4, 2017-09-25:
I'm sorry, but where do I find the carbide 3d?

X3msnake, 2017-09-26:
try the second test first

Preston4, 2017-09-26:
I'm not by the CNC right now, but ill work on it tomorrow, thank you very much.

X3msnake, 2017-09-26:
if the removing G28 works

set the properties of GRBL post processor like this useG28 > NO

[image: Imagem intercalada 2]

What is probably happening is that the machine is trying to go to machine
zero (g28) and hitting the endstops

2017-09-26 1:01 GMT+01:00 Vinicius Silva <x3msnake@gmail.com>:

> try the second test first
>
>
>> I'm sorry, but where do I find the carbide 3d?
>>
>> <https://github.com/grbl/grbl/issues/1289#issuecomment-332045763>, or mute
>> the thread
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

langwadt, 2017-09-26:
or rebuild grbl to set zero after pull-off from the switches

X3msnake, 2017-09-26:
@langwadt - Please correct me if i'm wrong, but won't that mean it will write to the flash memory every time it does a homing sequence?
And if so won't this mean one will be burning out life cycles out of that flash memory?
Or am I missing something here?

Preston4, 2017-10-10:
Ok so I got the code working and everything was going ok, till the maschine reached its hard limit, and kept going and made a terrible noise. The motors got so hot it hurt to touch them. I am unable to home the maschine, as it is past its limits.

vMeph, 2017-10-10:
Machine should stop if hits the switches if doesnt, and dont homes most like is not enable in the grbl configs