grbl/grbl #1292: Spindle active when eStop pushed

CLMNTDSGNR, 2017-09-28:
Hi all,

I have a small problem, my spindle starts turning when I press the eStop button. Surely because the eStop is wired to reset (and change the state of relay pin). How to solve this problem? It is a relay without octocoupler: / (I ordered one). With M3 and M5, everything works well.

Thank you.

Norwegian-Gadgetman, 2017-09-29:
Okay...  

You're aware that 'eStop' means EMERGENCY Stop? 
It's not supposed to be wired to the Reset pin, it's supposed to go on the POWER lines. 

Mine cuts power to the spindle and the motor drivers. 
(This is a bare minimum of what the eStop button must do.)

Cutting all power to the ShapeOko is pretty common.

On Professional machines it may even actively lock motors in order to bring the machine to a stop even faster.

Evil-Nemesis, 2017-11-28:
> I'd like to control my CNC without my computer, just for simple cuts and drills holes. So, I'm using a 5x5 matrix keypad with another arduino do control, before that I used 6 buttons to move the axis.

so just send G-codes to serial pins by another Arduino!
for example, to move it to left (X-axis) by 10mm send this codes:
G91
G0X10Y0
G90