grbl/grbl #1293: Potential loss of steps

adichell, 2017-09-29:
Hi, 
I was analyzing the stepper code and in the st_prep_buffer function I think there may be a problem that may cause a loss of steps.
If I am not wrong, in this function the segments for the current block are buffered, and then the mm_remaining variable is updated. 
Suppose we are executing the last step segments of a block, this function plans them, then it updates the mm_remaining and at the end of the function it enters into the else branch of the check:
`if (mm_remaining > 0.0)`
In this case, the current block is discarded and the last steps are not executed.

This does not happen usually since the processor speed and the number of operation in between the update of the mm_remaining variable and this check is such that there is enough time for the timer interrupt to be triggered and for the steps to be executed.
But if you will change the processor (speed) and/or optimize the code, this issue may present.

I'd suggest to add a check in the else branch and eventually a timeout like this:

`else 
{ // End of planner block
    	// Check if there is any segment pending to be executed before discarding block
    	volatile int index = 0;
    	while((segment_buffer_tail != segment_buffer_head) && index < TIMEOUT_COUNTER){index++;}
        // The planner block is complete. All steps are set to be executed in the segment buffer.
        pl_block = NULL; // Set pointer to indicate check and load next planner block.
        plan_discard_current_block();
      }`

Best regards,
A.

chamnit, 2017-09-29:
No. You’ve misread the code. The step segment is set to be executed when its head pointer is updated prior to the mm_remaining line. Planner blocks are closed after the final steps are placed in the segment buffer. They are always executed.

closed 2017-09-29 by chamnit

adichell, 2017-09-29:
I agree with "The step segment is set to be executed when its head pointer is updated prior to the mm_remaining line". It is set to be executed, and then the stepper ISR is triggered, it sees that there are steps to be executed and it executes, normally.

But imagine the processor executes the code quite fast to arrive to set mm_remaining to 0 (I am talking of the last steps) and it gets to the check before the ISR is triggered. What would happen?

chamnit, 2017-09-29:
The steps would sit in the step segment buffer until the ISR comes around to service it. I'm failing to see what you think is a problem. I think you are misinterpreting the code. The planner buffer is independent of the step segment buffer. Everything in the step segment buffer is always executed.

adichell, 2017-09-29:
Yes, you are right about this, the steps would sit in the step segment buffer, I was wrong about this, you would not lose it. 
The effect would be the system to go in idle, and then these steps would be executed during the execution of the next planned block. 
I still think this is a potential issue, but different from losing steps.

Best regards,
A.

chamnit, 2017-09-29:
No. It would not go into idle either. The timing of the step segment generator is designed to only require it to not starve. A faster processor will only help. In the future, please try to bring up issues that are related to trackable bugs, not misinterpretations or questions about the code. Grbl runs reliably on tens of thousands of CNC machines with a wide breadth of job types and lengths. There have been no reported issues of step loss in several years. Keep that in mind.