grbl/grbl #1297: Disable z homing

mogey336, 2017-10-03:
How can i disable z homing after a click on $H, I need X and Y z axis only

jahnj0584, 2017-10-03:
You edit the config file for 2 axis homing

mogey336, 2017-10-03:
@jahnj0584  how can i edit it?

jahnj0584, 2017-10-03:
Open config.h with a text editor then reflash the arduino

mogey336, 2017-10-03:
@jahnj0584 which line do i have to change?

jahnj0584, 2017-10-03:
Open and look yourself. It's not hard. I personally am not here to walk you
through it

X3msnake, 2017-10-04:
https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#how-do-i-configure-my-homing-cycle-to-just-do-2-axes-or-i-have-a-pen-plotter-and-dont-have-a-z-axis

aforww, 2017-10-25:
Wow, someone doesn't appreciate the importance of teaching.

winson2769, 2019-01-21:
hi where can i find the config.h file?

Rob2029, 2019-01-21:
In the repository on github for the grbl, all the files are there

winson2769, 2019-01-21:
pls directly show me the url of config.h ?i am a new to grbl, thanks

Rob2029, 2019-01-21:
https://github.com/grbl/grbl/tree/master/grbl
You should learn how to  navigate thru the github page, there are so much
information and cool stuff there

109JB, 2019-01-22:
The config.h you need to edit is in the grbl folder that you downloaded from github.