grbl/grbl #1298: Does grbl support Touch plate

mogey336, 2017-10-04:
How can i activate touch plate with grbl? With out typing a code can i use manually z aixs with NO connection to make touch plate

mogey336, 2017-10-04:
I mean manually click on home z axis

carneeki, 2017-11-06:
I think you need a connection between the Arduino and the touch plate. Probes and touch plates work based on an electrical path so they can provide feedback to grbl. Without it grbl can't really 'know' when contact is made.

As for the no programming part, checkout UGS by Will Winder, there's a Probe Helper in the GUI that can be used for touch plates.
https://github.com/winder/Universal-G-Code-Sender/issues/584#issuecomment-290307059