grbl/grbl #1301: Grbl not flashing correctly

grabc1001, 2017-10-09:
Hi

I'm new. 
Windows10 64
Installed Latest Arduino Ide. Flashed the Arduino with the latest Grbl. I'm not getting correct Grbl welcome message in the Serial monitor just a few random characters. (Correct port is selected and Baud is set to 112500)

I've tried flashing Grbl to the Arduino several times in an attempt to sort this. Could that be making the matter worse? If someone can advise on how to copy the orange readout from the ide window I can add that readout.

Would appreciate any advise.

jahnj0584, 2017-10-09:
Go to the grbl wiki and follow the instructions exactly

grabc1001, 2017-10-14:
Hi
I believe I have followed the wiki to the letter. I have also cleared the eeprom and tried to reflash the uno with the latest grbl.   When I open up the serial monitor I just get differing random characters each time.
I realise this forum may be regarded as too technical for new people so could you direct me to an appropriate forum. Would appreciate the help.

jbidinger, 2017-10-14:
Sounds like you may have a baud rate mismatch. Some documentation says that it should be 9600 but I found that newer grbl uses 115200 by default.

grabc1001, 2017-10-14:
Thanks for that, I knew baud should be 115200 but missed setting it in the serial monitor. That is now done but after typing $ for the Help list the output I get is not the list but this:

Grbl 1.1f ['$' for help]
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
ok

Can you let me know what this means. Does this mean grbl isn't set up right or am I missing something again. Also how to display the Help list properly.
Gonna get this pen plotter up and running even if it kills me... (it does seem like it's trying to kill me by the way)

chamnit, 2017-10-14:
Try the wiki on the Grbl v1.1 website. Not this one.

grabc1001, 2017-10-14:
Hi Chamnit... I see you guard your knowledge.....How did you start leaning this?

 thanks..I'll move on

closed 2017-10-14 by grabc1001

chamnit, 2017-10-14:
@grabc1001 :  No. I’m not guarding anything. Everything is written in the wiki pages and saves everyone the time of having to reply to every user requests for help. Please do us all a favor and try to read it before posting. It takes a lot of time out of our days to write and handle responses that we all do in our free time to service the community.

chamnit, 2017-10-14:
Also I meant “not this one” as in the Grbl v1.1 website. Not the Grbl v0.9 website where you posted. There is a link on the front page.