grbl/grbl #1303: Speed too high

Suuniil, 2017-10-15:
I'm building a laser engraver with grbl 0.9j but my laser diode is just 1W, so i have to move very slowly. My problem is, when i run something (I'm using the UniversalGcodeSender v1.0.5) like G1 X-20 F5 it goes to 2cm in just around 5 seconds. It doesn't seem like going lower with the speed helps, so why is it moving so fast? When I home, the Axis moves much slower at the set 50 mm/min, but when i use G1 like above with 5 mm/min it is much much more faster. Is there a speed minimum, and if there is, how can I get around it? I attached my grbl settings.
[GRBL Settings.txt](https://github.com/grbl/grbl/files/1385360/GRBL.Settings.txt)

Labruzketa, 2017-10-15:
You need to modify the settings, **$110**, **$111**, **$112**, for the desired max speed in mm/min. Below is a description of the configuration.

**$110, $111 and $112 – [X,Y,Z] Max rate, mm/min**

This sets the maximum rate each axis can move. Whenever Grbl plans a move, it checks whether or not the move causes any one of these individual axes to exceed their max rate. If so, it'll slow down the motion to ensure none of the axes exceed their max rate limits. This means that each axis has its own independent speed, which is extremely useful for limiting the typically slower Z-axis.

The simplest way to determine these values is to test each axis one at a time by slowly increasing max rate settings and moving it. For example, to test the X-axis, send Grbl something like G0 X50 with enough travel distance so that the axis accelerates to its max speed. You'll know you've hit the max rate threshold when your steppers stall. It'll make a bit of noise, but shouldn't hurt your motors. Enter a setting a 10-20% below this value, so you can account for wear, friction, and the mass of your workpiece/tool. Then, repeat for your other axes.

NOTE: This max rate setting also sets the G0 seek rates.

Sorry for my poor English.

Suuniil, 2017-10-16:
Yes, i know that. But if I try moving very slowly, grbl moves too fast, so my question is, is there sime value that sets the MINIMUM speed?

jahnj0584, 2017-10-16:
It's on the wiki. I think it's 1mm/s is the slowest grbl will send without
editing the config

chamnit, 2017-10-16:
@Suuniil : Your Grbl $$ settings are unusual. I would start by making sure that they are properly configured. For example, your step pulse time is very long (should be around 10), while your axis settings are haphazard.

Suuniil, 2017-10-16:
@chamnit the axis settings are all right - I was just playing around with the step pulse, but it was just the same before

Suuniil, 2017-10-16:
@jahnj0584 I'll try that as soon as I can, but sadly that's in at least one week or later.

chamnit, 2017-10-16:
By default, Grbl has a minimum step rate of around 240 steps/second. This is due to how the adaptive step smoothing algorithm works and the limited resolution of the AVR 16-bit timer. Your X-axis 10 steps/mm setting is causing it to hit this limit. If you can't increase this by upping your micro steps, then disable AMASS in config.h. You can go slower to about 4 steps/second.

closed 2017-10-16 by chamnit

langwadt, 2017-10-16:
wouldn't it be easy to make the prescaler a compile time option and change TICKS_PER_MICROSECOND accordingly for those special cases of systems that need to run very slow?

chamnit, 2017-10-16:
@langwadt : Yes it would be fairly simple to do, but this isn't a common problem for properly configured CNCs. Also, you can just disable AMASS to get to absolute minimum speed. The benefits of AMASS reduce significantly as you reduce the pre scalar. So, there isn't that much reason for having in-between settings. Plus, I don't want to have to support them.