grbl/grbl #1304: my stepper motor don't turn on high speed

uung, 2017-10-16:
please help me, i have arduino uno + cnc shield, and 20A/24v power supply with 3 stepper motor. but my stepper motor can't turn quickly, but it release nois(buzzing). i hope you  can tell me wath is happen.

chamnit, 2017-10-16:
Buzzing is a sign of losing steps. Your system is misconfigured.

closed 2017-10-16 by chamnit

vMeph, 2017-10-16:
Did you adjust trimpot on your drivers for your motor?

uung, 2017-10-17:
@chamnit how is to to configure my system so it can work normally and not release buzzing noise... sory, i am a newbe...

uung, 2017-10-17:
@vMeph yes, i have adjust trimpot to maximum value...

uung, 2017-10-17:
@vMeph trimpot that i adjust is on suplly power, not on the driver

ArjenD, 2017-10-17:
your acceleration for the motor is probably set too high.

vMeph, 2017-10-17:
So you got to adjust trimpot on drivers if you have drv 8825 or a4988 if they set to low or to high will afect motos, search on you tube on how to do it

uung, 2017-10-17:
@ArjenD if max acceleration motor set to minimum value it can work slowly. but buzzing and vibration too high, i fear it can affect to stability and durability of other component.

uung, 2017-10-17:
@vMeph okay i will do it, but what is the trimpot referred to as shown in the picture below?

uung, 2017-10-17:
@vMeph okay i will do it, but what is the trimpot referred to as shown in the picture below?
[image]

X3msnake, 2017-10-17:
to Ground Pin, bottom right

[image: Imagem intercalada 1]
[image: Imagem intercalada 2]