grbl/grbl #1308: GRBL Z axis homing delay

aforww, 2017-10-25:
So, normally I'll do my own problem solving but this one has me stumped. I just upgraded to the latest GRBL and installed a new Protoneer CNCShield V3.51. My issue comes when I do a homing cycle. There is a 2-3 second delay between the switch activating (indicated by the LED on the switch) and when GRBL and UGS recognize that it's been triggered. This issue is only on the Z axis. 

For reference, the switches are hall effect switches. They have an integrated LED on each switch which indicates it's being triggered. If anyone has any ideas, please throw em this way.

ariefadha, 2017-10-26:
have you probe the end cable that connect to the board with multimeter or scope? try rotate your hall effect switches to X axis, if same problem occurred, it might connector problem on your Z axis (might be voltage drop)

aforww, 2017-11-01:
ariefeda, I did swap them and the problem reproduces itself so, it's either something with the CNC shield v3.51 or it's a problem with the RJ11 breakout board I'm using for the switches, I know this because there's a delay on every switch and wire if it's plugged into a specific port on the breakout board.