grbl/grbl #1309: My stepper motor is not moving

fznmdhsn, 2017-10-28:
As we are checking our electronics of cnc machine, I will explain what I did or doing till now
I connected arduino uno with computer and load hex file successfully in it by x loader then i will done the connection from arduino uno to motor board tb6560 and then from board to stepper board and 12v DC supply to the motor board and then I run .nc file into ugs v0.9 and grbl controller but the motor doesn't show any movement. Please help us

With regards 
Fznmdhsn

fznmdhsn, 2017-10-28:
I mentioned that my stepper motor is nema18

jahnj0584, 2017-10-28:
You have the wiring wrong then

fznmdhsn, 2017-10-28:
Can I send you my wiring sir

fznmdhsn, 2017-10-28:
[image]

fznmdhsn, 2017-10-28:
[image]

fznmdhsn, 2017-10-28:
[image]

fznmdhsn, 2017-10-28:
Now I run an arduino code of stepper motor speed, so it's giving the pulse and the spindle is vibrating... But still from grbl, it's not showing any movement

vMeph, 2017-10-28:
Did you config grbl?
What settings you have

agjupiter, 2019-10-23:
Hi guys ..I have also such problem plz help me...plz plz.plz...
I have made a cnc pen plotter using arduino uno, cnc shield v3.0, nema 17 stepper with drv8825 driver..and use gebl v0.8c/ v0.9i/and v0.9j...with baud rate 9600' and 115200...use ide to flash grbl and grbl controller for senfing g code...
The problem is when I upload the g code to board...it experience the vibrating and unpleasant sound but the motor is not moving but in visualizer it shows miving...

I have adjusted vref to .8 amps as the rated current is 2amps....
I m providing 12v with 1amps ....
I tried changing the wire of motor....
Plz help me to solve the problem...

JesConsuegra, 2019-10-23:
Usually this happens when the motor wires are swapped, please check the motor cabling.