grbl/grbl #1312: New to GRBL - Constant ALARM:1 Issue but not Electrical issue when trying to Home.

aqibi2000, 2017-11-06:
I have a 2.5D machine (Z axis is a solenoid).

Disabled Z axis homing in the Config library

With homing switched off I can move the X and Y axis using commands such as: G91 G0 X-1

With homing enabled $21 =1 and $22=1 it moves 1mm towards home then halts stating ALARM:1

It's really annoying me and cannot figure out what is the cause. Please help.

X3msnake, 2017-11-06:
If you leave 21 disabled and enable only 22 does it do the same?

aqibi2000, 2017-11-06:
> If you leave 21 disabled and enable only 22 does it do the same?

I get Alarm:8 with 21 disabled

X3msnake, 2017-11-06:
@hdrider465 <https://github.com/hdrider465> : "Alarm" state is not an
"Idle" state. Homing upon power up sets an alarm, as well as a failed
homing cycle. Make sure you are editing the config.h file inside the
Arduino library folder, not your downloaded folder. The import makes a copy
there

https://github.com/gnea/grbl/issues/87

X3msnake, 2017-11-06:
How are your homing switches set up nc/no ?

aqibi2000, 2017-11-06:
> How are your homing switches set up nc/no ?

They are Normally Open switches, mounted at the limits of the X and Y axis (X = pin 8, Y = Pin 9). They are 4 in total thus the respective switches are paralleled together, with the appropriate supporting hardware to ensure it is pulled up to the rail voltage. I'm certain this isn't an Electrical issue.

In order to disable the Z homing I have done the following in the config file:
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  
#define N_HOMING_LOCATE_CYCLE 1 // Integer (1-128)

X3msnake, 2017-11-06:
just for sanity check try inverting the actual value of $5 setting a see if
it throws the same error

aqibi2000, 2017-11-06:
> ust for sanity check try inverting the actual value of $5 setting a see if
it throws the same error

I appreciate your help but the switches are working I can assure you.

X3msnake, 2017-11-06:
How do I configure my homing cycle to just do 2-axes? Or I have a pen
plotter and don't have a Z-axis.

Configuring Grbl for a 2-axis CNC machine is easy. The config.h file
contains a lot of compile-time options that you can enable or disable based
on particular needs, including altering the homing cycle to only perform it
on the X and Y axes. There are instructions and descriptions for each
option available, but some users may not be familiar with C programming,
comments, and macros. So, here's how to do it. You'll see two lines in the
config.h file like so:

#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First
move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then
move X,Y at the same time.

Change these two lines to look like:

#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)

That's it! Re-compile and re-flash Grbl via the uploading procedure
outlined in the Grbl Wiki.

https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#how-do-i-configure-my-homing-cycle-to-just-do-2-axes-or-i-have-a-pen-plotter-and-dont-have-a-z-axis

aqibi2000, 2017-11-06:
> How do I configure my homing cycle to just do 2-axes? Or I have a pen
plotter and don't have a Z-axis.

Configuring Grbl for a 2-axis CNC machine is easy. The config.h file
contains a lot of compile-time options that you can enable or disable based
on particular needs, including altering the homing cycle to only perform it
on the X and Y axes. There are instructions and descriptions for each
option available, but some users may not be familiar with C programming,
comments, and macros. So, here's how to do it. You'll see two lines in the
config.h file like so:

#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First
move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then
move X,Y at the same time.

Change these two lines to look like:

#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)

That's it! Re-compile and re-flash Grbl via the uploading procedure
outlined in the Grbl Wiki.

https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#how-do-i-configure-my-homing-cycle-to-just-do-2-axes-or-i-have-a-pen-plotter-and-dont-have-a-z-axis

Thanks for the reply. I have experience in C programming, changing the code alters the logic of the homing operation to perform X axis alignment then Y, compared to performing both simultaneously. It still yields the same end results: 'ok ok' even though it is not near the limit switch.

The machine moves the same distance (mm) as stipulated by $27, very strange behaviour

chamnit, 2017-11-06:
@aqibi2000 : Please post your $$ settings and $I build info output. This will help diagnose your issue.

aqibi2000, 2017-11-06:
> @aqibi2000 : Please post your $$ settings and $I build info output. This will help diagnose your issue.

The machine is not a CoreXY or anything like that, just a conventional machine with a moving gantry. The closest left corner is 0,0.

$I=
[VER:1.1f.20170801:]
[OPT:V,15,128]

$$ =
$0=10
$1=255
$2=0
$3=4
$4=0
$5=0
$6=0
$10=16
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=500.000
$26=100
$27=2.000
$30=1000
$31=0
$32=0
$100=160.000
$101=160.000
$102=250.000
$110=5000.000
$111=5000.000
$112=5000.000
$120=10.000
$121=10.000
$122=10.000
$130=315.000
$131=304.000
$132=30.000

$ = [HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]

chamnit, 2017-11-06:
0,0 should be the rear right corner, not left front. Grbl operates and sets the machine coordinates in the negative coordinate space. See the wiki as to why.

closed 2017-11-06 by chamnit

aqibi2000, 2017-11-06:
> 0,0 should be the rear right corner, not left front. Grbl operates and sets the machine coordinates in the negative coordinate space. See the wiki as to why.

I've corrected that positioning via $3 = 7, now rear right is the 0,0 point.

But that wouldn't correct the homing movement issue

aqibi2000, 2017-11-07:
Why has my thread been closed?

[image]

X3msnake, 2017-11-07:
this is not a GRBL bug

the tread is closed but you still get support