grbl/grbl #1313: #1312 was my post but has been closed strangely. Machine will not HOME only moves 10mm then states ok.

aqibi2000, 2017-11-07:
I have a 2.5D machine (Z axis is a solenoid). Not an Electrical issue, the limit switches are supported with appropriate hardware, and verified with Oscilloscope there is no false triggering and wired correctly to the X and Y limit pins.

Disabled Z axis homing in the Config library

Before I press home:
<Idle|WPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>

After pressing home:
<Idle|WPos:-2.000,-2.000,0.000|FS:0,0|Ov:100,100,100>

As you can see the -2mm displacement follows the parameter $27, yet the machine hasn't reached the limits yet nor did it reinitialise to 0,0 if it has reached 'home'.

$I=
[VER:1.1f.20170801:]
[OPT:V,15,128]

$$ =
$0=10
$1=255
$2=0
$3=7
$4=0
$5=0
$6=0
$10=16
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=500.000
$26=100
$27=2.000
$30=1000
$31=0
$32=0
$100=160.000
$101=160.000
$102=250.000
$110=5000.000
$111=5000.000
$112=5000.000
$120=10.000
$121=10.000
$122=10.000
$130=315.000
$131=304.000
$132=30.000

$ = [HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]

I have verified that the switches are correctly wired to pin 9 and 10.
[image]

chamnit, 2017-11-07:
Please don't open multiple threads. Your problem is a misunderstanding of the coordinate frame. Your origin is correct, but the directions are not. As I've said in your other thread, Grbl operates in NEGATIVE machine space, not positive. Read the FAQ wiki page.

closed 2017-11-07 by chamnit

aqibi2000, 2017-11-08:
@chamnit 

I appreciate I have the movement direction incorrect and thus will not actually correctly home ever, but however I feel the root issue if beyond this. Please see me below explanation.

> The Sequence s:
> X and Y axis
> X, Y Axis move both to Homing direction at fast rate ($25)
> The first Axis triggers the switch will stop and wait for the second axis to trigger
> When second axis triggers the switch, both axis back off a distance ($27)
> Both X and Y axis will move toward switches again slowly, until both switches triggered again ($24)
> Both X and Y axis will back off a small distance ($27)

> https://github.com/gnea/grbl/wiki/Set-u ... ming-Cycle

My machine after pressing $H the machine just moves the distance $27 on both X and Y once, which means the program is not processingthe homing sequence because it is meant to perform the triggering twice per axis (High speed movement to trigger the axis, then the low speed movement to perform precise homing).

chamnit, 2017-11-08:
You have noise in your limit switches that is causing the fast search cycle to be skipped. There is a wiki page on strategies for wiring them and protecting them from EMI. Please try to understand that Grbl reliably runs on tens of thousands of CNCs. The homing cycle works fine. 99% of the time, it's user error, often due to either misunderstanding how it works or electrical noise.

109JB, 2017-11-08:
Or, $5 is not correct. The below is from your post with added bold comments to what would happen if the $5 is wrong

The Sequence s:
X and Y axis
X, Y Axis move both to Homing direction at fast rate ($25) **Won't happen because Grbl believes the limits are already tripped**
The first Axis triggers the switch will stop and wait for the second axis to trigger **Grbl believes both are triggered, so this won't happen either**
When second axis triggers the switch, both axis back off a distance ($27) **Grbl does this because during this move limits are ignored**
Both X and Y axis will move toward switches again slowly, until both switches triggered again ($24) **Won't happen because Grbl believes the limits are already tripped**
Both X and Y axis will back off a small distance ($27) **Grbl does this because during this move limits are ignored**

So, if $5 is not correct you will get 2 movements totaling 2*($27 value)

Also, when homing is done the MPOS is reset to where 0,0,0 is the position where the switch activates, not the pull off position.

109JB, 2017-11-08:
Actually, I just tested it on my machine and here is what happens.

If $5 is wrong, the first defined axis/axes,( in my case Z, and in your case XY), will do the following:

$H sent
Z-axis moves the $27 pull off amount
Z-axis stops and an alarm is triggered

If the first defined homing axes are both X and Y, I would expect X and Y to move simultaneously by $27, stop, and then trigger an alarm. Are you getting an alarm message after attempting homing?