grbl/grbl #1315: Z axis not homing

allstar87, 2017-11-08:
Another guy where the Z axis isn't working.  GRBL 1.1 with Protoneer CNC shield.  All end stops functioning normally; if I drive the Z axis against the stop by jogging, it throws a proper error.  During the homing cycle, it acts as if there is no Z axis; X and Y axes begin homing.

Config.h has 
```
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
```
both in place and uncommented.  Attempted to unplug the end stop from the Arduino but no change in behavior.  $$ config is as follows:
 $0=10
 $1=25
 $2=0
 $3=0
 $4=0
 $5=0
 $6=0
 $10=0
 $11=0.010
 $12=0.002
 $13=0
 $20=0
 $21=1
 $22=1
 $23=0
 $24=100.000
 $25=800.000
 $26=250
 $27=4.000
 $30=255
 $31=0
 $32=1
 $100=80.000
 $101=80.000
 $102=4000.000
 $110=10000.000
 $111=10000.000
 $112=500.000
 $120=600.000
 $121=600.000
 $122=80.000
 $130=380.000
 $131=300.000
 $132=50.000

chamnit, 2017-11-08:
Check your wiring on Arduino pins D11 and D12. Grbl swapped these pins (z-axis limit, spindle enable) several years ago and some CNC shields are not designed for this. This was done to access the hardware PWM for spindle speed on pin D11. 

Also thanks for sharing your settings. Your z axis step/mm is unusually high and combined with your 500 mm/min max rate, you exceed Grbl's max 30kHz step rate. Lower either one and ensure you are below this threshold. Otherwise Grbl's processor will be over taxed and Grbl will become unstable.

closed 2017-11-08 by chamnit

allstar87, 2017-11-08:
Umm, issue's still open.  It's not the wiring of the shield.  If I jog the axis against the end stop, it correctly throws a hard limit.  Z steps/mm is high because that motor is driving an M5 threaded rod; I'll lower the mm/min setting.

chamnit, 2017-11-08:
It's closed because it's not a software bug. It's a implementing problem. What does the '?' status report say when you are holding the z-axis limit switch down and when you release it? Also test continuity of the z-axis limit to pin D12.

allstar87, 2017-11-08:
Roger.
Continuity from Z limit switch to pin D12 is correct.  Switch is wired Normally Open to ground, as per Protoneer's instructions. 
Sending a ? gets no response except an ok.

allstar87, 2017-11-08:
Check that; my console was giving me an issue.  When depressed, I get a PinZ, when released, no PinZ.

chamnit, 2017-11-08:
Does the z-axis move at all during the homing cycle? Like 4mm?

Did you have any prior Grbl installations that altered the homing cycle?

Protoneer, 2017-11-08:
What version shield do you have?

allstar87, 2017-11-08:
The Z axis does not move at all in response to a $H command.  The shield is V3.51.

chamnit, 2017-11-08:
The only way this can happen is if you are installing a modified version of Grbl that disables the z-axis homing. Check and post your `config.h` that is located in the grbl folder in your Arduino library. Not elsewhere.

allstar87, 2017-11-08:
Config.h, located in ~/sketchbook/grbl-master/grbl:
[config.h.zip](https://github.com/grbl/grbl/files/1454830/config.h.zip)

chamnit, 2017-11-08:
Wrong file. In your Arduino library. On Mac, it's in ~/Documents/Arduino/libraries/grbl

allstar87, 2017-11-08:
It's a Linux system; ah, wait.  Was the wrong file anyway.  ~/sketchbook/libraries/grbl had the Z axis line commented out.  For some reason I've got two copies of the firmware installed in broadly similar places.  Got it now.  Thank you so much for your help.

lee1933, 2018-04-13:
Hi Allstar87,

I am having a similar problem where my X and Y axis homes but not the Z axis. Could you show me where I need to uncomment the line for Z-axis homing? Thanks

chamnit, 2018-04-13:
@lee1933 : Z-axis is homed by default. Make sure that the z-limit is connected to pin D12. Not D11. This switched a few years back and some older and popular shields haven't updated this.