grbl/grbl #1316: Stuck at waking GRBL via Python

selotipkusut, 2017-11-09:
Hello, I'm working on a simple Python 3 GUI for a GRBL machine made by a previous student.

I'm struggling to initialize GRBL using a modified simple_stream.py, where 2 serial ports are opened. ttyACM0 is to configure the motor step mode and ttyUSB0 is to communicate with GRBL.

The ttyACM0 has no problem, but I got no feedback from GRBL. No startup message such as _Grbl v0.9j press $ for help_, etc.
Meanwhile, communicating with both ports using Arduino IDE's serial monitor works without problem, G-Codes were sent properly.

`##
import serial
import time

# Open grbl serial port
s = serial.Serial('/dev/ttyUSB0',115200)
step=serial.Serial('/dev/ttyACM0',9600)

# Open g-code file
f = open('test.gcode','r')

# Wake up grbl
s.write("\r\n\r\n".encode())
time.sleep(2)   # Wait for grbl to initialize
s.flushInput()  # Flush startup text in serial input

step.write("f".encode())
print((step.readline()).decode())

# Stream g-code to grbl
for line in f:
    l = line.strip() # Strip all EOL characters for consistency
    print('Sending: ' + l)
    s.write((l + '\n').encode()) # Send g-code block to grbl
    grbl_out = s.readline() # Wait for grbl response with carriage return
    print(' : ' + (grbl_out.strip()).decode())

# Wait here until grbl is finished to close serial port and file.
input("Enter to exit and disable grbl.")
f.close() #close file

step.write("s".encode()) #Exit step mode
print((step.readline()).decode())
`

winder, 2017-11-09:
Grbl only has one serial interface, why do you think you need two devices?

109JB, 2017-11-29:
I guess I don't understand what you mean by "wake" grbl.  Grbl is running anytime the Arduino is powered so there is nothing to wake.  If you are looking for the startup script then you need to send the soft reset command Ctrl-X.

cagordon2espe, 2020-05-08:
Hola, veo que utiliza algo similar al "Simple g-code streaming script for grbl", presenté el mismo problema y de antemano, encontré los cambios para python3...(ref.: [https://github.com/grbl/grbl/issues/1316#issue-272589073][1])

    import serial
    import time
    
    # Open grbl serial port
    s = serial.Serial('COM7',115200)
    
    # Open g-code file
    f = open('corte.gcode','r')
    
    # Wake up grbl
    s.write("\r\n\r\n".encode())
    time.sleep(2)   # Wait for grbl to initialize 
    s.flushInput()  # Flush startup text in serial input
    
    # Stream g-code to grbl
    for line in f:
        l = line.strip() # Strip all EOL characters for consistency
        print( 'Sending: ' + l)
        s.write((l + '\n').encode()) # Send g-code block to grbl
        grbl_out = s.readline() # Wait for grbl response with carriage return
        print( ' : ' + (grbl_out.strip()).decode())
    
    # Wait here until grbl is finished to close serial port and file.
    input("  Press <Enter> to exit and disable grbl.") 
    
    # Close file and serial port
    f.close()
    s.close()

Seguido, al correrlo aparecerá lo siguiente en el terminal:

> PS D:\GCODE_BUSCAR\grbl_python> & C:/Python37/python.exe d:/GCODE_BUSCAR/grbl_python/simple_stream.py 
> 
> Sending: G90  : error: Alarm lock  
>
> Sending: M5  : error: Alarm lock 
>
> Sending: X4.8Y22.5  : error: Alarm lock

Esto a su vez, nos indica que GRBL necesita ser desbloqueado o en GRBL.... "$X".
Para lo cual podría ser que en el archivo .gcode o depende la extención, cuente en su encabezado $X

> PS D:\GCODE_BUSCAR\grbl_python> & C:/Python37/python.exe d:/GCODE_BUSCAR/grbl_python/simple_stream.py 
>
> Sending: $X  : [Caution:Unlocked] 
>
> Sending: G90  : ok
>
> Sending: M5  : ok
>

Espero que haya sido de ayuda.

  [1]: https://github.com/grbl/grbl/issues/1316#issue-272589073