grbl/grbl #1317: Can i attach joystick to control CNC using another arduino.

Catatau52, 2017-11-10:


X3msnake, 2017-11-13:
There are a couple of addon projects for GRBL out there i think there is a discussion or tow running at the moment here on github

check it out on the latest version, this version has been superseeded by v1.1 that is called gnea/grbl

https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#extending-grbl

Evil-Nemesis, 2017-11-26:
hi everyone
I think it is better to connect the joystick to your PC!
You can find a lot on the internet about it!

109JB, 2017-11-27:
> hi everyone
 I think it is better to connect the joystick to your PC!
 You can find a lot on the internet about it!

This is actually on my list of things I want to do, but slightly different. What I plan to do is hook a second Arduino with a MPG wheel to a second USB port that will "talk" to my gui. Then the gui will interpret what to do and send the appropriate commands to grbl.

Catatau52, 2017-11-27:
i'd like to control my CNC without my computer ,just for simple cuts and drills holes. So, I'm using a 5x5 matrix keypad  with another arduino do control, before that I used 6 buttons to move the axis.

109JB, 2017-11-27:
for use without computer then yes, you can absolutely use another Arduino to control the Grbl Arduino.  I would opt for an MPG type control.

Evil-Nemesis, 2017-11-28:
> I'd like to control my CNC without my computer, just for simple cuts and drills holes. So, I'm using a 5x5 matrix keypad with another arduino do control, before that I used 6 buttons to move the axis.

so just send G-codes to serial pins by another Arduino!
for example, to move it to left (X-axis) by 10mm send this codes:
G91
G0X10Y0
G90