grbl/grbl #1319: Closed loop PID control for spindle PWM

Harvie, 2017-11-13:
Hello,
i know this might seem bit off the scope of GRBL, since traditionaly people use external motor driver that handles PID control of spindle speed. However i wanted to ask if it would be possible to add basic support directly to GRBL.

That means having input pin interrupt that would be able to receive pulses from optical or hall sensor, measure time interval between them and tune spindle PWM accordingly. So it would send more PWM power to motor if measured RPM will get lower than set RPM.

PID parameters should be configurable as GRBL variables.

I am not sure if there's enough memory to add this, but it would be super useful.

I know that you will probably tell me to go f*ck myself, because this is major new feature and nobody likes adding such big features to established projects, but please think about it.

Otherwise i'd have to use separate arduino just to receive PWM from GRBL, compare it with RPM pulses and send modified PWM to motor driver. I don't see point of having another arduino just for this simple task.

chamnit, 2017-11-13:
It’s not about being unwilling to add more to Grbl. It’s primarily about how much flash and RAM is left on the 328p chip. It’s literally a hundred bytes of flash and a handful of bytes of RAM. The little flash that is left is reserved to for bugs only. 

Closed loop control of a spindle is already on the development list for the ARM version, which is due out soonish. Aka when I get time.

closed 2017-11-13 by chamnit

Harvie, 2017-11-13:
Fair enough.
BTW what about using optiboot to save some space?

whitetd, 2017-11-13:
personally think its a matter of wait till a more advanced chip is fully
supported, rather then cram more into the old one. time to move forward
instead of hanging with the old standard. and just a quick Thank you to
Sonny for your work on GRBL.

chamnit, 2017-11-13:
That is with optiboot and with certain critical configurations that need to be supported. You can remove the boot loader altogether to get another 0.5KB of flash, but that doesn't help the RAM problem and makes it 100x more difficult for users to use.