grbl/grbl #1323: How to turn off spindle in pause mode?

Evil-Nemesis, 2017-11-26:
hello there
I need to turn off the spindle when I pressed feed hold button!
how can I do it!?

jahnj0584, 2017-11-26:
You can't and shouldn't

chamnit, 2017-11-26:
With Grbl v1.1, you can use spindle overrides. With v0.9 and after, you can enable the parking motion that will make Grbl retract the spindle and turn it off automatically. It'll perform a re-enable and plunge when resumed. See config.h file.

closed 2017-11-26 by chamnit

Evil-Nemesis, 2017-11-28:
I actually do that!!

Evil-Nemesis, 2017-11-28:
I actually do that!!
for hardware button i do this:

Evil-Nemesis, 2017-11-28:
I actually do that!!
for hardware button I do this:
uncomment this line in config.h:
#define ENABLE_SAFETY_DOOR_INPUT_PIN

and for the software button I changed this:
`
		void BtnStopClick(object sender, EventArgs e)
		{
			Core.FeedHold();
		}`
to this
`
		void BtnStopClick(object sender, EventArgs e)
		{
			Core.FeedHold();
            Core.SafetyDoor();
		}`
and it's OK!