grbl/grbl #1324: GRBL on MEGA X,Y,and Z only one direction. Direction outputs on pins 27,28, and 29 never go high.

765timb, 2017-11-26:
GRBL on MEGA X,Y,and Z only one direction. Direction outputs on pins 27,28, and 29 never go high. Drivers reverse direction if I put 5 volts on them, MEGA 2560 outputs never change state when jogging with Universal G-Code Sender machine control page, they turn same direction when selecting + or - moves. Is there something else needed in code or setup to get this to function both directions. 
[image]

109JB, 2017-11-27:
You should be posting this in the Grbl-mega fork: https://github.com/gnea/grbl-Mega/issues
however, a quick look at the cpu_map.h file shows that the "official" fork uses pins 30,31, and 32 for direction pins, not 27, 28, and 29.

https://github.com/gnea/grbl-Mega/blob/edge/grbl/cpu_map.h

765timb, 2017-11-27:
Thank you, I I will try swapping pins tonight after work. The pictorial
Mega 2560 diagram shows using pins 27,28 and 29. I am just getting started
with ardunio's and GRBL building a CNC router. I appreciate you quick
response and guidance.

109JB, 2017-11-27:
Just checked the diagram and it sure looks like pins 30, 31, 32 to me.

https://github.com/gnea/grbl-Mega/wiki/Connecting-Grbl-Mega

765timb, 2017-11-27:
This is the diagram I used to wire up. I don't remember off hand which site
I got it from, but obviously it was incorrect.Thanks Tim

https://github.com/gnea/grbl-Mega/wiki/Connecting-Grbl-Mega