grbl/grbl #1327: Erratic operation of Uno, CNC Shield and Grbl using UGCS

dre3rd, 2017-12-06:
grbl - 1
Have new Xyltech CNC shield and Genuine Uno. Have 3 Laptops - Toshiba Satellite w/ Win 10 -- Dell Latitude w/ Win7 and Toshiba Satellite w/ Unbuntu Linux 14.04
The Toshiba with Win10 (upgraded from 7) has run a Shark CNC Router for 5 years with no issues whatsoever.  Also have Dell Optiplex running LinuxCNC that controls a Sherline Lathe w/Milling Column. 
This Kit I bought from XYLtech contained shield, 4-drv 8825, 3-Nema 17 motors, power supply, and misc. associated small parts, along with a Xyltech branded Uno that I did NOT use and chose to use my Genuine Uno.

My installation so far: Followed instructions as closely as I could. Used Dell laptop first. Set current limit to Vref =.85 as motors are 1.7A per Pololu video and written instructions from internet. Installed drivers and left at default full step. Uno and Shield talk to each other with no problems ?? that I know of. Using UGCS. Using the X Y and Z buttons set to a distance of 1 the motors will run back and forth. Any thing over 1 gives unpredictable results with motors moving from maybe 1/2 turn to several turns and then stopping with motor humming and vibrating. Proceeded to move to Toshiba with Win10 as this machine has had no problems with the New Wave Automation Shark CNC system.  Exact same scenario. Moved to the Toshiba with Ubuntu Linux 14.04. Exact same scenario.  Moved all back to the Dell and made one change. Went from full step to 1/8 step. Works pretty good, well almost. Since motors are not connected to any mechanical parts yet, I can only observe the shafts turning.  X Y and Z buttons for movement again with distance set to 1.  X moves about 1 1/2 turns Y and Z move about 1/4 turn. Below are my GRBL settings with all at default except $100 $101 $102 to follow along with the example I was using. With my limited knowledge of these settings (Shark required no config on my part and Sherline has a set of config files for their machine on the LinuxCNC CD that comes with system.) I do not understand why with the settings below the motors do not step the same distance. I have swapped driver positions and motor lead connections on shield. Problems do NOT follow physical parts. Also the DRO on the UGCS screen does not always show movement and the Return To Zero button works sporadically and sending a manual G0 x0 y0 z0 does not work 100 percent of the time. It does however work better when typed in on the command bar rather than putting it in one of the macro keys as that particular instruction will not move the motors, but another of the macro keys with G0 x100 y100 z100 works every time. That's my story and I'm sticking to it. I need help, please.

>>> $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=314.961 (x, step/mm)
$101=314.961 (y, step/mm)
$102=314.961 (z, step/mm)
$110=250.000 (x max rate, mm/min)
$111=250.000 (y max rate, mm/min)
$112=250.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=1000.000 (x max travel, mm)
$131=1000.000 (y max travel, mm)
$132=1000.000 (z max travel, mm)
ok

winder, 2017-12-06:
What version of UGS are you using? You'll need to use one of the 2.0 builds to get full compatibility with GRBL 1.1. GRBL 1.1 and UGS 2.0 is a combination that many people are using successfully.

Your large paragraph is a lot to process, but there are a few possible problems I was able to pick out:
* the acceleration might be too high (or the motor voltage too low), causing the humming issue with larger step sizes.
* the UGS DRO issue suggests you might be using an older version like 1.0.9 with grbl 1.1.
* You might also be switching between G90 (absolute) and G91 (relative) positioning. When in G91 `G0 x0 y0 z0` won't do anything.

dre3rd, 2017-12-06:
Currently running UGS 1.0.9 with Grbl 0.9j and I believe that has fixed the DRO problem. Would like to stay with this config until I get everything ironed out unless there are known problems with this mix.
First run was with 1.0.9 and 1.1
You are dead on the money on the G90 and G91. Will have to find that and default it to G90.
I use vCarve Pro almost 100% of the time and am never concerned with g90 or g91. On the linux cnc I have used both CamBam Lathe files and VCP files. Again never am concerned with those due to the post processors for both packages.
So 2 of the three problems you addressed are I feel are fixed. (Crossed Fingers)
The third issue of acceleration is a mystery to me. I have read some about it in LinuxCNC when I got it 8 years ago. At my age I can't remember overnight sometimes. That of course means I draw a blank there. Is there some place particular that you can suggest for me to do some research on this to see if I can even begin to make sense of what the setting might need to be.

dre3rd, 2017-12-07:
Will,
Sometimes in our haste to have our problems solved we forget our social
graces. I want to thank you for taking the time to help me with this
problem I am having.

Dan

closed 2018-01-15 by dre3rd