grbl/grbl #1328: Problem GRBL v0.9 with Homing on the X axis

Thorgrlm, 2017-12-12:
Hi everyone, I have a CNC 3040 with Arduino Uno with firmware GRBL v0.9j + TB6600 drivers.

Everything works fine, but homing doesn't work.

When I press the homing button, the Z axis and the Y work fine but the X axis remains stopped and does not look for the Switch.

My configuration is this:
[img]https://i.imgur.com/txEqid0.png[/img]

I have changed the firmware to GRBL v1.1 and the same thing happens to it.

if someone could help me.... Thanks u

X3msnake, 2017-12-12:
Your image does not show

Can you jog all the axis?

Thorgrlm, 2017-12-12:
The link works perfectly for me, try again.

Yes, the CNC working fine and moving the axes works perfect, when doesn't work is in the process of homing

X3msnake, 2017-12-13:
Link works from the github not from e-mail. I dont see anything wrong with the setings. 

Can you post a link to a video?

Thorgrlm, 2017-12-13:
Yes, this is the link
https://youtu.be/cq8CVUX6l1o

In the first part I do the homing so that you see it, and in the second part I deactivate the homing alarm and move the axes freely without problems

X3msnake, 2017-12-14:
How did you upload the firmware? IDE or Hex?

X3msnake, 2017-12-14:
Check this thread

https://github.com/grbl/grbl/issues/1060

2017-12-14 0:58 GMT+00:00 Vinicius Silva <x3msnake@gmail.com>:

> How did you upload the firmware? IDE or Hex?
>
>
>> Yes, this is the link
>> https://youtu.be/cq8CVUX6l1o
>>
>> In the first part I do the homing so that you see it, and in the second
>> part I deactivate the homing alarm and move the axes freely without problems
>>
>> <https://github.com/grbl/grbl/issues/1328#issuecomment-351551601>, or mute
>> the thread
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

Thorgrlm, 2017-12-14:
Hi, on that pin this solution doesn't work, on my Arduino.

What other pin could I put a filter on?

X3msnake, 2017-12-14:
can you post a picture of your board and connections please

Thorgrlm, 2017-12-15:
Yes.
I don't know if you see well the picture.
The more strange thing is in the frist moment my cnc worked fine, the homing work fine and all worked fine, but in this moment the homing don't work, I don't know of happened.

[img]https://i.imgur.com/rChMd6y.jpg[/img]
[img]https://i.imgur.com/Wf6geMl.jpg[/img]
[img]https://i.imgur.com/UEY86ZS.jpg[/img]

Thorgrlm, 2017-12-16:
Hi, good morning at all people.

I have done a test. I have installed a logic analyzer in the signal input of the TB6600 driver to see the signals while the CNC does the work and these are the results

The eje X in homing mode
[img]https://i.imgur.com/h3MLm12.png[/img]

The eje X in normal work
[img]https://i.imgur.com/56XN24w.png[/img]

The eje Y in homing mode
[img]https://i.imgur.com/nDu3Qlz.png[/img]

The eje Y in normal work
[img]https://i.imgur.com/Y17RoQS.png[/img]

The eje Z in homing work
[img]https://i.imgur.com/7EC2Reu.png[/img]

The eje Z in normal work
[img]https://i.imgur.com/9l1MIcM.png[/img]

All the tests that I have done the axis X in the homing work doesn't work the signals that handle the TB6600, but in free movement that works well

X3msnake, 2017-12-16:
I would redo the connections with propper connectors. Those connections are
too shady

You are most likely suffering from some kind of noise. Try setting the
sender to verbose to see if a error is being sent

X3msnake, 2017-12-17:
Try disconnecting the endstop of the axis that is not moving.

If it is noise triggering it should move.

Beware the machine will not stop on that Axis (if indeed it moves)

What did you change from when it was working to when it stopped working

No dia sábado, 16 de dezembro de 2017, Vinicius Silva <x3msnake@gmail.com>

Thorgrlm, 2017-12-18:
Hi, the problem are fixed, the switch of eje X it was in short circuit

Thanks you for your help ;)

X3msnake, 2017-12-18:
Thank you for posting back

I'm glad you managed to fixed it.

Best regards and good machining ;)

closed 2017-12-18 by chamnit