grbl/grbl #1332: grbl 1.1 freeze in between

dhunjoshi, 2017-12-18:
I m using grbl 1.1 when sent command G0 X800Y400 grbl freeze in between and don't accept new command and need to restart Arduino uno.
I change max x length to 800 y length to 400 and x rate to 60000mm/min y rate 60000mm/min x acc 200mm/s2 y acc 200mm/s2

109JB, 2017-12-18:
Please show all of your settings.  I suspect that you are exceeding 30kHz step rate

jahnj0584, 2017-12-18:
Unless you have a coarse steps/mm this is just exceeding the Hz rate of the
arduino

dhunjoshi, 2017-12-18:
$12 | Arc tolerance | 0.002
-- | -- | --
$13 | Report in inches | 0
$20 | Soft limits enable | 0
$21 | Hard limits enable | 0
$22 | Homing cycle enable | 0
$23 | Homing direction invert | 0
$24 | Homing locate feed rate | 25.000
$25 | Homing search seek rate | 500.000
$26 | Homing switch debounce delay | 250
$27 | Homing switch pull-off distance | 1.000
$30 | Maximum spindle speed | 1000
$31 | Minimum spindle speed | 0
$32 | Laser-mode enable | 0
$100 | X-axis travel resolution | 250.000
$101 | Y-axis travel resolution | 250.000
$102 | Z-axis travel resolution | 250.000
$110 | X-axis maximum rate | 60000.000
$111 | Y-axis maximum rate | 60000.000
$112 | Z-axis maximum rate | 500.000
$120 | X-axis acceleration | 200.000
$121 | Y-axis acceleration | 200.000
$122 | Z-axis acceleration | 100.000
$130 | X-axis maximum travel | 800.000
$131 | Y-axis maximum travel | 400.000
$132 | Z-axis maximum travel | 150.000

109JB, 2017-12-18:
You are exceeding the capabilities of Grbl by about 8 times.

Take x for example.  Steps/mm = 250, max rate =60000 mm/min so,

250 * 60000 / 60 = 250000 steps/sec = 250kHz

Grbl on a 328P chip can only work reliably to 30 kHz, maybe up to 40 kHz but not higher unless ported to a faster chip.

chamnit, 2017-12-18:
250kHz is higher than most, if not all, ARM based CNC firmwares can do reliably. Don't expect other firmwares to do that.

closed 2017-12-18 by chamnit

ClarkDailey, 2020-02-12:
My DIY CNC end mill uses GRBL v1.1 Arduino AtMega2560.  The end mill is a heavy duty Grizzly G1006 with off-the-shelf stepper motors, drivers, and a custom Arduino CNC shield.  I ran into a problem when cutting one-hour 3D relief carvings.  It appeared that GRBL would freeze at random times, and would not accept any more commands from the Universal G-Code Sender program running on my laptop.  In order to debug the problem, I added 3 LEDs and hacked the GRBL firmware to toggle them.  The red LED is toggled in the stepper motor interrupt routine (ISR) every 1000 steps.  The yellow LED is toggled every line motion cut (mc_line subroutine).  The green LED is toggled whenever GRBL checks for a byte on the serial port (protocol_main_loop subroutine).   Running another one-hour job, motion again froze, as usual.  The green LED was still toggling and checking the serial port, which indicated to me that the GRBL firmware was not frozen, but the communications hardware was latched up.  Unplugging the USB cable at the laptop side, and plugging it in again, the 3 LEDs went through the boot-up process, which indicated to me that the Arduino microprocessor rebooted, and all positioning information was then lost.  This proved to me that it is was electro-magnetic-interference problem on the USB cable.  I use a gutted desktop computer case to contain the Arduino, the power supplies, and stepper motor drivers.  The wiring is clean.  No motor cables or power cables are close to the USB cable.  To continue testing, I replaced the shielded 2 foot USB cable, with a 2 foot USB cable with ferrite beads on both ends.  The iron ferrite beads are inductor chokes that filter out high frequency RF generated by the motors and electronic chopping circuitry.  This did the trick.  I have run several one-hour CNC jobs without any motion freeze.  Random problems caused by EMI are difficult to solve, but I think this one is solved for me.  There are a number of postings on the internet describing this GRBL freezing problem.  Although GRBL is taking the rap for this, it is not the problem.  The Arduino could benefit by adding some filtering on the USB differential signals, but this would add cost.   So the filtering has to be done on the front end cabling.  Hope this helps anyone having this problem.