grbl/grbl #1335: Work the Closed loop motors in GRBL?

Thorgrlm, 2018-01-03:
Hi my friends, I have a doubt.
looking on the internet, I found a hybrid closed loop stepper motor, with an encoder to avoid missing steps.
[https://es.aliexpress.com/store/product/57-closed-loop-57BH256E-1000-DC-stepper-motor-2HSS57-driver-1N-m-Nema-23-Hybird-closed/1021179_32686929785.html?spm=a219c.search0204.3.22.7tAzEv&ws_ab_test=searchweb0_0,searchweb201602_4_10152_10151_10065_10068_10344_10547_10342_10343_10340_10548_10341_10084_10083_10307_10301_10303_10059_10184_10314_10534_100031_10604_10103_10142,searchweb201603_36,ppcSwitch_3_ppcChannel&algo_expid=6f269521-23b6-479c-b907-f4b8a8162c86-2&algo_pvid=6f269521-23b6-479c-b907-f4b8a8162c86&priceBeautifyAB=0](url)

Are these types of engines compatible with GRBL?

thanks you

X3msnake, 2018-01-03:
> 57BH256E-1000-DC-stepper-motor-2HSS57-driver-1N-m-Nema-
> 23-Hybird-closed/1021179_32686929785.html?spm=a219c.
> search0204.3.22.7tAzEv&ws_ab_test=searchweb0_0,
> searchweb201602_4_10152_10151_10065_10068_10344_10547_10342_
> 10343_10340_10548_10341_10084_10083_10307_10301_10303_10059_
> 10184_10314_10534_100031_10604_10103_10142,searchweb201603_36,ppcSwitch_
> 3_ppcChannel&algo_expid=6f269521-23b6-479c-b907-f4b8a8162c86-2&algo_pvid=
> 6f269521-23b6-479c-b907-f4b8a8162c86&priceBeautifyAB=0 <http://url>

From what i understand the hybrid drivers do the closed loop internally so
for grbl these drivers are just the same as normal ones.

Thorgrlm, 2018-01-03:
ok, the doubt I was that if the GRBL would have to control the motor loop. if it works that way that you say, it would be the perfect solution for greater precision.

Thank you nd sorry my poor english :(

X3msnake, 2018-01-03:
https://www.automationtechnologiesinc.com/wp-content/uploads/downloads/2012/06/KL-5080H.pdf

If you look for example at the pinout for a hybrid stepper you see that it
is similar to a standart stepper with the exception of the self logic tied
to the encoder on the motors

https://images-na.ssl-images-amazon.com/images/I/416SnwdQvFL.jpg

Thorgrlm, 2018-01-03:
OK, I had not seen a scheme this device, thanks for your help, for my next evolution of my CNC I will install this system.

jahnj0584, 2018-01-03:
I don't think GRBL has an alarm pin for the lost steps so there would be no
benefit to using closed loop.

X3msnake, 2018-01-03:
I think you don't need the alarm pin the driver takes care of making sure
what you asked is executed the alarm ia only for failure modes. But i might
be wrong

https://youtu.be/xJ97x_2b4yk

X3msnake, 2018-01-03:
In fact you could probably repourpouse the door pin to use the alarm

109JB, 2018-01-03:
A few things.

1. Closed loop DOES NOT mean more precision. The closed loop steppers are still steppers and can only position based on the steps per revolution. If set up for say 250 steps/mm, then 1 step is still 1/250th of a millimeter regardless of closed loop or open loop.

2. The closed loop steppers do not really need a alarm pin to work. The way they work is not to throw an alarm, but to make up for lost steps by adding back in the steps that were lost. They do this internally and don't need an external controller to do this. The only reason you would need an alarm type pin is if the driver has the capability to report errors or following errors that exceed limits. Even then it would be optional.

X3msnake, 2018-01-04:
Since the driver can achieve higher microstepping it might be be dependable
on the encoder resolution more than the stepper motor step grade

X3msnake, 2018-01-04:
For example the HSS86 2-phase Hybrid Stepper Servo Driver

1.2.1 Stepper motor closed loop system, never lose step.
1.2.2 Improve motor output torque and working speed.
1.2.3 Automatic current adjustment based on load, lower temperature rising.
1.2.4 Suitable for all mechanical load conditions (include low rigidity
belt pulley and
wheel), no need to adjust gain parameter.
1.2.5 Motor work smoothly and low vibration, high dynamic performance at
acceleration and deceleration.
1.2.6 No vibration from high speed to zero speed
1.2.7 Drive nema 34 series 4N.m, 8N.m, 12N.m closed loop stepper motor.
1.2.8 Pulses response frequency can reach 200KHZ
1.2.9 16 kinds microsteps choice, highest 51200microsteps/rev.
1.2.10 Voltage range: AC24~80V or DC30V~110V
1.2.11 Over-current, over-voltage and position ultra difference protection
function.

109JB, 2018-01-04:
The fact is that one step will still be one step.  Open loop stepper drivers can also be set for high microstepping just like the closed loop steppers, even as high as 51,200 depending on the driver, but it is not usable especially on a Grbl controlled stepper.  If using 51,200 microstep/rev, with Grbl having a max of 30,000 step pulses per second you achieve roughly 3/5 rev per second or only 36 rpm. So, extremely high microstepping is useless in almost all applications.

A properly set up open loop stepper system won't lose steps. If steps are being lost then either the motor is improperly spec'd for the application, the accel is too high, or the rate is too high. 

The benefit of closed loop would only be possibly being able to push the acceleration and maximum rate up a small amount, and peace of mind, but it will not increase precision over a properly set up open loop system.

Thorgrlm, 2018-01-04:
But this system is safer when you use microsteps since the motor doesn't vibrate, instead the normal stepper motor has problems in the microsteps, or that is the theory.
by eliminating that small error, in my opinion the accuracy increases.
or no?

jahnj0584, 2018-01-04:
No. 1/10 of a mm is always 1/10 even with servos

Thorgrlm, 2018-01-04:
True, but the theory says that a stepper motor is not stable in microspasses and this solution is stable or that is what I read in the reports.

X3msnake, 2018-01-04:
Well it will work with GRBL :)

And of course it will be as good as the mechanical build and electrical
build is. Still it is more expensive, not so much as true servo drivers.

I wonder if these drivers are more imune to electrical noise than the
standart open loop.

mat60, 2018-01-07:
Hello, i try this kind of motor White grbl 1.1
I use a 3 phases motor. I don't use thé alarm pin.
Il works Good with à leadshine nema 23 3nm. 
Enable pin must be allways on.

Cromaglious, 2018-01-08:
Closed loop with alarm is always better than open loop.  There is always a chance of getting a bound up lead screw or something that'll cause a lost step. Most closed loop steppers offer a signal when on station. While on grbl there probably not a way to capitalize on that, i can see a pause routine in marlin being added in and linuxcnc as well.  My alarms on my closed loop steppers trigger an estop.

Cromaglious, 2018-01-08:
I'm running both JMC-Motor.com NEMA 23 2Nm iHSS57-36-20 and NEMA 24 3Nm iHSS60-36-30 Integrated closed loop steppers stepper motor, encoder, and driver is all in one unit.  Yes they are more expensive, but I have peace of mind when running a job.

andrewismoody, 2018-02-25:
The problem with closed loop (that I'm discovering first-hand) is that each axis is independent.  The encoder does notice the missed step and compensate for it, but at the expense of time.  As you might imagine, not all axes get 'bound up' at the same time - usually just one.  This means that one axis is slightly behind on motion while the other two (or three or whatever) continue along their merry way, none the wiser.  The result being that the spindle does eventually land in the right position, the path it took to get there may not be as planned.  Think about making an XY move where the X axis a half second behind.  The angle of the path will be much different than the expected angle and therefore result in cutting (or not cutting) material that was expected.
Would like to see GRBL support a feedback mechanism with closed loop controllers (like Smart Stepper and MechAduino) so that the encoder can tell the controller to hold up while it corrects its position.
I had thought about hooking the 'error' pin of the closed loop encoder to the feed hold pin, but I don't know if this would quite do what I want (will the feed hold be removed when the error pin reverts?).  And I'm not an EE, so I don't really know how to determine what would be required to 'convert' the signals to be compatible.
This feels like it would need to be a 'supported feature' of GRBL to really work properly.

andrewismoody, 2018-02-25:
Just realizing this is on the old Grbl/Grbl.  I'm going to open this on the Gnea/Grbl with my write-up.

langwadt, 2018-02-25:
@andrewismoody if that happens you are pushing your machine too hard

pauljay, 2018-03-07:
Every closed loop stepper operates in open loop in most cases. If error of loss step occurs, then closed loop control path starts. JMC Motor has low resolution optical encoder less than microstepping level they provide. If closed loop stepper works in steady closed loop mode, it’s Servo Motor.

X3msnake, 2018-03-07:
There are true closed loop steppers they are called hybrid

109JB, 2018-03-07:
They are closed loop back to the driver, but not back to the control software.  I would not call that "true closed loop".  In order to be true close loop it would have to have feedback to the control software so that if one axis stalls, the controller can slow the non stalled axes. So, imagine a 45 degree move in XY and part way into the move the X axis stalls.  If there is no feedback back to the control software it will  keep sending step pulses out and the Y axis keeps moving even though X has stalled.

X3msnake, 2018-03-07:
http://www.jvl.dk/1109/jvl-closed-loop-stepper-motors

chamnit, 2018-03-07:
In my view, closed loop control **motors** only give you the ability to more safely run at the maximum capacity of the motor and some feedback if that fails. Open loops always has to derate itself to ensure it never loses position. 

This is all fine and dandy, but the problem is syncing them all together to get a coherent multi-axis motion so none of the motors exceed their rating. Once again, this is what Grbl basically does. It tells the motors where and when to go based on its internal planner physics. The only thing closed loop motors can do is tell Grbl if it didn't make it there on time.

Just to be explicitly clear, there is no problem with Grbl running hybrid motors and closed-loop servos, as long as the motor controllers have a step and direction input. Grbl, as it is now, doesn't have the ability to do anything about position errors thrown by the motor controllers. That is not to say it won't have this in the near future with the ARM version, but a quick way to hack this is to tie in the error pin on the motor controllers to Grbl's feed hold or safety door input.

Cromaglious, 2018-03-08:
I exclusively run JMC-motor.com integrated closed loop steppers. iHSS57-36-20 (NEMA 23 295ozin), iHSS60-36-30 (NEMA 24 425oz in) as well as the closed loop NEMA 17 using Linuxcnc/Mach3/Smoothieware/Marlin/GRBL I tie the Alarm off the stepper/driver to the appropriate ESTOP or FEEDHOLD or SAFETY INTERLOCK pins.  They have saved my rearend multiple times from stock moving, leather curling up or other issues. using NEMA 17 closed loop in my K40 laser cutters as well.  How many times has air assist blown something or 3 things out of position and the head comes along and stalls on a pile of stuff.  I haven't gotten a 3d printer yet, but when I do, it'll be driven by closed loop steppers with ball screws, no belts for me.  One K40 is GRBL controlled, the other is currently Marlin.  CNC 3040 is LinuxCNC, Sieg X2 (Harbor Freight 44991) mini mill is Smoothieware and a Rong Fu 30 (Harbor Freight 33686) is Mach 3.

MarcusWIA, 2018-04-19:
@chamnit 

> The only thing closed loop motors can do is tell Grbl if it didn't make it there on time.
How do I modify/tell grbl this?
Can you give me some specifics?
I need to implement at least an error sending/ showing for this kind of thing, this being the only way I can pass my self made CNC as a "Automated machine" according to Diploma Project Coordinator and some other teachers in the graduation committee.

tsipa, 2020-09-13:
Half of the thread is obvious BS, i dunno what would be in head of person who saying that alarm is not needed. Why it's god damn there if it is not needed? What do you expect would prevent other 2 axis from doing stupid things when 1 axis lost step?