grbl/grbl #1336: error 22

lemons342, 2018-01-04:
Hi, I'm doing a project for a hardware class and I'm doing a CNC controller. I'm going off the tutorial here: http://www.tinkernut.com/portfolio/hack-old-cd-roms-into-a-cnc-machine/#lightbox[9633]/13/
But instead of having x, y, and z axis I only have an x and y. When I try to run the program I get an error 22 message. I believe I have the wrong code and or the wrong hex file. Could someone please give me some advice and possibly download links to the right software files and executables to use.
Thanks.

X3msnake, 2018-01-04:
> portfolio/hack-old-cd-roms-into-a-cnc-machine/#lightbox[9633]/13/

https://github.com/gnea/grbl/issues/119

That is a problem of the sender you are trying to use;

Try using UGS, BCNC, Candle or any other sender

https://github.com/gnea/grbl/wiki/Using-Grbl

lemons342, 2018-01-05:
Thanks for the response. Also, do you think that having 2 wire disk drive motors instead of 4 will be a problem?

X3msnake, 2018-01-06:
Yes it will be a problem

2 wire motors are dc motors not steppers :(

You need steppers to use with GRBL.

https://www.modmypi.com/blog/whats-the-difference-between-dc-servo-stepper-motors